mossdef-org/source.mossdef.org #1: Recursive Dependency Error when building against LEDE master

njfox, 2018-04-02:
Hello,

I'm receiving the following error when running `make defconfig` or `make menuconfig` when configuring a build for the LEDE `master` branch. This error does NOT occur when building against LEDE 17.01:

```
$ make defconfig
tmp/.config-package.in:90388:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:90388:	symbol PACKAGE_vpn-policy-routing depends on PACKAGE_vpnbypass
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:90407:	symbol PACKAGE_vpnbypass depends on PACKAGE_vpn-policy-routing
#
# configuration written to .config
#
```

I tried digging but I can't quite figure out what is causing the issues. Let me know if I can provide any other information that would be helpful for troubleshooting.

stangri, 2018-04-03:
Ah, there was an erroneous entry in the VPR makefile. I've removed it and pushed the fix to the repo. If it still gives you problems, remove [this line](https://github.com/stangri/openwrt_packages/blob/master/vpn-policy-routing/Makefile#L18) from Makefile.

njfox, 2018-04-03:
Thanks...I still received the error after the commit but removing the `CONFLICTS` line resolved the issue.

stangri, 2018-07-13:
Now that I've had time to reflect on that, do you have both vpnbypass and vpn-policy-routing selected (with the star) for build? I have a bare-bones .config file and I build packages from the command line, but I have no problems building either package.

closed 2018-07-15 by stangri

njfox, 2018-07-27:
Just an update, I've tried to build this in a completely clean environment with the latest version of this repo and OpenWRT master and I'm still getting the error:
```
$ cp ../policy_config .config
[nick@arch openwrt]$ make defconfig
Collecting package info: done
WARNING: Makefile 'package/feeds/luci/luci-app-samba4/Makefile' has a dependency on 'samba4-server', which does not exist
tmp/.config-package.in:92211:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:92211:	symbol PACKAGE_vpn-policy-routing depends on PACKAGE_vpnbypass
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:92230:	symbol PACKAGE_vpnbypass depends on PACKAGE_vpn-policy-routing
```
Here is an excerpt from my .config showing that vpnbypass is not selected:

```
$ grep vpn .config
# CONFIG_PACKAGE_luci-app-openvpn is not set
CONFIG_PACKAGE_luci-app-vpn-policy-routing=y
# CONFIG_PACKAGE_luci-app-vpnbypass is not set
# CONFIG_PACKAGE_luci-proto-vpnc is not set
# CONFIG_PACKAGE_chaosvpn is not set
# CONFIG_PACKAGE_openvpn-easy-rsa is not set
# CONFIG_PACKAGE_openvpn-mbedtls is not set
# CONFIG_PACKAGE_openvpn-nossl is not set
CONFIG_PACKAGE_openvpn-openssl=y
# CONFIG_PACKAGE_softethervpn is not set
# CONFIG_PACKAGE_vpnc is not set
# CONFIG_PACKAGE_vpnc-scripts is not set
CONFIG_PACKAGE_vpn-policy-routing=y
# CONFIG_PACKAGE_vpnbypass is not set
```
Any ideas?