mossdef-org/source.mossdef.org #2: Invalid luci.mk path

njfox, 2018-04-02:
I'm getting the following error when updating the openwrt_packages feed from git. I also tried cloning the repo locally and using `src-cpy` and `src-link` but the results were the same:

```
Updating feed 'stangri' from 'https://github.com/stangri/openwrt_packages.git' ...
Cloning into './feeds/stangri'...
remote: Counting objects: 148, done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 148 (delta 13), reused 132 (delta 7), pack-reused 0
Receiving objects: 100% (148/148), 1.05 MiB | 1.26 MiB/s, done.
Resolving deltas: 100% (13/13), done.
Create index file './feeds/stangri.index' 
ERROR: please fix feeds/stangri/luci-app-advanced-reboot/Makefile - see logs/feeds/stangri/luci-app-advanced-reboot/dump.txt for details
ERROR: please fix feeds/stangri/luci-app-easyflash/Makefile - see logs/feeds/stangri/luci-app-easyflash/dump.txt for details
ERROR: please fix feeds/stangri/luci-app-simple-adblock/Makefile - see logs/feeds/stangri/luci-app-simple-adblock/dump.txt for details
ERROR: please fix feeds/stangri/luci-app-vpn-policy-routing/Makefile - see logs/feeds/stangri/luci-app-vpn-policy-routing/dump.txt for details
ERROR: please fix feeds/stangri/luci-app-vpnbypass/Makefile - see logs/feeds/stangri/luci-app-vpnbypass/dump.txt for details
ERROR: please fix feeds/stangri/luci-mod-alt-reboot/Makefile - see logs/feeds/stangri/luci-mod-alt-reboot/dump.txt for details
```

Editing the `Makefile`s and changing the `../../luci.mk` path to `../../luci/luci.mk` resolves the issue for me. I can submit a PR if you want but I wanted to make sure I wasn't doing something wrong on my end first.

stangri, 2018-04-03:
I didn't intend this repo to be used as a feed, try cloning individual packages to: `packages/` and `packages/luci` of your SDK.

I will try to look into converting it to a proper feed in the future.

njfox, 2018-04-03:
Thanks, sorry about the confusion!

stangri, 2018-04-11:
Ah, I found the reason for the confusion -- the README actually stated this repo can be used as a feed, which I fixed. I will try to set up another repo with aliases for this one and a proper directory structure to work as a feed.

closed 2018-05-07 by stangri