mossdef-org/source.mossdef.org #6: split DNS

n8v8R, 2018-05-06:
Would be possible to add split DNS, say:

- client 1 routed via WAN with WAN DNS resolver
- client 2 routed via WAN but with router (127.0.0.1) DNS resolver
- client 3 routed via VPN with VPN DNS resolver
- client 4 routed via VPN but with router (127.0.0.1) DNS resolver

stangri, 2018-05-07:
Do you have an idea of how to implement this?

n8v8R, 2018-05-07:
Had thought initally of ipset-dns
https://git.zx2c4.com/ipset-dns/about/
https://wiki.openwrt.org/doc/uci/ipset-dns
https://wiki.openwrt.org/doc/howto/netfilter

But reading from the first it seems to have been implemented in dnsmasq

> This functionality has now been written directly into dnsmasq, which should be much easier to use than this project. See the --ipset option.

Which may though conflict or not work with boxes utilizing a different resolver, e.g. unbound or knot. 

Whilst I would like such split DNS feature (apparently since posting it here) it may not be that common of a request and perhaps not worth pursuing further, meaning please feel liberty at any time to close the ticket.

stangri, 2018-07-13:
Yeah, if you want to use different resolvers for different local clients, it can be implemented with dnsmasq already.

closed 2018-07-13 by stangri