mossdef-org/source.mossdef.org #9: udp traffic

n8v8R, 2018-05-12:
Router Turri Omnia
OpenWrt omnia 15.05 r47055 / LuCI cc8b99aacec15490bb725fe53abefa1156cb5fe3 branch (git-18.123.42193-cc8b99a)
Kernel 4.4.131-a2dbf3bef3d0c1f725e0a5f0801935a1-2
BusyBox v1.25.1

With `traceroute` wrapped in BusyBox it is lacking the `-T` option (TCP) and thus only UDP can be utilized.
It seems that UDP traffice however is routed outside the VPN iface designated by PBR despite being set in advanved options:

```
config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	list ignored_interface 'guest'
	option enabled '1'
	option udp_proto_enabled '1'
```

Running `traceroute` is showing as heading trough the WAN only. Suppose that holds true for other UDP traffic then too.

stangri, 2018-07-07:
Did you run traceroute on router? Enable OUTPUT chain.

n8v8R, 2018-07-14:
Yes, did that https://github.com/stangri/openwrt_packages/issues/8#issuecomment-405020057

Discounting the Win client (to my understanding `tracert` is using `icmp` only) it would have been expected that traceroute from the router and the android client being `udp` to be routed through the vpn iface but it does not.

n8v8R, 2018-07-15:
With VPR _disabled_ and 

```
config wireguard_wg0
	list allowed_ips '0.0.0.0/0'
	option route_allowed_ips '1'
```
udp traffic from the router and its clients are routed through the vpn with no issue (as expected).