mossdef-org/source.mossdef.org #10: [bug?] no wireguard traffic

n8v8R, 2018-05-13:
Router Turri Omnia
OpenWrt omnia 15.05 r47055 / LuCI cc8b99aacec15490bb725fe53abefa1156cb5fe3 branch (git-18.123.42193-cc8b99a)
Kernel 4.4.131-a2dbf3bef3d0c1f725e0a5f0801935a1-2
kmod-wireguard  4.4.131+0.0.20171017-..1-2 (`option route_allowed_ips '0'`)
vpn-policy-routing 0.0.1-25

There is no traffic between the `wg ifaces`, peers cannot be pinged or tracerouted. Switching option `udp_proto_enabled` did not make a difference. Made sure that the WG tunnel is actual up and the peers are connected.

```
config policy
	option local_addresses '192.168.112.120'
	option comment 's5'
	option interface 'wan'

config policy
	option comment 's1'
	option local_addresses '192.168.112.121'
	option interface 'wan'

config policy
	option comment 'da'
	option local_addresses '192.168.112.160'
	option interface 'wan'

config policy
	option comment '1750e'
	option local_addresses '192.168.112.190'
	option interface 'wan'

config policy
	option comment 'dvb-c'
	option local_addresses '192.168.112.191'
	option interface 'wan'

config policy
	option comment 'stv9s'
	option local_addresses '192.168.112.122'
	option interface 'wan'

config policy
	option interface 'wan'
	option comment 'oc_vpscp'
	option remote_addresses '<redacted>'
	option remote_ports '4083'

config policy
	option comment 'lan'
	option local_addresses '192.168.112.12/24'
	option interface 'wg0'

config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	list ignored_interface 'guest'
	option enabled '1'
	option udp_proto_enabled '1'
```

```
/etc/init.d/vpn-policy-routing reload
Creating table 'wan/pppoe-wan/<redacted' [✓]
Creating table 'wg0/wg0/172.25.120.2' [✓]
Routing 's5' via wan [✓]
Routing 's1' via wan [✓]
Routing 'da' via wan [✓]
Routing '1750e' via wan [✓]
Routing 'dvb-c' via wan [✓]
Routing 'stv9s' via wan [✓]
Routing 'oc_vpscp' via wan [✓]
Routing 'lan' via wg0 [✓]
vpn-policy-routing 0.0.1-25 started on wan/pppoe-wan/<redacted> wg0/wg0/172.25.120.2 [✓]
vpn-policy-routing 0.0.1-25 monitoring interfaces: wan wg0 [✓]
```

```
/etc/init.d/vpn-policy-routing support
vpn-policy-routing 0.0.1-25 running on OpenWrt Chaos Calmer. WAN (IPv4): wan/dev/<redacted>. WAN (IPv6): wan/dev6/::/0.
============================================================
Dnsmasq version 2.78  Copyright (c) 2000-2017 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect inotify
============================================================
Routes/IP Rules
default         ppp-gw-pop107.r 0.0.0.0         UG    0      0        0 pppoe-wan
32712:  from all fwmark 0x20000 lookup 202
32713:  from all fwmark 0x10000 lookup 201
IPv4 Table 201: default via <redacted> dev pppoe-wan
IPv4 Table 202: default via 172.25.120.2 dev wg0
============================================================
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s 192.168.112.0/24 -m comment --comment lan -c 213 33355 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d <redacted>/32 -p udp -m multiport --dports 4083 -m comment --comment oc_vpscp_<redacted> -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -d <redacted>/32 -p tcp -m multiport --dports 4083 -m comment --comment oc_vpscp_<redacted> -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.122/32 -m comment --comment stv9s -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.191/32 -m comment --comment dvb-c -c 2 120 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.190/32 -m comment --comment 1750e -c 2 120 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.160/32 -m comment --comment da -c 1 40 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.121/32 -m comment --comment s1 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.112.120/32 -m comment --comment s5 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -m set --match-set wg0 dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create wg0 hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]
```

stangri, 2018-07-10:
I can't reproduce. With the properly set up wg tunnel I have no problems sending some traffic thru the tunnel. How did you test the wg tunnel connectivity before starting VPR?

n8v8R, 2018-07-10:
I suggest to close this issue then. It was a while ago and VPR as well as TO have evolved since then. And I dropped WG for now.

n8v8R, 2018-07-13:
Back to it, I am afraid. WG tunnel is up and each peer can ping/traceroute the other, ssh working as well.

Then starting VPR for the wg iface on the router and there is no tcp traffic in the tunnel and thus ssh sessions are cut off. udp traffic is apparently not impeded since the wg tunnel remains intact and the peers can still ping each other, including those behind NAT at the remote end.

Reversing VPR wg to the router's eth is restoring connectivity to the router's clients except for the ssh client. For some reason only after restarting the router's network the ssh client is able to get back into remote host.

This with

* luci-app-vpn-policy-routing | git-18.192.22043-6f77..-27
* vpn-policy-routing | 0.0.2-11
* kmod-wireguard | 4.4.138+0.0.20171017-..e-1
* luci-proto-wireguard | git-18.145.30016-526a876-1
* wireguard-tools | 0.0.20171017-1
* TOS 3.10.3 (OpenWrt omnia 15.05 r47055 / LuCI 526a8767846acbe57c521912b35feb4d97354db6 branch (git-18.145.30016-526a876) 
* kernel 4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1

n8v8R, 2018-07-13:
`/etc/init.d/vpn-policy-routing support`

> vpn-policy-routing 0.0.2-11 running on OpenWrt Chaos Calmer. WAN (IPv4): wan/dev/84.[truncated].

> ============================================================
> Dnsmasq version 2.78  Copyright (c) 2000-2017 Simon Kelley
> Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect inotify

> ============================================================
> Routes/IP Rules
> default         ppp-gw-pop107.r 0.0.0.0         UG    0      0        0 pppoe-wan
> IPv4 Table 201: default via 84.[truncated] dev pppoe-wan
> IPv4 Table 201 Rules:
> 32747:  from all fwmark 0x10000 lookup 201
> IPv4 Table 202: default via 192.168.112.12 dev br-lan
> IPv4 Table 202 Rules:
> 32746:  from all fwmark 0x20000 lookup 202

> ============================================================
> IP Tables PREROUTING
> -N VPR_PREROUTING
> -A VPR_PREROUTING -s 192.168.112.0/24 -m comment --comment lan -c 86 4426 -j MARK --set-xmark 0x20000/0xff0000
> -A VPR_PREROUTING -d 179.[truncated]/32 -p udp -m multiport --dports 56009 -m comment --comment oc_ssh -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -d 179.[truncated]/32 -p tcp -m multiport --dports 56009 -m comment --comment oc_ssh -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -d 179.[truncated]/32 -p udp -m multiport --dports 5904 -m comment --comment oc_vnc -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -d 179.[truncated]/32 -p tcp -m multiport --dports 5904 -m comment --comment oc_vnc -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -d 94.[truncated]/32 -p udp -m multiport --dports 4083 -m comment --comment oc_[truncated]_com -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -d 94.[truncated]/32 -p tcp -m multiport --dports 4083 -m comment --comment oc_[truncated]_com -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.122/32 -m comment --comment stv9s -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.191/32 -m comment --comment dvb-c -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.190/32 -m comment --comment 1750e -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.160/32 -m comment --comment da -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.121/32 -m comment --comment s1 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -s 192.168.112.120/32 -m comment --comment s5 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -m set --match-set wg0 dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
> -A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000

> ============================================================
> Current ipsets
> create wan hash:net family inet hashsize 1024 maxelem 65536 comment
> create tun hash:net family inet hashsize 1024 maxelem 65536 comment
> create wg0 hash:net family inet hashsize 1024 maxelem 65536 comment

> ============================================================
>

stangri, 2018-07-13:
I would have to look into over the next week.

n8v8R, 2018-07-13:
Just hold your horses please. It might be a WG issue. I will get back if it still persists afer some WG debugging

n8v8R, 2018-07-13:
Done with WG debugging, that is sorted but unfortunately still no WG tcp traffic with VPR. Sytoms are as decribed in the recent posts.

n8v8R, 2018-07-13:
With VPR _enabled and all traffic assigned to WAN_ it ceases all TCP traffic of the WG tunnel.

Only when VPR is _disabled_ is there TCP traffic in the WG tunnel

n8v8R, 2018-07-15:
With VPR _disabled_ and 

```
config wireguard_wg0
	list allowed_ips '0.0.0.0/0'
	option route_allowed_ips '1'
```
all (tcp/udp) traffice from the route and its clients are routed through the vpn with no issue (as expected).

---
If it would help I could provide a WG remote endpoint for testing.

n8v8R, 2018-07-18:
Tested different scenarios and it is apparent that VPR is somehow disrupting the tcp traffice in the wg tunnel.

All scenarios with VPR `enabled`

---
1. default route via wg iface and all VPR policies only via wan iface

```
config wireguard_wg0
	list allowed_ips '0.0.0.0/0'
	option route_allowed_ips '1'
```

No issues, all wan traffic assigned via VPR is routed accordingly and the remainder via wg tunnel.

---
2. certain remote subnets are routed via wg and routes added to the table by wg, various VPR policies via wan and one VPR policy (at the bottom) for the remainder traffice via wg
```
config wireguard_wg0
	list allowed_ips '172.23.0.0/16'
	option route_allowed_ips '1'
```

```
config policy
	option comment 'lan'
	option local_addresses '192.168.112.0/24'
	option interface 'wg0'
```

Traffic via wan no issues, no tcp traffic inside the wg tunnel. Disabling VPR revives the tcp traffic in the wg tunnel.

---
3. certain remote subnets are routed via wg and _no_ routes added to the table by wg, various VPR policies via wan and one VPR policy (at the bottom) for the remainder traffice via wg

```
config wireguard_wg0
	list allowed_ips '172.23.0.0/16'
	option route_allowed_ips '0'
```

```
config policy
	option comment 'lan'
	option local_addresses '192.168.112.0/24'
	option interface 'wg0'
```
Traffic via wan no issues, no tcp traffic inside the wg tunnel. Disabling VPR revives the tcp traffic in the wg tunnel.

n8v8R, 2018-11-29:
Sorted, been a misconception my end in the settings of WG

closed 2018-11-29 by n8v8R