mossdef-org/source.mossdef.org #12: (openconnect → l2tp → dhcp wan) sequence breaks some logic

pohmelie, 2018-05-23:
I have such weird configuration: openconnect → l2tp → dhcp wan. And your script detects l2tp as wan [here](https://github.com/stangri/openwrt_packages/blob/master/vpn-policy-routing/files/vpn-policy-routing.init#L375). When l2tp detected as wan branch there is some error in `ipt` function call at point `--set-xmark /0xff0000`, it looks like `mark` variable is empty, so `iptables` call fails.
I change order of checks (tunnel before wan) and it looks like my case is resolved, but I don't really now how this change will affect other cases.
I'm tired of diggin into bash hieroglyph-spaghetti :rofl: so no extra information. If you want I can paste any config you need.

stangri, 2018-05-29:
Hey Nikita,
can you please post/e-mail me the output of `ifconfig` and `ip -4 route`?

pohmelie, 2018-05-29:
```
root@LEDE:~# ifconfig 
eth1      Link encap:Ethernet  HWaddr DC:EF:09:E6:9A:38  
          inet6 addr: fe80::deef:9ff:fee6:9a38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9742131 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6302257 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2381854139 (2.2 GiB)  TX bytes:1421802856 (1.3 GiB)
          Interrupt:5 

eth1.1    Link encap:Ethernet  HWaddr DC:EF:09:E6:9A:38  
          inet addr:10.42.201.96  Bcast:10.42.207.255  Mask:255.255.248.0
          inet6 addr: fe80::deef:9ff:fee6:9a38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9742131 errors:0 dropped:257508 overruns:0 frame:0
          TX packets:6302236 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:10796430373 (10.0 GiB)  TX bytes:1396591270 (1.2 GiB)

eth1.2    Link encap:Ethernet  HWaddr DC:EF:09:E6:9A:38  
          inet addr:192.168.3.1  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: fe80::deef:9ff:fee6:9a38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:1410 (1.3 KiB)

l2tp-l2tp Link encap:Point-to-Point Protocol  
          inet addr:95.27.87.7  P-t-P:95.27.64.1  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1456  Metric:1
          RX packets:8802153 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6017558 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3 
          RX bytes:10340135431 (9.6 GiB)  TX bytes:1065062519 (1015.7 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9794 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9794 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:886995 (866.2 KiB)  TX bytes:886995 (866.2 KiB)

vpn-vpn   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:192.168.1.184  P-t-P:192.168.1.184  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1325  Metric:1
          RX packets:8731433 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6014195 errors:0 dropped:1166 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:1096227389 (1.0 GiB)  TX bytes:667690019 (636.7 MiB)

wlan1     Link encap:Ethernet  HWaddr DC:EF:09:E6:9A:38  
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::deef:9ff:fee6:9a38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6535040 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9300889 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:828652820 (790.2 MiB)  TX bytes:10322897012 (9.6 GiB)

wlan1-1   Link encap:Ethernet  HWaddr DE:EF:09:E6:9A:38  
          inet addr:192.168.2.1  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::dcef:9ff:fee6:9a38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:88 errors:0 dropped:0 overruns:0 frame:0
          TX packets:112 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11124 (10.8 KiB)  TX bytes:20708 (20.2 KiB)
```
```
root@LEDE:~# ip -4 route
default dev vpn-vpn 
10.42.200.0/21 dev eth1.1  src 10.42.201.96 
10.42.200.1 dev eth1.1  src 10.42.201.96 
83.102.254.213 via 10.42.200.1 dev eth1.1 
83.102.255.51 via 10.42.200.1 dev eth1.1 
91.134.120.51 via 95.27.64.1 dev l2tp-l2tp 
95.27.64.1 dev l2tp-l2tp  src 95.27.87.7 
192.168.0.0/24 dev wlan1  src 192.168.0.1 
192.168.1.0/24 dev vpn-vpn 
192.168.2.0/24 dev wlan1-1  src 192.168.2.1 
192.168.3.0/24 dev eth1.2  src 192.168.3.1
```

stangri, 2018-05-30:
I've updated both interface and gateway detection logic in `vpn-policy-routing 0.0.2-1`, please test it and let me know.

pohmelie, 2018-05-30:
Works fine. Thank you for fast resolution! :+1:

closed 2018-05-30 by pohmelie