mossdef-org/source.mossdef.org #13: [bug] Save & Apply not having any effect after router reboot

n8v8R, 2018-06-09:
Router Turri Omnia
OpenWrt omnia 15.05 r47055 / LuCI 526a8767846acbe57c521912b35feb4d97354db6 branch (git-18.145.30016-526a876)
Kernel 4.4.134-8f7f4132dc88fb7034ce9648e5961be5-0
luci-app-vpn-policy-routing | git-18.151.61791-fed3..-23
vpn-policy-routing | 0.0.2-2

---
mentioned https://forum.turris.cz/t/vpn-policy-based-routing-possible/7204/23

After updating the router's OS from TOS 3.10 to 3.10.1 and issue #7 the workaround of getting PBR to work after a reboot is not working anymore, as in pressing `Save & Apply `(/cgi-bin/luci/admin/services/vpn-policy-routing) is not having any effect.

Only way to get PBR after a router's reboot is connecting to the TO over ssh and then from the command line

`/etc/init.d/vpn-policy-routing reload`

stangri, 2018-06-09:
Thanks for the report. I've switched from doing ```/etc/init.d/dnsmasq restart``` to just SIGHUP to dnsmasq to kill it in a few of my projects and I suspect that could be the issue.
Could you please do the following:
1. Start with the router with VPR not working.
2. Go to WebUI and do Save & Apply.
3. Connect via SSH and post the output of ```/etc/init.d/vpn-policy-routing status```.

Second test case is:
1. Start with the router with VPR not working.
2. Go to WebUI and add a new non-domain policy (like a policy for a local IP address).
3. Do Save & Apply.
4. Connect via SSH and post the output of ```/etc/init.d/vpn-policy-routing status```.

n8v8R, 2018-07-04:
luci-app-vpn-policy-routing git-18.181.76096-1fd3…-24
vpn-policy-routing 0.0.2-6
TOS 3.10.3

the issue is not apparent anymore

closed 2018-07-04 by n8v8R