mossdef-org/source.mossdef.org #14: [feature request] support for nft

n8v8R, 2018-06-20:
I am intending to switch from `ipt` to `nft` but that would render this package not working as currently being based on `ipt `and `ipset`. 

One of the many benefits of `nft` is the independency of `ipset` since it features its own `sets`

stangri, 2018-07-20:
AFAIK, the OpenWrt's fw3 is dependent on `iptables`. I don't think it's possible to "switch to nftables" for any current OpenWrt distribution. 

If at any point in the future, fw3 changes dependency to nftables, I'll work on converting current iptables rules to fw3-compatible rules to avoid dependency on iptables for VPR. 

However, from my experience with `fakeinternet` package, the 15.05-based builds require special handling for the PROCD firewall rules comparing to 17.01/18.06, so I wouldn't want to do it now as it would be twice the work.

Feel free to re-open this if my assumption about fw3 and nftables incompatibility is incorrect.

n8v8R, 2018-07-20:
Sure, understand and appreciate the toughts.

 TOS 4 is supposed to be based on OpenWRT 18.06, perhaps around end of year.
fw3 will probably stay with ipt indefinitely though.

But that does not mean a user has pass up nft. I am using it already on a couple of other boxes and came to like it, particularly its own sets as opposed to ipsets and its performance over ipt.

Likely I am going to compile nft on my own then and kick out fw3 (which I am not really fond of).
And then apparently will need to put something like VPR together based on nft, likely without LuCI though.

stangri, 2018-07-20:
Well, if you look at the VPR code, all the `iptables` creations/deletions are separated into the `ipt` function. Also, the creation of parameters is happening in the `insert_policy` function, so if you know what you're doing and have a way of testing it, it shouldn't be hard to adapt the code to `nftables`.

There're some calls to `ipt` from `table_create` and `start_process`. 

If you know the `nft` syntax, shouldn't take you more than an hour to convert everything.

If/when you do that, send me the updated code and I can try to incorporate it my with possibly a setting for which filter to use or a dynamic detection based on the presence of `nftables` binary.

n8v8R, 2019-12-26:
https://www.netfilter.org/projects/nftables/manpage.html

[image]

StaRipper, 2022-08-10:
OpenWRT in SNAPSHOT already fully in nft and showing a message : “Legacy rules detected
There are legacy iptables rules present on the system. Mixing iptables and nftables rules is discouraged and may lead to incomplete traffic filtering.”

stangri, 2022-08-10:
Unfortuntely `dnsmasq-full` in OpenWrt still supports ipsets only, not nft sets and moving to nftables now would mean losing a widely used feature. Transition to nft is in the works but can't be completed until OpenWrt's `dnsmasq-full` supports nft sets.

jonozzz, 2022-08-19:
Is this relevant ? https://github.com/imp/dnsmasq/commit/47aefca5e405b4b6627ef952fdc42e61b1baa770

stangri, 2022-08-19:
> Is this relevant ? [imp/dnsmasq@47aefca](https://github.com/imp/dnsmasq/commit/47aefca5e405b4b6627ef952fdc42e61b1baa770)

AFAIK the upstream dnsmasq had nftsets support implemented for a while, I'm talking about OpenWrt package. I've asked multiple times every time the transition to nft sets was mentioned and I was never given an answer on the dnsmasq-full future from the core team.

jonozzz, 2022-08-19:
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git&a=search&h=HEAD&st=commit&s=nftset
upstream dnsmasq introduced this about 10 months ago in 2.87test2. The latest stable version is 2.86. So even though the support is in the upstream, is not included the latest version. Once Simon is gonna release 2.87 I'm sure openwrt folks won't have an issue bumping the version.

stangri, 2022-09-29:
nft support has been implemented in `pbr` package, I intend to send a PR to have it in the official snapshots and 22.03 repo, if there are any specific nft-related bugs, please open separate issues.