mossdef-org/source.mossdef.org #16: Remove the dependency on package `luci`

LGA1150, 2018-07-06:
This commit https://github.com/openwrt/luci/commit/90bf08af829a8e13c3caf6a389bff5212d5204dc introduced a new package named `luci-nginx`, which uses nginx as its web server, while the old `luci` package uses uHTTPd.
With the dependency on `luci`, I can't deselect `uhttpd` despite no need for it.

stangri, 2018-07-07:
Which package(s) are you including in your build? I'd like you to test a change on a single package before I update all of them.

LGA1150, 2018-07-07:
w/ `luci-ssl-nginx`
w/o `luci` and `uhttpd`

stangri, 2018-07-07:
Ok, let's try it another way, which one of **my** packages are you trying to include in your build?

LGA1150, 2018-07-07:
Any one of them, such as `luci-app-advanced-reboot`

LGA1150, 2018-07-07:
Compiled with latest master
[image]
Seems to be working, and `uhttpd` isn't included now, thx!

stangri, 2018-07-07:
Thanks for the report, I'll fix other packages over the weekend!

closed 2018-07-07 by stangri