mossdef-org/source.mossdef.org #17: [bug] LuCI VPR status stays disabled/stopped

n8v8R, 2018-07-13:
- luci-app-vpn-policy-routing | git-18.192.22043-6f77..-27
- vpn-policy-routing | 0.0.2-11
- TOS 3.10.3 (OpenWrt omnia 15.05 r47055 / LuCI 526a8767846acbe57c521912b35feb4d97354db6 branch (git-18.145.30016-526a876)
- kernel 4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1
---

Service status remains static @ `Service is disabled/stopped` despite running and subsequent the button constantly showing `Enable/Start`

stangri, 2018-07-13:
Can you see if ```ubus call service status``` yields anything on Turris.

stangri, 2018-07-13:
Actually, I think it's been established that Turris Omnia doesn't have PROCD support, so scratch that.

Please upgrade to ```luci-app-vpn-policy-routing_git-18.193.61091-777ffa9-28_all.ipk```.

n8v8R, 2018-07-13:
The update fixed it, thank you.

Would reckon that TO is not going fix PROCD support for their current flavour but rather drag it till the release of TOS 4 (supposedly end of year), which then will be basically true OpenWRT as foundation with TO tweaks on top.

closed 2018-07-13 by n8v8R