mossdef-org/source.mossdef.org #18: vpn-policy-routing: failure to reload on systems without PROCD

stangri, 2018-07-13:
@n8v8R -- can you please test something for me,
can you please place the following into file ```/etc/hotplug.d/iface/70-vpn-policy-routing```:
```sh
#!/bin/sh

if [[ "$ACTION" == "ifup" ]]; then
  sleep 10 && /etc/init.d/vpn-policy-routing reload
fi
```

Might need to also do ```chmod +x /etc/hotplug.d/iface/70-vpn-policy-routing```.

Then manually stop and restart openvpn and see if you see any traces of VPR reloaded in the system log (wherever it is on TurrisOS).

n8v8R, 2018-07-13:
Thank for the effort to cater for those few TO users, big of you!

---

Just to be sure - manually stop/restart I did from LuCI, or did you mean from cli?

Anyway, from LuCI pressing the `stop/disable` button produced in the router's log

> notice [1083]: service stopped [✓]

And then on pressing `start/enable`

> err modprobe[]: xt_set is already loaded
> err modprobe[]: ip_set is already loaded
> err modprobe[]: ip_set_hash_ip is already loaded
> notice [2840]: Creating table 'wan/84.46.104.107' [✓]
> notice [2840]: Creating table 'tun/0.0.0.0' [✗]
> notice [2840]: Creating table 'wg0/192.168.112.12' [✓]
> notice [2840]: Routing 's5' via wan [✓]
> notice [2840]: Routing 's1' via wan [✓]
> notice [2840]: Routing 'da' via wan [✓]
> notice [2840]: Routing '1750e' via wan [✓]
> notice [2840]: Routing 'dvb-c' via wan [✓]
> notice [2840]: Routing 'stv9s' via wan [✓]
> notice [2840]: Routing 'oc_vpscp' via wan [✓]
> notice [2840]: Routing 'oc_vnc' via wan [✓]
> notice [2840]: Routing 'oc_ssh' via wan [✓]
> notice [2840]: Routing 'lan' via wan [✓]
> notice [2840]: service started on wan/84.46.104.107 wg0/192.168.112.12 with errors [✗]
> notice [2840]: ERROR: Failed to set up 'tun/0.0.0.0'
> notice [2840]: service monitoring interfaces: wan tun wg0 [✓]

---

Looks like nothing about the status of ifaces. From the router's log it appears that iface are logged with `netifd`. Perhaps different from OpenWRT?

> notice netifd[]: radio0 (2072): wlan0: ACS-COMPLETED freq=5745 channel=149
> notice netifd[]: radio0 (2072): wlan0: interface state ACS->HT_SCAN
> notice netifd[]: radio0 (2072): Using interface wlan0 with hwaddr 04:f0:21:31:d0:ba and ssid "km"
> info kernel[]: [   45.096037] br-lan: port 3(wlan0) entered forwarding state
> info kernel[]: [   45.096068] br-lan: port 3(wlan0) entered forwarding state
> info kernel[]: [   45.098610] device guest0 entered promiscuous mode
> info kernel[]: [   45.100788] br-guest: port 1(guest0) entered forwarding state
> info kernel[]: [   45.100816] br-guest: port 1(guest0) entered forwarding state
> notice netifd[]: radio0 (2072): Using interface guest0 with hwaddr 06:f0:21:31:d0:ba and ssid "b_me_g8st"
> netifd[]: bridge 'br-guest' link is up
> netifd[]: Interface 'guest' has link connectivity 
> netifd[]: Network device 'guest0' link is up
> netifd[]: radio0 (2072): wlan0: interface state HT_SCAN->ENABLED
> netifd[]: radio0 (2072): wlan0: AP-ENABLED 
> netifd[]: Network device 'wlan0' link is up

---

The router is logging to this file `/tmp/log/messages`

stangri, 2018-07-13:
So if you manually stop and start (or just restart) OpenVPN (not VPR) either from cli or from WebUI -- does the VPR get reloaded afterwards? Not immediately, as OpenVPN may take some time to settle down.

n8v8R, 2018-07-13:
Well, reading a sometimes helps... you asked for manually starting/stopping... sorry about that

Don't have a OpenVPN at the moment, just WG, and hope it does not matter for debugging.

`ifdwn wg0` produced in the log

> notice netifd[]: Network device 'wg0' link is down
> notice netifd[]: Interface 'wg0' is now down

and `ifup wg0`

> notice netifd[]: Interface 'wg0' is setting up now
> notice netifd[]: Interface 'wg0' is now up
> notice netifd[]: Network device 'wg0' link is up
> notice firewall[]: Reloading firewall due to ifup of wg0 (wg0)
> info dnsmasq[12864]: reading /etc/resolv.conf
> info dnsmasq[12864]: using local addresses only for domain km.home
> info dnsmasq[12864]: using nameserver 127.0.0.1#53
> info dnsmasq[12864]: reading /etc/resolv.conf
> info dnsmasq[12864]: using local addresses only for domain km.home
> info dnsmasq[12864]: using nameserver 127.0.0.1#53

---
Nothing about VPR.  But since the fw is apparently aware of changes in the iface status here is the relevant script

```
#!/bin/sh

[ "$ACTION" = ifup -o "$ACTION" = ifupdate ] || exit 0
[ "$ACTION" = ifupdate -a -z "$IFUPDATE_ADDRESSES" -a -z "$IFUPDATE_DATA" ] && exit 0

/etc/init.d/firewall enabled || exit 0

fw3 -q network "$INTERFACE" >/dev/null || exit 0

logger -t firewall "Reloading firewall due to $ACTION of $INTERFACE ($DEVICE)"
fw3 -q reload
```
---
Suppose the TO developer could have shared that from the beginning...

n8v8R, 2018-07-13:
And there is also this netstat script forUCI, maybe are more global OpenWRT solution

[ ifup = "$ACTION" ] && {
	uci_toggle_state network "$INTERFACE" up 1
	[ -n "$DEVICE" ] && {
		uci_toggle_state network "$INTERFACE" device "$(uci -q get network.$INTERFACE.ifname)"
		uci_toggle_state network "$INTERFACE" ifname "$DEVICE"
	}
}

stangri, 2018-07-13:
Well, last attempt, please try to change the script to:
```sh
#!/bin/sh

if [ "$ACTION" = "ifup" -o "$ACTION" = "ifupdate" ]; then
  logger -t vpn-policy-routing "Reloading vpn-policy-routing due to $ACTION of $INTERFACE ($DEVICE)"
  /etc/init.d/vpn-policy-routing reload
fi
```

If that doesn't help right away (ie: no record in the log for reloading VPR) maybe try to reboot the router and then try again.

n8v8R, 2018-07-13:
There you go, without reboot and no `chmod +x`

`ifup wg0`
> notice netifd[]: Interface 'wg0' is setting up now
> notice netifd[]: Interface 'wg0' is now up
> notice netifd[]: Network device 'wg0' link is up
> notice firewall[]: Reloading firewall due to ifup of wg0 (wg0)
> info dnsmasq[3836]: reading /etc/resolv.conf
> info dnsmasq[3836]: using local addresses only for domain km.home
> info dnsmasq[3836]: using nameserver 127.0.0.1#53
> info dnsmasq[3836]: reading /etc/resolv.conf
> info dnsmasq[3836]: using local addresses only for domain km.home
> info dnsmasq[3836]: using nameserver 127.0.0.1#53
> notice vpn-policy-routing[]: Reloading vpn-policy-routing due to ifup of wg0 (wg0)
> err modprobe[]: xt_set is already loaded
> err modprobe[]: ip_set is already loaded
> err modprobe[]: ip_set_hash_ip is already loaded
> notice [6624]: Creating table 'wan/84.46.104.107' [✓]
> notice [6624]: Creating table 'wg0/192.168.112.12' [✓]
> notice [6624]: Routing 's5' via wan [✓]
> notice [6624]: Routing 's1' via wan [✓]
> notice [6624]: Routing 'da' via wan [✓]
> notice [6624]: Routing '1750e' via wan [✓]
> notice [6624]: Routing 'dvb-c' via wan [✓]
> notice [6624]: Routing 'stv9s' via wan [✓]
> notice [6624]: Routing 'oc_vpscp' via wan [✓]
> notice [6624]: Routing 'oc_vnc' via wan [✓]
> notice [6624]: Routing 'oc_ssh' via wan [✓]
> notice [6624]: Routing 'lan' via wan [✓]
> notice [6624]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> notice [6624]: service monitoring interfaces: wan wg0 [✓]

---
`ls -la /etc/hotplug.d/iface/`

> drwxr-xr-x    1 root     root           212 Jul 13 20:52 .
> drwxr-xr-x    1 root     root            54 May  4 09:58 ..
> -rw-r--r--    1 root     root           244 Mar 26 07:37 00-netstate
> -rwxr-xr-x    1 root     root            98 May  4 10:33 11-sqm
> -rwxr-xr-x    1 root     root           486 May  4 10:03 15-teql
> -rw-r--r--    1 root     root           336 May  4 10:32 20-firewall
> -rwxr-xr-x    1 root     root           205 May  4 09:58 20-ntpd
> -rw-r--r--    1 root     root           100 May  4 10:33 25-dnsmasq
> -rw-r--r--    1 root     root            65 May  4 09:34 30-relay
> -rwxr-xr-x    1 root     root           322 May  4 08:57 40-resolver-reload
> -rw-r--r--    1 root     root           213 Jul 13 20:53 70-vpn-policy-routing
> -rwxr-xr-x    1 root     root           204 Jun  6 16:30 95-ddns

stangri, 2018-07-13:
Woohoo, amazing progress man!

A few more things I need output of to only use the hotplug script on Turris:
```which ubus```
```ubus call service list```

Thanks!

n8v8R, 2018-07-13:
`which ubus`

> /bin/ubus

---
`ubus call service list`

> {
>         "atd": {
>                 "instances": {
>                         "instance1": {
>                                 "running": true,
>                                 "pid": 1876,
>                                 "command": [
>                                         "\/usr\/sbin\/atd",
>                                         "-f"
>                                 ],
>                                 "respawn": {
>                                         "threshold": 3600,
>                                         "timeout": 5,
>                                         "retry": 5
>                                 }
>                         }
>                 }
>         },
>         "cron": {
>                 "instances": {
>                         "instance1": {
>                                 "running": true,
>                                 "pid": 2623,
>                                 "command": [
>                                         "\/usr\/sbin\/cron",
>                                         "-n"
>                                 ],
>                                 "respawn": {
>                                         "threshold": 3600,
>                                         "timeout": 5,
>                                         "retry": 5
>                                 }
>                         }
>                 }
>         },
>         "dnsmasq": {
>                 "instances": {
>                         "instance1": {
>                                 "running": true,
>                                 "pid": 3836,
>                                 "command": [
>                                         "\/usr\/sbin\/dnsmasq",
>                                         "-C",
>                                         "\/var\/etc\/dnsmasq.conf",
>                                         "-k",
>                                         "-x",
>                                         "\/var\/run\/dnsmasq\/dnsmasq.pid"
>                                 ],
>                                 "respawn": {
>                                         "threshold": 3600,
>                                         "timeout": 5,
>                                         "retry": 5
>                                 }
>                         }
>                 }
>         },
>         "firewall": {
> 
>         },
>         "foris-controller": {
>                 "instances": {
>                         "instance1": {
>                                 "running": true,
[... 264 lines cut ...]
>                                         "-d"
>                                 ],
>                                 "respawn": {
>                                         "threshold": 3600,
>                                         "timeout": 5,
>                                         "retry": 5
>                                 }
>                         }
>                 }
>         },
>         "usbmode": {
>                 "instances": {
>                         "instance1": {
>                                 "running": false,
>                                 "command": [
>                                         "\/sbin\/usbmode",
>                                         "-s"
>                                 ]
>                         }
>                 }
>         },
>         "vpn-policy-routing": {
> 
>         }
> }

stangri, 2018-07-13:
```ubus -S call system board```
```cat /etc/turris-version```
please?

n8v8R, 2018-07-13:
`ubus -S call system board`

> ubus -S call system board
> {"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}

---
`cat /etc/turris-version`

> 3.10.3

n8v8R, 2018-07-13:
There is something about parsing message data - see below

> INFO:Queue install of vpn-policy-routing/stangri_repo/0.0.2-12
> INFO:Queue install of luci-app-vpn-policy-routing/stangri_repo/git-18.194.53747-b38306d-28
> Press return to continue, CTRL+C to abort
> 
> INFO:Executing preupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_preupdate/05_schnapps.sh:
> Snapshot number 7 created
> 
> INFO:End of subprogram output
> INFO:Unpacking download packages
> INFO:Checking for file collisions between packages
> INFO:Running pre-install scripts and merging packages to root file system
> WARN:Config file /etc/config/vpn-policy-routing modified by the user. Backing up the new one into /etc/config/vpn-policy-routing-opkg
> INFO:Running post-install and post-rm scripts
> Output from vpn-policy-routing.postinst:
> **Failed to parse message data**
> INFO:Removing packages and leftover files
> INFO:Cleaning up control files
> INFO:Executing postupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_postupdate/05_schnapps.sh:
> Snapshot number 8 created
> 
> INFO:End of subprogram output
> INFO:Executed: /etc/updater/hook_postupdate/99_approvals_cleanup

stangri, 2018-07-13:
Please try vpn-policy-routing 0.0.2-14.

Also, the code I referred to was:
```sh
source /etc/init.d/vpn-policy-routing
ubus_status set 'Testing'
ubus_status get
```

Thanks!

n8v8R, 2018-07-13:
Oh, I c now - browser did not refresh after your edit

the parsing error is still present though

> INFO:Queue install of vpn-policy-routing/stangri_repo/0.0.2-14
> Press return to continue, CTRL+C to abort
> 
> INFO:Executing preupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_preupdate/05_schnapps.sh:
> Snapshot number 9 created
> 
> INFO:End of subprogram output
> INFO:Unpacking download packages
> INFO:Checking for file collisions between packages
> INFO:Running pre-install scripts and merging packages to root file system
> WARN:Config file /etc/config/vpn-policy-routing modified by the user. Backing up the new one into /etc/config/vpn-policy-routing-opkg
> INFO:Running post-install and post-rm scripts
> Output from vpn-policy-routing.postinst:
> **Failed to parse message data**
> INFO:Removing packages and leftover files
> INFO:Cleaning up control files
> INFO:Executing postupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_postupdate/05_schnapps.sh:
> Snapshot number 10 created
> 
> INFO:End of subprogram output
> INFO:Executed: /etc/updater/hook_postupdate/99_approvals_cleanup

---
`source /etc/init.d/vpn-policy-routing`
**-ash: readonly: line 8: _OK_: is read only**

---
`ubus_status set 'Testing'`
_no output_

---
`ubus_status get`

> Testing

stangri, 2018-07-13:
Please upgrade both packages to:
```
[build]   vpn-policy-routing 0.0.2-15...                    [✓]
[build]   luci-app-vpn-policy-routing 29...                 [✓]
```

n8v8R, 2018-07-13:
it is not showing yet as available

stangri, 2018-07-13:
Can you see if you can kill the package manager cache first (maybe even reboot) and then try to install again?

n8v8R, 2018-07-13:
Removed both packages and then rebooted and then from LuCI installed vpn-policy-routing (0.0.2-15) and I am dismayed having to report

> Installing vpn-policy-routing (0.0.2-15) to root...
> Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/vpn-policy-routing_0.0.2-15_all.ipk
> Configuring vpn-policy-routing.
> 
>  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> 100  9790  100  9790    0     0   8240      0  0:00:01  0:00:01 --:--:--  9183
> **Failed to parse message data**
> Collected errors:
>  * resolve_conffiles: Existing conffile /etc/config/vpn-policy-routing is different from the conffile in the new package. The new conffile will be placed at /etc/config/vpn-policy-routing-opkg.

stangri, 2018-07-13:
But did it install or not?

n8v8R, 2018-07-13:
Sure it (always) did.

n8v8R, 2018-07-13:
Just changed some setting via LuCI and then `save & apply` and observed that the same process is repeated several times within a few seconds?

> 58:10 err modprobe[]: xt_set is already loaded
> 58:10 err modprobe[]: ip_set is already loaded
> 58:10 err modprobe[]: ip_set_hash_ip is already loaded
> 58:10 notice [15396]: Creating table 'wan/84.46.104.107' [✓]
> 58:10 notice [15396]: Creating table 'wg0/192.168.112.12' [✓]
> 58:10 notice [15396]: Routing 's5' via wan [✓]
> 58:10 notice [15396]: Routing 's1' via wan [✓]
> 58:10 notice [15396]: Routing 'da' via wan [✓]
> 58:10 notice [15396]: Routing '1750e' via wan [✓]
> 58:10 notice [15396]: Routing 'dvb-c' via wan [✓]
> 58:10 notice [15396]: Routing 'stv9s' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_vpscp' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_vnc' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_ssh' via wan [✓]
> 58:10 notice [15396]: Routing 'lan' via wan [✓]
> 58:10 notice [15396]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:11 notice [15396]: service monitoring interfaces: wan wg0 [✓]
> 58:11 err modprobe[]: xt_set is already loaded
> 58:11 err modprobe[]: ip_set is already loaded
> 58:11 err modprobe[]: ip_set_hash_ip is already loaded
> 58:11 notice [16076]: Creating table 'wan/84.46.104.107' [✓]
> 58:11 notice [16076]: Creating table 'wg0/192.168.112.12' [✓]
> 58:11 notice [16076]: Routing 's5' via wan [✓]
> 58:11 notice [16076]: Routing 's1' via wan [✓]
> 58:11 notice [16076]: Routing 'da' via wan [✓]
> 58:11 notice [16076]: Routing '1750e' via wan [✓]
> 58:11 notice [16076]: Routing 'dvb-c' via wan [✓]
> 58:12 notice [16076]: Routing 'stv9s' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_vpscp' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_vnc' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_ssh' via wan [✓]
> 58:12 notice [16076]: Routing 'lan' via wan [✓]
> 58:12 notice [16076]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:12 notice [16076]: service monitoring interfaces: wan wg0 [✓]
> 58:12 err modprobe[]: xt_set is already loaded
> 58:12 err modprobe[]: ip_set is already loaded
> 58:12 err modprobe[]: ip_set_hash_ip is already loaded
> 58:12 notice [16771]: Creating table 'wan/84.46.104.107' [✓]
> 58:13 notice [16771]: Creating table 'wg0/192.168.112.12' [✓]
> 58:13 notice [16771]: Routing 's5' via wan [✓]
> 58:13 notice [16771]: Routing 's1' via wan [✓]
> 58:13 notice [16771]: Routing 'da' via wan [✓]
> 58:13 notice [16771]: Routing '1750e' via wan [✓]
> 58:13 notice [16771]: Routing 'dvb-c' via wan [✓]
> 58:13 notice [16771]: Routing 'stv9s' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_vpscp' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_vnc' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_ssh' via wan [✓]
> 58:13 notice [16771]: Routing 'lan' via wan [✓]
> 58:13 notice [16771]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:13 notice [16771]: service monitoring interfaces: wan wg0 [✓]
> 58:13 err modprobe[]: xt_set is already loaded
> 58:13 err modprobe[]: ip_set is already loaded
> 58:13 err modprobe[]: ip_set_hash_ip is already loaded
> 58:14 notice [17451]: Creating table 'wan/84.46.104.107' [✓]
> 58:14 notice [17451]: Creating table 'wg0/192.168.112.12' [✓]
> 58:14 notice [17451]: Routing 's5' via wan [✓]
> 58:14 notice [17451]: Routing 's1' via wan [✓]
> 58:14 notice [17451]: Routing 'da' via wan [✓]
> 58:14 notice [17451]: Routing '1750e' via wan [✓]
> 58:14 notice [17451]: Routing 'dvb-c' via wan [✓]
> 58:14 notice [17451]: Routing 'stv9s' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_vpscp' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_vnc' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_ssh' via wan [✓]
> 58:14 notice [17451]: Routing 'lan' via wan [✓]
> 58:14 notice [17451]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:14 notice [17451]: service monitoring interfaces: wan wg0 [✓]
[... 60 lines cut ...]
> 58:19 notice [20230]: Routing 'dvb-c' via wan [✓]
> 58:19 notice [20230]: Routing 'stv9s' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_vpscp' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_vnc' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_ssh' via wan [✓]
> 58:19 notice [20230]: Routing 'lan' via wan [✓]
> 58:19 notice [20230]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:20 notice [20230]: service monitoring interfaces: wan wg0 [✓]
> 58:20 err modprobe[]: xt_set is already loaded
> 58:20 err modprobe[]: ip_set is already loaded
> 58:20 err modprobe[]: ip_set_hash_ip is already loaded
> 58:20 notice [20928]: Creating table 'wan/84.46.104.107' [✓]
> 58:20 notice [20928]: Creating table 'wg0/192.168.112.12' [✓]
> 58:20 notice [20928]: Routing 's5' via wan [✓]
> 58:20 notice [20928]: Routing 's1' via wan [✓]
> 58:20 notice [20928]: Routing 'da' via wan [✓]
> 58:20 notice [20928]: Routing '1750e' via wan [✓]
> 58:20 notice [20928]: Routing 'dvb-c' via wan [✓]
> 58:20 notice [20928]: Routing 'stv9s' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_vpscp' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_vnc' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_ssh' via wan [✓]
> 58:21 notice [20928]: Routing 'lan' via wan [✓]
> 58:21 notice [20928]: service started on wan/84.46.104.107 wg0/192.168.112.12 [✓]
> 58:21 notice [20928]: service monitoring interfaces: wan wg0 [✓]

n8v8R, 2018-07-13:
Just changed some setting via LuCI and then `save & apply` and observed that the same process is repeated several times within a few seconds?

> 58:10 err modprobe[]: xt_set is already loaded
> 58:10 err modprobe[]: ip_set is already loaded
> 58:10 err modprobe[]: ip_set_hash_ip is already loaded
> 58:10 notice [15396]: Creating table 'wan/84.[truncated]' [✓]
> 58:10 notice [15396]: Creating table 'wg0/192.168.112.12' [✓]
> 58:10 notice [15396]: Routing 's5' via wan [✓]
> 58:10 notice [15396]: Routing 's1' via wan [✓]
> 58:10 notice [15396]: Routing 'da' via wan [✓]
> 58:10 notice [15396]: Routing '1750e' via wan [✓]
> 58:10 notice [15396]: Routing 'dvb-c' via wan [✓]
> 58:10 notice [15396]: Routing 'stv9s' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_vpscp' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_vnc' via wan [✓]
> 58:10 notice [15396]: Routing 'oc_ssh' via wan [✓]
> 58:10 notice [15396]: Routing 'lan' via wan [✓]
> 58:10 notice [15396]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:11 notice [15396]: service monitoring interfaces: wan wg0 [✓]
> 58:11 err modprobe[]: xt_set is already loaded
> 58:11 err modprobe[]: ip_set is already loaded
> 58:11 err modprobe[]: ip_set_hash_ip is already loaded
> 58:11 notice [16076]: Creating table 'wan/84.[truncated]' [✓]
> 58:11 notice [16076]: Creating table 'wg0/192.168.112.12' [✓]
> 58:11 notice [16076]: Routing 's5' via wan [✓]
> 58:11 notice [16076]: Routing 's1' via wan [✓]
> 58:11 notice [16076]: Routing 'da' via wan [✓]
> 58:11 notice [16076]: Routing '1750e' via wan [✓]
> 58:11 notice [16076]: Routing 'dvb-c' via wan [✓]
> 58:12 notice [16076]: Routing 'stv9s' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_vpscp' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_vnc' via wan [✓]
> 58:12 notice [16076]: Routing 'oc_ssh' via wan [✓]
> 58:12 notice [16076]: Routing 'lan' via wan [✓]
> 58:12 notice [16076]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:12 notice [16076]: service monitoring interfaces: wan wg0 [✓]
> 58:12 err modprobe[]: xt_set is already loaded
> 58:12 err modprobe[]: ip_set is already loaded
> 58:12 err modprobe[]: ip_set_hash_ip is already loaded
> 58:12 notice [16771]: Creating table 'wan/84.[truncated]' [✓]
> 58:13 notice [16771]: Creating table 'wg0/192.168.112.12' [✓]
> 58:13 notice [16771]: Routing 's5' via wan [✓]
> 58:13 notice [16771]: Routing 's1' via wan [✓]
> 58:13 notice [16771]: Routing 'da' via wan [✓]
> 58:13 notice [16771]: Routing '1750e' via wan [✓]
> 58:13 notice [16771]: Routing 'dvb-c' via wan [✓]
> 58:13 notice [16771]: Routing 'stv9s' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_vpscp' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_vnc' via wan [✓]
> 58:13 notice [16771]: Routing 'oc_ssh' via wan [✓]
> 58:13 notice [16771]: Routing 'lan' via wan [✓]
> 58:13 notice [16771]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:13 notice [16771]: service monitoring interfaces: wan wg0 [✓]
> 58:13 err modprobe[]: xt_set is already loaded
> 58:13 err modprobe[]: ip_set is already loaded
> 58:13 err modprobe[]: ip_set_hash_ip is already loaded
> 58:14 notice [17451]: Creating table 'wan/84.[truncated]' [✓]
> 58:14 notice [17451]: Creating table 'wg0/192.168.112.12' [✓]
> 58:14 notice [17451]: Routing 's5' via wan [✓]
> 58:14 notice [17451]: Routing 's1' via wan [✓]
> 58:14 notice [17451]: Routing 'da' via wan [✓]
> 58:14 notice [17451]: Routing '1750e' via wan [✓]
> 58:14 notice [17451]: Routing 'dvb-c' via wan [✓]
> 58:14 notice [17451]: Routing 'stv9s' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_vpscp' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_vnc' via wan [✓]
> 58:14 notice [17451]: Routing 'oc_ssh' via wan [✓]
> 58:14 notice [17451]: Routing 'lan' via wan [✓]
> 58:14 notice [17451]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:14 notice [17451]: service monitoring interfaces: wan wg0 [✓]
[... 60 lines cut ...]
> 58:19 notice [20230]: Routing 'dvb-c' via wan [✓]
> 58:19 notice [20230]: Routing 'stv9s' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_vpscp' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_vnc' via wan [✓]
> 58:19 notice [20230]: Routing 'oc_ssh' via wan [✓]
> 58:19 notice [20230]: Routing 'lan' via wan [✓]
> 58:19 notice [20230]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:20 notice [20230]: service monitoring interfaces: wan wg0 [✓]
> 58:20 err modprobe[]: xt_set is already loaded
> 58:20 err modprobe[]: ip_set is already loaded
> 58:20 err modprobe[]: ip_set_hash_ip is already loaded
> 58:20 notice [20928]: Creating table 'wan/84.[truncated]' [✓]
> 58:20 notice [20928]: Creating table 'wg0/192.168.112.12' [✓]
> 58:20 notice [20928]: Routing 's5' via wan [✓]
> 58:20 notice [20928]: Routing 's1' via wan [✓]
> 58:20 notice [20928]: Routing 'da' via wan [✓]
> 58:20 notice [20928]: Routing '1750e' via wan [✓]
> 58:20 notice [20928]: Routing 'dvb-c' via wan [✓]
> 58:20 notice [20928]: Routing 'stv9s' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_vpscp' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_vnc' via wan [✓]
> 58:21 notice [20928]: Routing 'oc_ssh' via wan [✓]
> 58:21 notice [20928]: Routing 'lan' via wan [✓]
> 58:21 notice [20928]: service started on wan/84.[truncated] wg0/192.168.112.12 [✓]
> 58:21 notice [20928]: service monitoring interfaces: wan wg0 [✓]

stangri, 2018-07-13:
Oh, didn't realize it still always installed despite the parser error. I'll switch back to the version where extra hotplug script was deleted in the post-install.

Can you check if WebUI now properly shows the status?

n8v8R, 2018-07-13:
No parsing error with 0.0.2-16

> INFO:Queue install of vpn-policy-routing/stangri_repo/0.0.2-16
> Press return to continue, CTRL+C to abort
> 
> INFO:Executing preupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_preupdate/05_schnapps.sh:
> Snapshot number 13 created
> 
> INFO:End of subprogram output
> INFO:Unpacking download packages
> INFO:Checking for file collisions between packages
> INFO:Running pre-install scripts and merging packages to root file system
> WARN:Config file /etc/config/vpn-policy-routing modified by the user. Backing up the new one into /etc/config/vpn-policy-routing-opkg
> INFO:Running post-install and post-rm scripts
> INFO:Removing packages and leftover files
> INFO:Cleaning up control files
> INFO:Executing postupdate hooks...
> INFO:Subprogram output: /etc/updater/hook_postupdate/05_schnapps.sh:
> Snapshot number 14 created
> 
> INFO:End of subprogram output
> INFO:Executed: /etc/updater/hook_postupdate/99_approvals_cleanup

n8v8R, 2018-07-13:
One more oversation - when VPR been _disabled from LuCI_ then `/etc/init.d/vpn-policy-routing start` from the cli does nothing and `/etc/init.d/vpn-policy-routing restart` produces

> vpn-policy-routing 0.0.2-16 stopped [✓]

stangri, 2018-07-14:
I have no idea why is it happening, if you feel like opening another bug, I'll try to investigate.

closed 2018-07-14 by stangri