mossdef-org/source.mossdef.org #20: [bug or design?] VPR cannot be started from cli when previously disabled in LuCI

n8v8R, 2018-07-14:
luci-app-vpn-policy-routing | git-18.194.53747-b383..-29
Remove | vpn-policy-routing | 0.0.2-17

**TOS 3.10.3** ({"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}})

---
Not sure whether it is a bug or by design, however having VPR disabled via LuCI and afterwards trying to start it from cli with `/etc/init.d/vpn-policy-routing start` produces

> vpn-policy-routing is currently disabled.
> Run the following commands before starting service again:
> uci set vpn-policy-routing.config.enabled='1'; uci commit;

Same happens when VPR is disabled via LuCI and in that state changing VPR settings via LuCI and `pressing save & apply` in LuCI

stangri, 2018-07-14:
By design. The button labels in WebUI are: Enable/Start and Stop/Disable. If you think the labels can be improved to better describe their function, please advise.

The reasoning behind is that for the WebUI users there's just one button to enable and start the service and also just one button to stop and disable. If they keep using WebUI that should cover most of their needs. There's also more fine-tuned control in the System/Startup page. 

The cli users should figure things out with the on-start message if VPR is disabled in case of the mixed (WebUI and cli) usage scenario.

n8v8R, 2018-07-14:
It is fine, appreciate the clarification

closed 2018-07-14 by n8v8R