mossdef-org/source.mossdef.org #21: [bug] reported wan ip remains the same despite been rolledover by ISP

n8v8R, 2018-07-14:
luci-app-vpn-policy-routing | git-18.194.53747-b383..-29
Remove | vpn-policy-routing | 0.0.2-17

TOS 3.10.3 ({"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}})

---

ISP is rolling over WAN ip every 24 hours and also issuing a new ip on router reboot, however noticed that VPR is not picking up those new ips and instead is stuck with the same old one.

This becomes only apparent from the logs

> Creating table 'wan/**[same old ip]**
> service started on wan/**[same old ip]**

Since VPR is creating routing tables this may have an inclement impact on its functionality

stangri, 2018-07-14:
Can you please capture more of the logs with the timestamps included -- starting from the point where WAN IP changes and to the point where VPR is finished reloading.
Does VPR get reloaded at all after WAN IP change thru the hotplug script?

n8v8R, 2018-07-14:
Here you go, that is from a reboot. VPR appears to get reloaded on `ifup` of wan (pppoe-wan). Will have to wait for an ip rollover by the ISP but if am not mistaken it involves `ifdown` and subsequent `ifup`, yet not 100% sure.

> :32:44 info kernel[]: [   13.655039] ip_set: protocol 6
> :32:44 info kernel[]: [   15.398164] Ebtables v2.0 registered
> :32:44 info kernel[]: [   15.403252] ip_tables: (C) 2000-2006 Netfilter Core Team
> :32:44 info kernel[]: [   15.415690] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
> :32:44 info kernel[]: [   15.483625] ctnetlink v0.93: registering with nfnetlink.
> :32:44 info kernel[]: [   15.585571] PPP MPPE Compression module registered
> :32:44 info kernel[]: [   15.590935] NET: Registered protocol family 24
> :32:44 info kernel[]: [   15.695321] wireguard: WireGuard 0.0.20171017 loaded. See www.wireguard.com for information.
> :32:44 info kernel[]: [   15.703797] wireguard: Copyright (C) 2015-2017 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
> :32:44 info kernel[]: [   22.959315] device eth0 entered promiscuous mode
> :32:44 info kernel[]: [   22.966369] device eth2 entered promiscuous mode
> :32:44 info kernel[]: [   24.950917] mvneta f1030000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
> :32:44 info kernel[]: [   24.960959] mvneta f1070000.ethernet eth2: Link is Up - 1Gbps/Full - flow control off
> :32:44 info kernel[]: [   25.919170] br-lan: port 1(eth0) entered forwarding state
> :32:44 info kernel[]: [   25.919208] br-lan: port 1(eth0) entered forwarding state
> :32:44 info kernel[]: [   25.919253] br-lan: port 2(eth2) entered forwarding state
> :32:44 info kernel[]: [   25.919265] br-lan: port 2(eth2) entered forwarding state
> :32:44 info kernel[]: [   27.910959] br-lan: port 2(eth2) entered forwarding state
> :32:44 info kernel[]: [   27.911014] br-lan: port 1(eth0) entered forwarding state
> :32:44 info kernel[]: [   27.930965] mvneta f1034000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
> :32:44 notice [2759]: service waiting for wan gateway...
> :32:45 notice [2690]: service waiting for wan gateway...
> :32:47 notice [2690]: service waiting for wan gateway...
> :32:48 info pppd[3002]: Plugin rp-pppoe.so loaded.
> :32:48 info pppd[3002]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
> :32:48 notice pppd[3002]: pppd 2.4.7 started by root, uid 0
> :32:48 notice [2759]: service waiting for wan gateway...
> :32:49 info pppd[3002]: PPP session is 54831
> :32:49 warning pppd[3002]: Connected to e0:ac:f1:65:51:b3 via interface eth1
> :32:49 info kernel[]: [   38.429333] pppoe-wan: renamed from ppp0
> :32:49 info pppd[3002]: Using interface pppoe-wan
> :32:49 notice pppd[3002]: Connect: pppoe-wan <--> eth1
> :32:49 notice [2690]: service waiting for wan gateway...
> :32:50 notice [2759]: service waiting for wan gateway...
> :32:51 notice [2690]: service waiting for wan gateway...
> :32:52 notice pppd[3002]: PAP authentication succeeded
> :32:52 notice pppd[3002]: peer from calling number E0:AC:F1:65:51:B3 authorized
> :32:52 notice pppd[3002]: local  IP address 46.x.x.x
> :32:52 notice pppd[3002]: remote IP address 84.x.x.x
> :32:52 notice pppd[3002]: primary   DNS address 213.x.x.x
> :32:52 notice pppd[3002]: secondary DNS address 213.x.x.x
> :32:52 notice netifd[]: Network device 'pppoe-wan' link is up
> :32:52 notice netifd[]: Interface 'wg0' is setting up now
> :32:52 notice netifd[]: Interface 'wan' is now up
> :32:53 notice netifd[]: Interface 'wg0' is now up
> :32:53 notice netifd[]: Network device 'wg0' link is up
> :32:53 err modprobe[]: xt_set is already loaded
> :32:53 err modprobe[]: ip_set is already loaded
> :32:53 err modprobe[]: ip_set_hash_ip is already loaded
> :32:53 notice [2759]: Creating table 'wan/84.x.x.x' [✓]
> :32:53 notice [2759]: Creating table 'wg0/84.x.x.x' [✓]
> :32:53 notice [2759]: Routing 's5' via wan [✓]
> :32:53 notice [2759]: service started on wan/84.x.x.x wg0/84.x.x.x [✓]
> :32:53 notice [2759]: service monitoring interfaces: wan wg0 [✓]
> :32:53 err modprobe[]: xt_set is already loaded
> :32:53 err modprobe[]: ip_set is already loaded
> :32:53 err modprobe[]: ip_set_hash_ip is already loaded
> :32:54 notice [2690]: Creating table 'wan/84.x.x.x' [✓]
> :32:54 notice vpn-policy-routing[]: Reloading vpn-policy-routing due to ifup of lan (br-lan)
> :32:54 notice [2690]: Creating table 'wg0/192.168.112.12' [✓]
> :32:54 notice [2690]: Routing 's5' via wan [✓]
> :32:54 notice [2690]: service started on wan/84.x.x.x wg0/192.168.112.12 [✓]
> :32:54 err modprobe[]: xt_set is already loaded
> :32:54 err modprobe[]: ip_set is already loaded
> :32:54 err modprobe[]: ip_set_hash_ip is already loaded
> :32:54 notice [2690]: service monitoring interfaces: wan wg0 [✓]
> :32:54 notice [3963]: Creating table 'wan/84.x.x.x' [✓]
> :32:54 notice [3963]: Creating table 'wg0/192.168.112.12' [✓]
[... 35 lines cut ...]
> :33:00 err modprobe[]: ip_set_hash_ip is already loaded
> :33:00 notice [6571]: Creating table 'wan/84.x.x.x' [✓]
> :33:00 notice [6571]: Creating table 'wg0/192.168.112.12' [✓]
> :33:00 notice [6571]: Routing 's5' via wan [✓]
> :33:00 notice [6571]: service started on wan/84.x.x.x wg0/192.168.112.12 [✓]
> :33:00 notice [6571]: service monitoring interfaces: wan wg0 [✓]
> **:33:01 notice vpn-policy-routing[]: Reloading vpn-policy-routing due to ifup of wan (pppoe-wan)**
> :33:01 err modprobe[]: xt_set is already loaded
> :33:01 err modprobe[]: ip_set is already loaded
[... cut ...]
> :33:02 notice vpn-policy-routing[]: Reloading vpn-policy-routing due to ifup of wg0 (wg0)
> :33:02 err modprobe[]: xt_set is already loaded
> :33:02 err modprobe[]: ip_set is already loaded
> :33:02 err modprobe[]: ip_set_hash_ip is already loaded
> :33:02 notice [7841]: Creating table 'wan/84.x.x.x' [✓]
> :33:03 notice [7841]: Creating table 'wg0/192.168.112.12' [✓]
> :33:03 notice [7841]: Routing 's5' via wan [✓]
> :33:03 notice [7841]: service started on wan/84.x.x.x wg0/192.168.112.12 [✓]
> :33:03 notice [7841]: service monitoring interfaces: wan wg0 [✓]

stangri, 2018-07-15:
OK, it's not the local interface IP it's the "via" address in the routing table. It's taken from the output of ```ip -4 route```.

closed 2018-07-15 by stangri

n8v8R, 2018-07-15:
That is what I was trying to covey, it is the IPS's static gateway ip and not the ip assigned by the ISP. No issue then for the routing table created by VPR?

stangri, 2018-07-15:
It's only a problem if the IP used by VPR is different from gateway IP for the WAN interface reported by ```ip -4 route```. Is it?

n8v8R, 2018-07-15:
The WAN IP used/reported by VPR is matching the ISP gateyway fron the router's WAN ip as printed by `ip r` or` ip -4 route`

> default via 84.x.x.x dev pppoe-wan  proto static
> 84.x.x.x dev pppoe-wan  proto kernel  scope link  src 46.x.x.x

> Creating table 'wan/84.x.x.x' [✓]
> service started on wan/84.x.x.x wg0/192.168.112.12 [✓]

stangri, 2018-07-15:
Then it is the expected behaviour.