mossdef-org/source.mossdef.org #22: [bug] repeated/redundant VPR reload through LuCI save & apply

n8v8R, 2018-07-14:
luci-app-vpn-policy-routing | git-18.194.53747-b383..-29
vpn-policy-routing | 0.0.2-17

TOS 3.10.3 ({"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}})

---

referenced https://github.com/stangri/openwrt_packages/issues/18#issuecomment-404954076 and https://github.com/stangri/openwrt_packages/issues/18#issuecomment-404958726

So far discovered no correlation between the VPR settings in general or the number of assigned policies in particular. In this box the total number of reloads always counts to 11 and thus making it 10 redundant reloads.

n8v8R, 2018-07-14:
Potential correlation or just coincidence - number of online ifaces?

1. br-guest
2. br-lan 
3. eth0
4. eth1
5. eth2 
6. guest0 
7. lo
8. pppoe-wan 
9. wg0
10. wlan0     
11. wlan1

n8v8R, 2018-07-15:
Just a feedback since updated to

vpn-policy-routing/stangri_repo/0.0.2-19
luci-app-vpn-policy-routing/stangri_repo/git-18.196.37553-f29b2a2-29

the issue is still present

stangri, 2018-07-15:
Fixed in 0.0.2-20.

n8v8R, 2018-07-15:
Confirmed fixed!

Yet it seems to have introduced another bug...

Starting the service via LuCI it shows

> Service Status Started without PROCD support
> Service started with error

and a `Reload` button is showing.

The log is not showing an error when having started the service.

> err modprobe[]: xt_set is already loaded
> err modprobe[]: ip_set is already loaded
> err modprobe[]: ip_set_hash_ip is already loaded
> notice [32433]: Creating table 'wan/84.x.x.x' [✓]
> notice [32433]: Creating table 'wg0/192.168.112.12' [✓]
> notice [32433]: Routing 's5' via wan [✓]
> notice [32433]: Routing 's1' via wan [✓]
> notice [32433]: Routing 'da' via wan [✓]
> notice [32433]: Routing '1750e' via wan [✓]
> notice [32433]: Routing 'dvb-c' via wan [✓]
> notice [32433]: Routing 'stv9s' via wan [✓]
> notice [32433]: Routing 'oc_vpscp' via wan [✓]
> notice [32433]: Routing 'oc_vnc' via wan [✓]
> notice [32433]: Routing 'oc_ssh' via wan [✓]
> notice [32433]: Routing 'lan' via wan [✓]
> notice [32433]: service started on wan/84.x.x.x wg0/192.168.112.12 [✓]
> notice [32433]: service monitoring interfaces: wan wg0 [✓]

---
Pressing said `Reload` button then disables the service however...

> notice [8999]: vpn-policy-routing is currently disabled.
> notice [8999]: Run the following commands before starting service again:
> notice [8999]: uci set vpn-policy-routing.config.enabled='1'; uci commit;

stangri, 2018-07-15:
Both reload button behaviour and the conditions it is shown are fixed in ```luci-app-vpn-policy-routing 30```.

n8v8R, 2018-07-15:
Hmm... something gone awry

`/cgi-bin/luci/admin/services/vpn-policy-routing`

> /usr/lib/lua/luci/dispatcher.lua:460: Failed to execute cbi dispatcher target for entry '/admin/services/vpn-policy-routing'.
> The called action terminated with an exception:
> /usr/lib/lua/luci/model/cbi/vpn-policy-routing.lua:84: attempt to call method 'match' (a nil value)
> stack traceback:
> 	[C]: in function 'assert'
> 	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
> 	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

stangri, 2018-07-15:
My bad, try 31.

n8v8R, 2018-07-15:
No bad, you are hard at work providing something valuable for the OpenWRT community. I am grateful at least.

Seems fine now.

closed 2018-07-15 by n8v8R