mossdef-org/source.mossdef.org #23: [bug] uci parsing error

n8v8R, 2018-07-15:
vpn-policy-routing | 0.0.2-20

TOS 3.10.3 ({"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}})

---

excecuting from cli 
* `uci set vpn-policy-routing.config.enabled='1'; uci commit;` 
* `uci set vpn-policy-routing.config.enabled='0'; uci commit;`

produces

> uci: Parse error (invalid command) at line 12, byte 0

stangri, 2018-07-15:
Cannot reproduce.

n8v8R, 2018-07-15:
Maybe something specific to the (outdated) TO repo implementation of OpenWRT and will go away once they around to release TOS 4. Not impotant anyway, only a few users likely to use it, if ever. Hence please feel at liberty to close this.

> -ash: /etc/config/vpn-policy-routing: Permission denied
> -ash: /etc/config/ucitrack: Permission denied

stangri, 2018-07-15:
I meant the content of the files.

n8v8R, 2018-07-15:
Oops, me simpleton

```
config policy
	option local_addresses '192.168.112.120'
	option comment 's5'
	option interface 'wan'

config policy
	option comment 's1'
	option local_addresses '192.168.112.121'
	option interface 'wan'

config policy
	option comment 'da'
	option local_addresses '192.168.112.160'
	option interface 'wan'

config policy
	option comment '1750e'
	option local_addresses '192.168.112.190'
	option interface 'wan'

config policy
	option comment 'dvb-c'
	option local_addresses '192.168.112.191'
	option interface 'wan'

config policy
	option comment 'stv9s'
	option local_addresses '192.168.112.122'
	option interface 'wan'

config policy
	option interface 'wan'
	option comment 'oc_vpscp'
	option remote_addresses 'vpscp.[obfuscated].com'
	option remote_ports '4083'

config policy
	option interface 'wan'
	option comment 'oc_vnc'
	option remote_addresses '179.x.x.x'
	option remote_ports '5904'

config policy
	option interface 'wan'
	option comment 'oc_ssh'
	option remote_addresses '179.x.x.x'
	option remote_ports '26009'

config policy
	option comment 'lan'
	option local_addresses '192.168.112.12/24'
	option interface 'wan'

config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option udp_proto_enabled '1'
	option strict_enforcement '1'
	list ignored_interface 'guest'
	list ignored_interface 'tun'
	option dnsmasq_enabled '0'
	option ipset_enabled '1'
	option output_chain_enabled '1'
	option icmp_interface 'wg0'
	option enabled '0'
```
---

```
config network
	option init 'network'
	list affects 'dhcp'
	list affects 'radvd'

config wireless
	list affects 'network'

config firewall
	option init 'firewall'
	list affects 'luci-splash'
	list affects 'qos'
	list affects 'miniupnpd'
	list affects 'vpn-policy-routing'

config olsr
	option init 'olsrd'

config dhcp
	option init 'dnsmasq'
	list affects 'odhcpd'

config odhcpd
	option init 'odhcpd'

config dropbear
	option init 'dropbear'

config httpd
	option init 'httpd'

config fstab
	option init 'fstab'

config qos
	option init 'qos'

config system
	option init 'led'
	list affects 'luci_statistics'

config luci_splash
	option init 'luci_splash'

config upnpd
	option init 'miniupnpd'

config ntpclient
	option init 'ntpclient'

config samba
	option init 'samba'

config tinyproxy
	option init 'tinyproxy'

config hd-idle
	option init 'hd-idle'

config minidlna 'minidlna'

config transmission
	option init 'transmission'

config unbound
	option init 'unbound'

config vpn-policy-routing
	option init 'vpn-policy-routing'
```

stangri, 2018-07-15:
Uhm, these files look legit and VPR doesn't change anything else. I suspect some other file in ```/etc/config``` is broken. ```uci commit``` then complains.

Try:
```sh
uci set vpn-policy-routing.config.enabled='1'; uci commit vpn-policy-routing;
uci set vpn-policy-routing.config.enabled='0'; uci commit vpn-policy-routing;
```
instead.

n8v8R, 2018-07-15:
yup, that works, no issue

closed 2018-07-15 by n8v8R