mossdef-org/source.mossdef.org #24: [bug] update to 0.0.2-22 send router into reboot loop

n8v8R, 2018-07-21:
TOS 3.10.3 ({"kernel":"4.4.138-1e8e1b4c23f383e990eb3c4f490f5f2e-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}})

---
After having updated to 0.0.2-22 and applied some settings the router went into a reboot loop, rebooting every few minutes.
Even disbaling VPR did not solve it.

Had to rollback the router (0.0.2-21) and since then the reboots have ceased.

Since the systemlog is rolled over at each reboot it was not possible to gather information of how VPR exactly cause the reboots.

n8v8R, 2018-07-21:
Well, it been a little surpise indeed.

> ERROR: curl, libopenssl or ca-bundle were not found!
> Run 'opkg update; opkg install curl libopenssl ca-bundle' to install them.

`curl` and `libopenssl` are present whilst `ca-bundle` is not as not being provided by the TO repo (sigh)...

The [TO team been made aware of it](https://forum.turris.cz/t/location-of-ca-certificate-bundle-file/7548) since it is needed for TLS authentication of the unbound resolver but apparently they cannot be bothered.

Out of desperation I got `/etc/ssl/certs/tls-ca-bundle.pem` from one of my linux boxes...

TO repo currently only provides `ca-certificates` which is a bunch of _single_ root certificates and which perhaos could be converted to a bundle with `cat /etc/ssl/certs/* > /etc/ssl/certs/ca-bundle.pem`. 

Suppose I should not install 0.0.2-23 without `ca-bundle`?

stangri, 2018-07-21:
Uhm, if you could just capture the output of ```/etc/init.d/vpn-policy-routing support``` before upgrade and e-mail it to me if 0.0.2-23 fails, that'd also work.

n8v8R, 2018-07-21:
Unfortunately that did not fix it, still rebooting in a loop...

Seems that whatever is causing it been introduced prior 0.0.2-21 already because after the rollback I repated the step with the same outcome (reboot loop).

I kept on testing new releases for #10 and thus the steps were adding a lan segment to be routed trough a WG iface.

Just to be clear the reboot loop does not occur after the update but only after the previous step and pressing `save & apply`

Now I tried the same steps with 0.0.2-21 except adding the lan segment to WAN instead and _no_ reboot loop after pressing `save & apply`!

From that point made the upgrade to 0.0.2-23 and then changed the aforementioned lan segement from WAN to WG and pressed `save & apply` and _bang_ the rebooting loop starts all over again.

The last documented testing on #10 was with 0.0.2-11 but I reckon that I might have checked also in between 2-11 and 2-20, just cannot recall which was the last.

It seems that whenever the WG iface gets active/online and picked up by VPR the reboot is initiated.

Still need the output from `/etc/init.d/vpn-policy-routing support`?

n8v8R, 2018-08-03:
Not that I expected it to be related to luci-app-vpn-policy-routing I gave a it try nonetheless after having updated to luci-app-vpn-policy-routing/stangri_repo/git-18.212.55585-176e246-32, just for good measure.

It did not remedy the issue however.

Do you reckon this being another TO specific issue, or rather something WG specific?

n8v8R, 2018-08-10:
updated to vpn-policy-routing 0.0.2-23 | luci-app-vpn-policy-routing git-18.220.31345-7609919-32 but the issue is still present

n8v8R, 2018-08-15:
meantime upgraded to

- TOS  3.10.**4** {"kernel":"4.4.**147**-0a333a8e606ab056173befac424900d2-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}
- vpn-policy-routing  0.0.2-2**4**
- luci-app-vpn-policy-routing git-18.225.48207-809d3b9-33

the issue is presistent and there when monitoring the system log there is no print out as to the cause of the reboot

n8v8R, 2018-08-23:
migrated to

- vpn-policy-routing 0.0.2-2**5**
- {"kernel":"4.4.**150**-0a333a8e606ab056173befac424900d2-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}
- TOS 3.10.**5**

---
the issue is presistent but for the first time now **got a log print out!**

> [88410.513258] Unable to handle kernel NULL pointer dereference at virtual address 00000002
> alert kernel[]: [88410.521389] pgd = ea118000

n8v8R, 2018-09-07:
migrated to vpn-policy-routing 0.0.2-2**8** with the issue persistent, the log showing:

> alert kernel[]: [1071442.984542] Unable to handle kernel paging request at virtual address 40a5f134
> alert kernel[]: [1071442.991974] pgd = ecbc4000
> alert kernel[]: [1071442.994877] [40a5f134] *pgd=00000000
> emerg kernel[]: [1071442.998655] Internal error: Oops: 5 [#1] SMP ARM

n8v8R, 2018-09-20:
migrated to

- {"kernel":"4.4.**156**-0a333a8e606ab056173befac424900d2-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}
-  TOS 3.10.**6**
- kmod-wireguard 	4.4.156+0.0.20180910-..2-1

---
issue persistent, the log showing:

> alert kernel[]: [ 4591.684808] Unable to handle kernel paging request at virtual address 312f736d
> alert kernel[]: [ 4591.692084] pgd = ea5a4000
> alert kernel[]: [ 4591.694796] [312f736d] *pgd=00000000
> emerg kernel[]: [ 4591.698432] Internal error: Oops: 5 [#1] SMP ARM
> warning kernel[]: [ 4591.703059] Modules linked in: qcserial option iptable_nat ath9k usb_wwan rndis_host qmi_wwan pppoe nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cdc_ether ath9k_common armada_thermal xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm thermal_sys pppox ppp_mppe ppp_async nfnetlink_queue nfnetlink_log nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast mvsdio iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hwmon ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt cdc_wdm ath9k_hw fuse sch_cake nf_conntrack sch_teql sch_tbf sch_sfq sch_red sch_prio sch_pie sch_netem sch_htb sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_vlan act_police act_pedit act_nat act_ipt act_gact act_csum act_bpf act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress ath10k_pci ath10k_core ath mac80211 cfg80211 compat ledtrig_usbdev ledtrig_oneshot xt_LED ledtrig_morse ledtrig_heartbeat ledtrig_gpio cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables pppoatm ppp_generic slhc nfsd nfsv3 msdos ip_gre gre ifb ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel veth tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd input_core soundcore vfat fat udf crc_itu_t ntfs nfs_layout_nfsv41_files nfsv4 nfs auth_rpcgss oid_registry lockd sunrpc grace minix isofs hfsplus hfs cramfs configfs cifs autofs4 dns_resolver dm_crypt dm_mirror dm_region_hash dm_log dm_mod br2684 atm multipath fscache raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod nls_utf8 nls_koi8_r nls_cp1255 nls_iso8859_6 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp932 nls_cp866 nls_cp864 nls_cp862 nls_cp852 nls_cp850 nls_cp775 nls_cp437 nls_cp1251 nls_cp1250 xts algif_skcipher algif_hash af_alg sha512_generic sha256_generic sha1_generic seqiv jitterentropy_rng drbg pcbc md5 md4 marvell_cesa hmac gf128mul ecb des_generic ctr cmac ccm cbc authenc xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ledtrig_transient ehci_orion ehci_platform ehci_hcd ahci_platform uas usb_storage xfs libcrc32c jfs f2fs exfat usbcore nls_base usb_common mii aead crypto_null
> warning kernel[]: [ 4592.014034] CPU: 0 PID: 2796 Comm: mountd Not tainted 4.4.156-0a333a8e606ab056173befac424900d2-1 #1
> warning kernel[]: [ 4592.023101] Hardware name: Marvell Armada 380/385 (Device Tree)
> warning kernel[]: [ 4592.029033] task: ee9a9680 ti: ec9f6000 task.ti: ec9f6000
> warning kernel[]: [ 4592.034448] PC is at kmem_cache_alloc+0x44/0x110
> warning kernel[]: [ 4592.039077] LR is at locks_alloc_lock+0x24/0x70
> warning kernel[]: [ 4592.043618] pc : [<c00e6268>]    lr : [<c013a178>]    psr: 20000013
> warning kernel[]: [ 4592.043618] sp : ec9f7ec8  ip : ec9f7ef0  fp : ec9f7eec
> warning kernel[]: [ 4592.055121] r10: 00000000  r9 : 00000000  r8 : 0008de76
> warning kernel[]: [ 4592.060356] r7 : 00000000  r6 : 024080c0  r5 : 312f736d  r4 : ee801e80
> warning kernel[]: [ 4592.066897] r3 : 00000000  r2 : eedce570  r1 : 024080c0  r0 : 2e6a4000
> warning kernel[]: [ 4592.073438] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
> warning kernel[]: [ 4592.080589] Control: 10c5387d  Table: 2a5a404a  DAC: 00000051
> emerg kernel[]: [ 4592.086346] Process mountd (pid: 2796, stack limit = 0xec9f6210)
> emerg kernel[]: [ 4592.092364] Stack: (0xec9f7ec8 to 0xec9f8000)
[... cut ...]
> warning kernel[]: [ 4592.280961] ath10k_pci 0000:02:00.0: SWBA overrun on vdev 1, skipped old beacon
> warning kernel[]: [ 4592.288315] ath10k_pci 0000:02:00.0: SWBA overrun on vdev 0, skipped old beacon
> warning kernel[]: [ 4592.295656] ath10k_pci 0000:02:00.0: SWBA overrun on vdev 1, skipped old beacon
> warning kernel[]: [ 4592.302986] ath10k_pci 0000:02:00.0: SWBA overrun on vdev 0, skipped old beacon
> warning kernel[]: [ 4592.310330] ath10k_pci 0000:02:00.0: SWBA overrun on vdev 1, skipped old beacon

n8v8R, 2018-10-15:
migrated to

- {"kernel":"4.4.**161**-0a333a8e606ab056173befac424900d2-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}}
- TOS 3.10.**8**
- kmod-wireguard 4.4.161+0.0.2018**0918**-..2-1

with the issue still present

n8v8R, 2018-11-29:
- SQM enabled on WAN and disabled on WG then it is working.
- Enabling SQM on WG and starting VPR causes the kernel panic.

I am not sure whether it makes sense to have SQM enabled on WAN and WG but for now it remains disabled on WG to prevent said kernel panic with VPR

n8v8R, 2018-11-29:
from the TO repo 

sqm-scripts 	1.0.7-3
sqm-scripts-extra 	2016-06-08-1

I would not know how to debug the potential collisions.

stangri, 2018-11-29:
I asked because I'm not aware how they're building packages for their repo.
They use the same mask as the mainline OpenWrt SQM (0x0000ff), which shouldn't conflict with what VPR uses (0xff0000). I'm at loss what's causing the kernel panic.
@tohojo -- do you have any idea why enabling SQM on a wireguard tunnel and starting VPR can cause the kernel panic?

tohojo, 2018-11-29:
Might be because running SQM re-enables a qdisc on the interface, and wg expects to run in queue-less mode?

Why are you running SQM on the wg interface instead of the underlying physical interface?

n8v8R, 2018-11-29:
@tohojo 

SQM is running on WAN interface, which is supposedly the underlying interface for WG. But I was not sure whether SQM on the WAN would also deal with the packages on the WG interface and thus enabled it there as well. 

Having said that SQM enabled on both interfaces, WAN and WG, does works without VPR enabled. Starting VPR in this setting is causing the kernel panic however with this error reported by the kernel

> Unable to handle kernel paging request at virtual address

tohojo, 2018-11-29:
Ah, scrolled up some more now.

Don't have a good explanation for the crash; but with the current structure of wireguard, you won't get much benefit from running sqm on the wg interface, so a solution for now could be to just not do that? :)

closed 2018-12-05 by stangri

n8v8R, 2018-12-05:
so v031 fixed this then? Enabled SQM on WG and restarted VPR and no more crash :+1: :smiley: