mossdef-org/source.mossdef.org #28: [ feature suggestion ] route via lxc container ip

n8v8R, 2018-08-15:
Placing the VPN client inside an unpriviliged lxc container the VPN interface is not available at the router host. Whilst the lxc container being paired with a bridge (interface) at the (router) host the bridge interface could hold other containers than just the one for the VPN service.

Thus routing VPN traffic via an unpriviliged lxc container would require an option to route via a designated ip rather than interface.

stangri, 2019-09-27:
I don't have an LXC-capable router to test -- how does OpenWrt handle the VPN client inside the LXC container generally?

n8v8R, 2019-09-28:
LXC utilizes the `veth` interface on the host side and `eth` interface in the guest. 

Currently I do not have a LXC guest up but if not mistaken the `veth` is cannot be routed in OpenWrt since it requires to be slaved into a bridge.

Anyway, this is probably rather a fringe case and not likely to be utilized by many users. I initially opened it because the TOS repo been lagging back then to keep up with upstream development and the LXC guest been a way to utilize more up-to-date software. Since then the repo's upkeep has improved and thus closing this issue.

closed 2019-09-28 by n8v8R