mossdef-org/source.mossdef.org #29: [VPR bug] vpn-policy-routing package not available from this repo

n8v8R, 2018-08-16:
- TOS  3.10.4 {"kernel":"4.4.147-0a333a8e606ab056173befac424900d2-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}}
- vpn-policy-routing  0.0.2-24
- luci-app-vpn-policy-routing  git-18.225.48207-809d3b9-33

---
trying to install VPR via LuCI the following is reported

> Installing luci-app-vpn-policy-routing (git-18.225.48207-809d3b9-33) to root...
> Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/luci-app-vpn-policy-routing_git-18.225.48207-809d3b9-33_all.ipk
> 
> Collected errors:
>  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-vpn-policy-routing:
>  * 	vpn-policy-routing * 
>  * opkg_install_cmd: Cannot install package luci-app-vpn-policy-routing.

---
`opkg install vpn-policy-routing`

> Unknown package 'vpn-policy-routing'.
> Collected errors:
>  * opkg_install_cmd: Cannot install package vpn-policy-routing.

stangri, 2018-08-16:
Stumbled upon it about same time you've created the bug. Please try again.

n8v8R, 2018-08-16:
working now

closed 2018-08-16 by n8v8R