mossdef-org/source.mossdef.org #38: [VPR - feature suggestion] add option to pause/resume policy

n8v8R, 2018-11-30:
It may not be a widely deployed user scenario but right now it is possible to add/delete a policy but not to pause/resume such, only globally for VPR (or editing the config via ssh)
When one is testing or deploying policies only intermittent it thus requires policies to be deleted and added.

stangri, 2018-11-30:
Thanks for the suggestion. 

My concern is that the Web UI is already pretty wide and on the less than 15" monitors it barely fits, I don't want to introduce another column.

And if you'd have to edit the config file manually, commenting out the policy is just as quick as adding ```option disabled '1'```.

PS. As long as you don't use WebUI, the commented out policy should stay in the config file.

n8v8R, 2018-11-30:
> My concern is that the Web UI is already pretty wide and on the less than 15" monitors it barely fits, I don't want to introduce another column.

Sure, can appreciate that concern.

I had been thinking though to split the delete button 3-ways and use icons only instead. e.g. `II > X`

closed 2018-11-30 by n8v8R

stangri, 2018-11-30:
The delete button (as well as the rest of the policies table) comes from the Luci template, with my limited knowledge of Luci, it would be a major undertaking to change the delete button to something else.