mossdef-org/source.mossdef.org #39: [VPR bug] 0.0.2-29 failed to load kernel modules!

n8v8R, 2018-12-05:
@stangri 

Just updated to 0.0.2-29 and  `/etc/init.d/vpn-policy-routing restart` producing

> vpn-policy-routing 0.0.2-29 stopped [✓]
> ERROR: vpn-policy-routing 0.0.2-29 failed to load kernel modules!

That is on TOS 3.10.8 (OpenWrt omnia 15.05 r47055 with kernel 4.4.161)

stangri, 2018-12-05:
Can you try with 29 installed: in file /etc/init.d/vpn-policy-routing comment out (put a # as a first symbol) lines 533-537?

Thanks!

n8v8R, 2018-12-05:
with those lines commented out VPR starts with no error reported, however to my understanding that is just suppression the status check _but the actual VPN routing is not working_ 

All traffic is only passing through the standard gateway, in this case WAN and none via VPN as designated in the policies. Thus suppose suppressing the status check is not restoring functionality as it been in version 0.0.2-28

stangri, 2018-12-05:
what about -030?

n8v8R, 2018-12-05:
I am afraid not

~# "/etc/init.d/vpn-policy-routing" restart

> vpn-policy-routing 0.0.2-30 stopped [✓]
> ERROR: vpn-policy-routing 0.0.2-30 failed to load kernel modules!
> vpn-policy-routing 0.0.2-30 monitoring interfaces: wan wg0 [✓]

`~# "/etc/init.d/vpn-policy-routing" stop`

> vpn-policy-routing 0.0.2-30 stopped [✓]

`~# "/etc/init.d/vpn-policy-routing" start`

> ERROR: vpn-policy-routing 0.0.2-30 failed to load kernel modules!
> vpn-policy-routing 0.0.2-30 monitoring interfaces: wan wg0 [✓]

stangri, 2018-12-05:
And yet, the 028 having identical code works?

n8v8R, 2018-12-05:
Oh wait, you meant with those line commented out, did you not?

Yes, with those lines commented out VPR starts and the routing policies are working

stangri, 2018-12-05:
Should be fixed in 031, please confirm.

n8v8R, 2018-12-05:
Confirmed, fixed and working

closed 2018-12-05 by n8v8R