mossdef-org/source.mossdef.org #40: luci-app-advanced-reboot

celticnestor, 2018-12-11:


celticnestor, 2018-12-11:
Hi @stangri,

Appreciate the hard work on packages that really make a difference!

I am running luci-app-advanced-reboot on Lede SNAPSHOT r8466-251c350727 / LuCI Master (git-18.320.75602-93fdac8), a Davidc502 LEDE Snapshot.

When taking action "Reboot to alternative partition..." I am getting following error message:
`Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/alternative_reboot'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:124: attempt to concatenate global 'errorMessage' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/advanced_reboot.lua:124: in function </usr/lib/lua/luci/controller/advanced_reboot.lua:97>
	(tail call): ?`

Is there anything I can do to help debug this issue?
Is there an alternative command to force the router boot from the other partition?

Thx for your help!

celticnestor, 2018-12-12:
Silly me.... device is Linksys WRT3200ACM-RM2, should be supported.

Apologies for the basic q: where can I get a build 034? I am unclear on how to relate that to Davidc shapshots...

Thx again!

stangri, 2019-01-13:
As far as I understand, all flavours of WRT3200 family are supported. Check the repo README for instructions.

closed 2019-01-13 by stangri

stangri, 2019-01-14:
Which version of the advanced reboot app are you running?

celticnestor, 2019-01-15:
luci-app-advanced-reboot   git-18.320.75602-93fdac8-33

celticnestor, 2019-01-23:
@stangri Any comments on this?

stangri, 2019-01-27:
Please update to latest version.

celticnestor, 2019-02-07:
Hi @stangri 

Upgraded to luci-app-advanced-reboot git-19.011.54860-47d9205-33 and the issue continues to be there. This is the error message I am getting now:

> Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/alternative_reboot'.
> The called action terminated with an exception: [...]
</usr/lib/lua/luci/controller/advanced_reboot.lua:97>
>	(tail call): ?

Anything I can do to recover this functionality? THX!

stangri, 2019-02-07:
Latest version should have -37 at the end of the version string.