mossdef-org/source.mossdef.org #41: [VPR bug] no WG traffic with kernel reverse-path filter enabled

n8v8R, 2018-12-12:
TOS 3.11
{"kernel":"4.4.165-4a7a81f8db0ad743e54c68e1845c60b6-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}

VPR 0.0.2-31

---
enabling kernel reverse-path filter via sysctl

```
net.ipv4.conf.default.rp_filter=1
net.ipv4.conf.all.rp_filter=1
```

ceases the VPR routed traffic on the WG iface. Instead packages showing up as Martians.

stangri, 2018-12-19:
I'm guessing some of the kernel modules required by VPR are incompatible with RP-path filter.

As I do not currently have the WG set up, it may take me a while to test this on 18.06.1.

n8v8R, 2018-12-19:
Sure, no worries. Is not a big issue. Happy season! :mrs_claus:

n8v8R, 2019-02-06:
setting `rp_filter = 2` solves the matter.

> 2 - Loose mode as defined in RFC3704 Loose Reverse Path
>     Each incoming packet's source address is also tested against the FIB
>     and if the source address is not reachable via any interface
>     the packet check will fail.
> Current recommended practice in RFC3704 is to enable strict mode
> to prevent IP spoofing from DDos attacks. **If using asymmetric routing
> or other complicated routing, then loose mode is recommended.**

closed 2019-02-06 by n8v8R