mossdef-org/source.mossdef.org #42: [VPR bug] bridge name parsing

n8v8R, 2018-12-19:
@stangri 

There seems to be a mismatch in parsing a bridge iface name `br-guest` from supported interface - it turns up as `guest` only in the other options. Since there is no actual `guest` interface on the system the routing policy fails and the log output shows:

> service started on (strict mode): guest/0.0.0.0 [✓]

```
config vpn-policy-routing 'config'
	option enabled '1'
	option verbosity '2'
	option ipv6_enabled '0'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	option udp_proto_enabled '1'
	option wan_dscp '1'
	option wg0_dscp '2'
	option guest_dscp '3'
	list supported_interface 'br-guest'

config policy
	option name 'router-hpot'
	option remote_addresses '10.168.112.67'
	option remote_ports '56007'
	option interface 'guest'
	option local_addresses '192.168.112.12/24'
```

Changing the `guest` entries manually via ssh edit into `br-guest ` and then refreshing the LuCI page produces:

> The configuration file could not be loaded due to the following error:
> uci: Parse error (invalid character in name field)

Is this a UCI bug/limitation rather than VPR?

stangri, 2018-12-19:
Cannot reproduce the Luci bug on mainline 18.06.1.

With the proper ("br-guest") interface specified in the config, could you please run and post the output of:
```sh
source /lib/functions.sh
packageName="vpn-policy-routing"
config_load "$packageName"
config_get supportedIfaces          'config' 'supported_interface'
echo "$supportedIfaces"
process_interface(){ echo "$1"; }
config_load 'network'; config_foreach process_interface 'interface' 'create';
```

n8v8R, 2018-12-19:
> loopback
> lan
> wan
> wan6
> wg0
> guest

It seems it does not matter, even with `list supported_interface 'br-guest'` the output still reads

>  service started on (strict mode): guest/0.0.0.0 [✓]

Seems like VPR is not picking up the guest ip for some reason and thus the policy is not working

[image]

stangri, 2018-12-19:
So the interface name is just guest.
Can you please post (maybe in a short-lived pastie) the output of:
```sh
ip -4 route
ifconfig
```

n8v8R, 2018-12-19:
> default via XX.46.104.107 dev pppoe-wan  proto static
> 10.168.112.0/24 dev br-guest  proto kernel  scope link  src 10.168.112.12
> XX.46.104.107 dev pppoe-wan  proto kernel  scope link  src XX.233.133.121
> XX.43.174.98 via XX.46.104.107 dev pppoe-wan  proto static
> 192.168.112.0/24 dev br-lan  proto kernel  scope link  src 192.168.112.12

> br-guest  Link encap:Ethernet  HWaddr D8:58:D7:00:87:4D
>           inet addr:10.168.112.12  Bcast:10.168.112.255  Mask:255.255.255.0
>           UP BROADCAST RUNNING MULTICAST  MTU:2304  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

stangri, 2018-12-19:
VPR shows the gateway, not an IP of the interface. Without the gateway VPR doesn't know how to route the packages from the interface.

n8v8R, 2018-12-19:
I had set a static route but that been superseded by this VPR policy
```

config policy
	option name 'lan'
	option local_addresses '192.168.112.12/24'
	option interface 'wg0'
	option remote_addresses '0.0.0.0/0'
```

n8v8R, 2018-12-19:
I put the static route back in but with type `unicast ` and get

> vpn-policy-routing 0.0.2-32 started on wan/84.46.104.107 wg0/192.168.45.2 guest/10.168.112.12 [✓]

but the local traffic is still escaping via the WG policy, which is at the bottom of the policies

`tracert hpot`

> Tracing route to hpot.lan [10.168.112.67]
> over a maximum of 30 hops:
> 
>   1     1 ms     1 ms     1 ms  192.168.112.12
>   2    21 ms    21 ms    20 ms  172.25.121.1
>   3    21 ms    21 ms    21 ms  172.25.120.1

stangri, 2018-12-19:
Well, the only policy you've set is for port 56007, so any traffic but destined for port 56007 should be routed thru default gateway.

n8v8R, 2018-12-19:
Point taken, but still no luck. Right now I had to stop VPR as the WG went dead.

I will see if I can figure it out somehow. It just I had hoped that VPR would honour static routes but it appears that is not aware of those.

n8v8R, 2018-12-20:
seems that 0.0.2-33 introduced some regression in this regard as the iface is not being recognised. :frowning_face: 

`ERROR: policy 'router-hpot' has an unknown interface: iced!`

Ran the script from above and got this output

> loopback
> lan
> wan
> wan6
> wg0
> guest
> iced

---

> default via xx46.104.107 dev pppoe-wan  proto static
> 10.168.112.0/24 dev br-guest  proto kernel  scope link  src 10.168.112.12
> xx.46.104.107 dev pppoe-wan  proto kernel  scope link  src xx.46.10.175
> 172.20.112.0/24 dev br-iced  proto kernel  scope link  src 172.20.112.12
> 179.43.174.98 via 84.46.104.107 dev pppoe-wan  proto static
> 192.168.112.0/24 dev br-lan  proto kernel  scope link  src 192.168.112.12

> br-iced   Link encap:Ethernet  HWaddr D8:58:D7:00:87:4B
>           inet addr:172.20.112.12  Bcast:172.20.112.255  Mask:255.255.255.0
>           UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1

---
```
config policy
	option name 'router-hpot'
	option remote_ports '56007 2222'
	option interface 'iced'
	option remote_addresses '172.20.112.190'
```

n8v8R, 2018-12-23:
Making changes to the routing table(s), e.g. setting static routes/gateway such are not being picked up by VPR and thus require a manual restart of VPR.

Might be worth a hint mentioning in the read.me?

n8v8R, 2018-12-25:
@stangri 

Would not want to be a bother but there is a potential VPR bug

| Destination | Gateway | Genmask | Flags | Metric | Iface |
|----------------|----------------|-----------------|-------|--------|-----------|
| 0.0.0.0 | xxx.46.104.107 | 0.0.0.0 | UG | 0 | pppoe-wan |
| 10.168.112.0 | 0.0.0.0 | 255.255.255.0 | U | 0 | br-guest |
| xxx.46.104.107 | 0.0.0.0 | 255.255.255.255 | UH | 0 | pppoe-wan |
| 172.20.112.0 | 0.0.0.0 | 255.255.255.0 | U | 0 | br-iced |
| xxx.43.174.98 | xxx.46.104.107 | 255.255.255.255 | UGH | 0 | pppoe-wan |
| 192.168.112.0 | 0.0.0.0 | 255.255.255.0 | U | 0 | br-lan |

xxx.43.174.98 -> xxx.46.104.107 cretaed by wireguard but the route is not active.

Network is working fine, public traffic escaping via `pppoe-wan` and lan clients are accessible in the respective subnets.
___
Adding `VPR`

config policy
	option name 'router-iced'
	option remote_addresses '172.20.112.12/24'
	option interface 'iced'

config policy
	option name 'lan'
	option local_addresses '192.168.112.12/24'
	option interface 'wg0'

config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	list ignored_interface 'guest'
	option udp_proto_enabled '1'
	list supported_interface 'iced'
	option enabled '1'
___
`ip r flush cache && ip r show cache`

`ip r get 172.20.112.190`

> 172.20.112.190 dev br-iced  src 172.20.112.12
>     cache

`ip r show to root 172.20.112.190/24`

> 172.20.112.0/24 dev br-iced  proto kernel  scope link  src 172.20.112.12

Clearly the kernel has the correct routing. However VPR is incorrectly routing that subnet via the wireguard interface.

`ipset l`

```
Name: wg0
Type: hash:net
Revision: 6
Header: family inet hashsize 1024 maxelem 65536 comment
Size in memory: 344
References: 1
Members:
```

```
Name: iced
Type: hash:net
Revision: 6
Header: family inet hashsize 1024 maxelem 65536 comment
Size in memory: 344
References: 1
Members:
```

And here I am wondering whether the order of the listings matters, i.e. whether those listed top are being priotized over the lower ranks? 
Surely the list is not aphabetically and neither appears randomly but aribiratily sorted.

n8v8R, 2018-12-25:
Just had look at the generated iptable and it seems strange that is has **2** entries for `MARK xset 0x30000/0xff0000` but different destination ip and thus likely being the root cause. Hope that helps to get this ironed out :sweat_smile: 

| Target | Prot. | In | Out | Source | Destination | Options |
|--------|-------|----|-----|------------------|-----------------|-----------------------------------------------|
| MARK | all | * | * | 192.168.112.0/24 | 0.0.0.0/0 | /* lan */ MARK xset 0x20000/0xff0000 |
| MARK | all | * | * | 0.0.0.0/0 | 172.20.112.0/24 | /* router-iced */ MARK xset 0x30000/0xff0000 |
| MARK | all | * | * | 0.0.0.0/0 | 0.0.0.0/0 | match-set iced dst MARK xset 0x30000/0xff0000 |
| MARK | all | * | * | 0.0.0.0/0 | 0.0.0.0/0 | match-set wg0 dst MARK xset 0x20000/0xff0000 |

n8v8R, 2018-12-31:
One rule, with the correct gateway, gets written for the VPR rule name and then a second is being written for the iface name, that one with the wrong gateway.

Being completely script illiterate I for sure failed to discover the culprit :sob:

n8v8R, 2019-01-02:
@stangri I have run the init script with

```
set -v
set -x
set -o xtrace
```

But neither produces a viable console output to figure out which part of the init script is causing those double rules with different gateways for the same iface. Is there any (other) debug switch that would help in debugging?

n8v8R, 2019-01-05:
@stangri found the issue - there is discrepancy in the ifname convention between `uci `and `iproute2`, which both are deployed in the init script.

For some curious reason `uci` is omitting the br-prefix for network bridges, which seems like bug (, like I suspected in the initial post) since the br- suffix is the common naming convention and even automatically added by the OS, at least TOS but would expect the same at upstream OpenWRT.

`uci show network` prints

> network.**iced**=interface
> network.**iced**.type='bridge'
> network.**iced**.proto='static'
> network.**iced**.ifname='eth0.2

whilst `ip l` prints

> br-iced: <BROADCAST,MULTICAST,UP,LOWER_UP>

---
```
source /lib/functions.sh
packageName="vpn-policy-routing"
config_load "$packageName"
config_get supportedIfaces          'config' 'supported_interface'
echo "$supportedIfaces"
```
> br-iced

```
process_interface(){ echo "$1"; }
config_load 'network'; config_foreach process_interface 'interface' 'create';
```

> iced
---
Unfortunately I have to stick with a bridge device since running a LXC container requires a bridge iface.

n8v8R, 2019-01-06:
The issue seems somewhere along device vs. iface.

`network_get_device if_icced iced; echo $if_iced` prints

> br-iced

and 

`ubus call network.interface.iced status`

>         "l3_device": "br-iced",
>         "device": "br-iced",
___
so there is a dev named _br-iced_ and then there is an iface named _iced_.

This becomes apparent by extending l469 of the init script

`network_get_device dev "$iface"; echo $if_$iface; echo $if_$dev`

> iced
> br-iced
> Creating table 'iced/0.0.0.0' [✓]

stangri, 2019-01-07:
First of all, thank you for the investigation.

Maybe I didn't name the setting correctly, it probably should be called not supported interfaces, but supported networks. So in VPR settings it should be iced, not br-iced.

Did you also post something about ```uci``` and ```ip/ifconfig``` using different device names rather than interface/network names in some output resulting in duplicate rules?

n8v8R, 2019-01-07:
That prints

> guest - br-guest
> iced - br-iced
---

`network_get_gateway() ` seems not to work for a br-iface since it being a L3 dev and thus its connected iface not being logical in the way it is interpreted by netifd & ubus. That would be my understanding.

From my understanding `iproute2` does not make a distinction between dev and iface but prints the ifname of the dev and not the iface. Now you got the gateway info for _br-iced_ but it is not matching the netifd iface ifname.

Which explains the wrong gateway.
---
Now about the fw rules being created twice by VPR I thought it was because of `process_policy` and `process_interface` but could not not find a fault in the logic. Could be related though to issue with dev name vs. face name.

stangri, 2019-01-07:
So dev names are being properly identified for interfaces on your system.

Can you please post the full output of ```ip -4 route | grep br-iced``` and ```ifconfig | grep br-iced``` (as text, not a screenshot).

stangri, 2019-01-07:
Also, what's the exact version of ```luci-app-vpn-policy-routing```?

n8v8R, 2019-01-07:
luci-app-vpn-policy-routing 	git-18.356.64384-31d2..-34
vpn-policy-routing 0.0.2-34

---
`ip -4 route | grep br-iced`

> 172.20.112.0/24 dev br-iced  proto kernel  scope link  src 172.20.112.12

---
`ifconfig | grep br-iced`

> br-iced   Link encap:Ethernet  HWaddr D8:58:D7:00:87:5C

stangri, 2019-01-07:
Would ```172.20.112.12``` be the correct gateway for VPR to use for that bridge interface?

n8v8R, 2019-01-07:
yes, indeed it is the correct gateway.

Yet VPR produces 

Chain VPR_PREROUTING (References: 5)

| Target | Prot. | In | Out |   Source  |   Destination   |                    Options                    |
|:------:|:-----:|:--:|:---:|:---------:|:---------------:|:---------------------------------------------:|
| MARK   | all   | *  | *   | 0.0.0.0/0 | 172.20.112.0/24 | /* test */ MARK xset 0x30000/0xff0000         |
| MARK   | all   | *  | *   | 0.0.0.0/0 | 0.0.0.0/0       | match-set iced dst MARK xset 0x30000/0xff0000 |

stangri, 2019-01-07:
Please test ```vpn-policy-routing 0.0.2-35```.

n8v8R, 2019-01-07:
The routing appears to be working correctly/as intended :+1: 

`/etc/init.d/vpn-policy-routing" restart`

> Routing 'router-iced' via iced [✓]
> vpn-policy-routing 0.0.2-35 started on iced/172.20.112.12 [✓]
> vpn-policy-routing 0.0.2-35 monitoring interfaces: iced [✓]

---
`iptables -t mangle -S VPR_PREROUTING` produces

```
-N VPR_PREROUTING
-A VPR_PREROUTING -d 172.20.112.0/24 -m comment --comment router-iced -j MARK --set-xmark 0x30000/0xff0000
-A VPR_PREROUTING -m set --match-set iced dst -j MARK --set-xmark 0x30000/0xff0000
```
___
`luci/admin/status/iptables#rule_mangle_VPR_PREROUTING` is still showing, and what confuses me but perhaps is to be expected?

| Target | Prot. | In | Out | Source    | Destination     | Options                                       |
|--------|-------|----|-----|-----------|-----------------|-----------------------------------------------|
| MARK   | all   | *  | *   | 0.0.0.0/0 | 172.20.112.0/24 | /* router-iced */ MARK xset 0x30000/0xff0000  |
| MARK   | all   | *  | *   | 0.0.0.0/0 | 0.0.0.0/0       | match-set iced dst MARK xset 0x30000/0xff0000 |

n8v8R, 2019-01-07:
closing this issue since it is working now and the double pit rule is expected or is this double ipt rule still something you want to look into?

stangri, 2019-01-07:
Definitely not expected. I'm struggling to understand where it comes from tho.

n8v8R, 2019-01-07:
Yeah, it was driving me bonkers to find the logic causing this... 

Commenting lines 573 - 575 just generates

| Target | Prot. | In | Out | Source    | Destination     | Options                                       |
|--------|-------|----|-----|-----------|-----------------|-----------------------------------------------|
| MARK   | all   | *  | *   | 0.0.0.0/0 | 0.0.0.0/0       | match-set iced dst MARK xset 0x30000/0xff0000 |

n8v8R, 2019-01-07:
I start wondering whether it is caused by netifd being version 2016-12-23 in TOS 3.11.x versus version 2018-11-19 in OWRT 18.06 and the related scripts.

stangri, 2019-01-07:
> The routing appears to be working correctly/as expected 👍
> 
> Target	Prot.	In	Out	Source	Destination	Options
> MARK	all	*	*	0.0.0.0/0	172.20.112.0/24	/* router-iced */ MARK xset 0x30000/0xff0000
> MARK	all	*	*	0.0.0.0/0	0.0.0.0/0	match-set iced dst MARK xset 0x30000/0xff0000

Ah, right, sorry, the output you've posted is expected! Sorry!

n8v8R, 2019-01-07:
Oh, I see. Thank you for having fixed the issue! :bowing_woman:

closed 2019-01-07 by n8v8R