mossdef-org/source.mossdef.org #43: [VPR 2-33 regression bug] iface name not recognized

n8v8R, 2018-12-22:
@stangri 

TOS 3.11.1 | vpn-policy-routing 0.0.2-33 | luci git-18.328.59464-9636605-1

{"kernel":"4.4.167-4a7a81f8db0ad743e54c68e1845c60b6-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu/generic","description":"OpenWrt omnia 15.05"}}

---

started to happen after upgrade to 0.0.2-33

`ERROR: policy 'router-hpot' has an unknown interface: iced!`

```
source /lib/functions.sh
packageName="vpn-policy-routing"
config_load "$packageName"
config_get supportedIfaces          'config' 'supported_interface'
echo "$supportedIfaces"
process_interface(){ echo "$1"; }
config_load 'network'; config_foreach process_interface 'interface' 'create';
```

produces

> loopback
> lan
> wan
> wan6
> wg0
> guest
> iced

---

> default via xx46.104.107 dev pppoe-wan  proto static
> 10.168.112.0/24 dev br-guest  proto kernel  scope link  src 10.168.112.12
> xx.46.104.107 dev pppoe-wan  proto kernel  scope link  src xx.46.10.175
> 172.20.112.0/24 dev br-iced  proto kernel  scope link  src 172.20.112.12
> 179.43.174.98 via 84.46.104.107 dev pppoe-wan  proto static
> 192.168.112.0/24 dev br-lan  proto kernel  scope link  src 192.168.112.12

> br-iced   Link encap:Ethernet  HWaddr D8:58:D7:00:87:4B
>           inet addr:172.20.112.12  Bcast:172.20.112.255  Mask:255.255.255.0
>           UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1

---
```
config policy
	option name 'router-hpot'
	option remote_ports '56007 2222'
	option interface 'iced'
	option remote_addresses '172.20.112.190'
```

stangri, 2018-12-23:
Doesn't seem that 'iced' is listed in the supported_interfaces in the config. Otherwise the output of the commands would have had 'iced' before 'loopback'.

n8v8R, 2018-12-23:
[image]

[image]

n8v8R, 2018-12-23:
The output (order) is similar to https://github.com/stangri/openwrt_packages/issues/42#issuecomment-448756930

stangri, 2018-12-23:
So the
```sh
source /lib/functions.sh
packageName="vpn-policy-routing"
config_load "$packageName"
config_get supportedIfaces          'config' 'supported_interface'
echo "$supportedIfaces"
```

doesn't produce anything?

n8v8R, 2018-12-23:
[image]

stangri, 2018-12-23:
I cannot reproduce your issue, I also have an explicitly allowed interface in supported_interface, but it works for me, so I can't figure out what part of the code may be failing.

If you just comment out lines 371-374 in /etc/init.d/vpn-policy-routing, does it process the policy?

stangri, 2018-12-23:
Also, please post the output of:
```sh
source /lib/functions.sh
packageName="vpn-policy-routing"
config_load "$packageName"
config_get supportedIfaces          'config' 'supported_interface'
	if [ -n "$supportedIfaces" ]; then
		for i in $supportedIfaces; do temp="$temp/$i/"; done
		supportedIfaces="$temp"
	fi
is_l2tp() { local proto; proto=$(uci -q get network."$1".proto); [ "${proto:0:4}" = "l2tp" ]; }
is_oc() { local proto; proto=$(uci -q get network."$1".proto); [ "${proto:0:11}" = "openconnect" ]; }
is_ovpn() { local dev; dev=$(uci -q get network."$1".ifname); [ "${dev:0:3}" = "tun" ] || [ "${dev:0:3}" = "tap" ] || [ -f "/sys/devices/virtual/net/${dev}/tun_flags" ]; }
is_pptp() { local proto; proto=$(uci -q get network."$1".proto); [ "${proto:0:4}" = "pptp" ]; }
is_tor() { local dev; dev=$(uci -q get network."$1".ifname); [ "${dev:0:3}" = "tor" ]; }
is_wg() { local proto; proto=$(uci -q get network."$1".proto); [ "${proto:0:9}" = "wireguard" ]; }
is_tunnel() { is_l2tp "$1" || is_oc "$1" || is_ovpn "$1" || is_pptp "$1" || is_tor "$1" || is_wg "$1"; }
is_wan() { [ -n "$wanIface4" ] && [ "$1" = "$wanIface4" ]; }
is_wan6() { [ -n "$wanIface6" ] && [ "$1" = "$wanIface6" ]; }
is_ignored_interface() { [ "${ignoredIfaces/"/$1/"}" != "$ignoredIfaces" ]; }
is_supported_interface() { [ "${supportedIfaces/"/$1/"}" != "$supportedIfaces" ] || { ! is_ignored_interface "$1" && { is_wan "$1" || is_wan6 "$1" || is_tunnel "$1"; }; }; }
is_supported_interface iced && echo "iced interface is supported"

stangri, 2018-12-23:
And the output of the "/etc/init.d/vpn-policy-routing status" please.

n8v8R, 2018-12-23:
> If you just comment out lines 371-374 in /etc/init.d/vpn-policy-routing, does it process the policy?

[image]

n8v8R, 2018-12-23:
> Also, please post the output of:

Uncommented 371 - 374

[image]

n8v8R, 2018-12-23:
> And the output of the "/etc/init.d/vpn-policy-routing status" please.

[image]

stangri, 2018-12-23:
I'm guessing the shell I'm using supports different string manipulation than what's default in 15.05. Please test both luci and main app of revision 34.

n8v8R, 2018-12-23:
Right on the mark, yet another TOS special/suffering... Thank you again for making the extra effort!
Another app developer made it clear what he thinks of TOS in its current state

https://forum.openwrt.org/t/travelmate-support-thread/5155/139

> Travelmate doesn't support old CC stuff - please wait until Turris 4.0 has been released

rev -34 does not exhibit the issue and thus closing. :mrs_claus:

closed 2018-12-23 by n8v8R