mossdef-org/source.mossdef.org #44: [VPR  bug] fails to detect a firewall restart

n8v8R, 2018-12-25:
@stangri 

Since recently VPR is not picking up a firewall restart which flushes all rules and thus any policy based connectivity ceases until VPR is manually restarted.

Had a look at the TOS code base to see if anything changed in their firewall init script between the last releases but it has not. It seems thus likely being a regression bug introduced with 0.0.2-34.

Please let me know if there is anything to debug the issue.
___
TOS 3.11.1 | vpn-policy-routing 0.0.2-34 | firewall 2018-06-13

{"kernel":"4.4.167-4a7a81f8db0ad743e54c68e1845c60b6-1","hostname":"router","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}

n8v8R, 2019-01-02:
Suppose this is just another case of the current TOS3.x branch lacking `procd ` support and a firewall restart not leaving a trace in the system log (/tmp/log/messages) and thus nothing for VPR to detect a firewall restart?

On the other hand the init script seems void of a procd restart trigger (only `procd_add_reload_trigger`)

stangri, 2019-01-08:
reload trigger is activated when the config file change causes the monitored service to be reloaded. I'm not aware of any hook options for when the service (firewall) is manually restarted.

VPR reload is also triggered on any supported interface up/down event tho.

n8v8R, 2019-01-08:
reload is working fine, it just been intriguing that all VPR connectivity got lost and not re-established with firewall restart.

the firewall unit script has a QUIET variable which at least during boot is exercised and thus not leaving a trace to pick upon.

```
boot() {
	# Be silent on boot, firewall might be started by hotplug already,
	# so don't complain in syslog.
	QUIET=-q
	start
}
```

Anyway, is probably just me conducting firewall restart exercises every now and then and not worth bothering further. Least I am aware of it now and know what to do.

closed 2019-01-08 by n8v8R