mossdef-org/source.mossdef.org #45: [VPR feature?] Virtual Routing and Forwarding (VRF)

n8v8R, 2019-01-04:
Whilst reading on https://www.kernel.org/doc/Documentation/networking/vrf.txt it occurred to me that it could potentially make sense for VPR to use and thus be liberated of any nf/ipt/nft/ipset bloating/complications? Even running a VPN server/client in a lxc container routing via/to/from via such container https://github.com/stangri/openwrt_packages/issues/28 sounds feasible?

stangri, 2022-09-29:
I see that VRF support has been added to OpenWrt: https://github.com/torvalds/linux/blob/master/drivers/net/Kconfig#L430 however I'm not sure if it's enabled/supported by default or needs a specially built kernel. Have you tested the VRF support in OpenWrt?

stangri, 2023-01-14:
Closing due to lack of reply.

closed 2023-01-14 by stangri