mossdef-org/source.mossdef.org #47: [VPR question] tor routing

n8v8R, 2019-01-11:
Whilst the readme is bare of any hint about tor routing the init script code though has some related snippets in display and thus wondering whether/how routing via tor might work, considering tor is not providing an iface but a SOCKET5 proxy?

I am currently running a tor client node with username/password credentials and would like to route `onion.` centrally from the router rather with a proxy browser plugin for each client.
One other issue comes to mind - resolving tor domains, which currently is set in the browser plugin with `Send DNS through SOCKS5 proxy`
I am also looking whether that DNS can be handled centrally by the router's DNS server (`unbound`) but have not reached a solution yet.

stangri, 2019-01-11:
The Tor support is planned but not at all implemented. Whatever code is there is just a stub, so that if anyone wants to contribute, they can.

y-alexey, 2019-12-15:
Tor routing can be easily realized by port forwarding to 127.0.0.1:9040 in fw3. You just need criteria or ready ipset sets of tragets.
It's already work for me.

And I try to find ability to forward the same manner to VPN tunnel.

stangri, 2019-12-15:
> Tor routing can be easily realized by port forwarding to 127.0.0.1:9040 in fw3. You just need criteria or ready ipset sets of tragets.
> It's already work for me.
Do you mind sharing the iptables/fw3 entries you have to make VPR work with the TOR tunnel?

stangri, 2020-05-23:
Are you still willing to test this out? I've made some steps towards TOR support in recent builds.

h4de5, 2020-10-19:
@stangri I am here to test this.

my setup:
i installed tor and privoxy using this guide: https://www.lan-monitor.de/openwrt-tor-proxy-einrichten/

using an additional PAC file, the proxy is picked up automatically in windows.
still, some devices ignore those proxy settings. for them i would like to force them through a (currently not existing) tor interface.

let me know if I can help.

stangri, 2020-10-20:
> @stangri I am here to test this.
>
> my setup:
> i installed tor and privoxy using this guide: https://www.lan-monitor.de/openwrt-tor-proxy-einrichten/
>
> using an additional PAC file, the proxy is picked [...]

Installing a package from my repo would be a great start. 👍

h4de5, 2020-10-20:
Ok. Forgot to mention that: a private openvpn interface and the vpn-policy-routing extension is up and running without problems since ages ;)

stangri, 2021-02-11:
Should be working in 0.3 versions and up. Please reopen if it doesn't.

closed 2021-02-11 by stangri

h4de5, 2021-02-22:
@stangri  can you provide on how to do this?
my web proxy is listening on 8118 and forwards requests to the tor client on port 9050.

this is what vpr shows - in the interface dropdown I only got WAN and openvpn
[image]

ilshatms, 2021-11-13:
> Are you still willing to test this out? I've made some steps towards TOR support in recent builds.

Hey
still relevant? I use vpr version 0.3.4-8, I don't see support TOR

stangri, 2021-11-13:
> > Are you still willing to test this out? I've made some steps towards TOR support in recent builds.
> 
> @stangri Hey
> still relevant? I use vpr version 0.3.4-8, I don't see support TOR

How did you check the tor support?

h4de5, 2021-11-13:
maybe we got a different understanding on that feature. i would expect that a Tor i entry is showing up in the drop down next to where i can select the interface in each rule row. if it is supported differently i did not find out how yet.

ilshatms, 2021-11-14:
> How did you check the tor support?

@stangri,

I see the TOR interface, but when setting up policies by domain or port, I see service errors.

[image]
[image]