mossdef-org/source.mossdef.org #52: luci-app-advanced-reboot GUI inaccurate when boot partition modified from U-Boot

timodug, 2019-02-02:
When Linksys boot partition was modified from U-Boot (with setenv bootcmd ...; saveenv), the 'Current' and 'Alternative' partitions are swapped-out in the luci-app-advanced-reboot Luci GUI (and thus, do not depict reality).
Sometimes it is required to set the boot partition from U-Boot, for example when a partition does not have luci-app-advanced-reboot installed. In such a case, luci-app-advanced-reboot GUI should update its view.

stangri, 2019-02-02:
1. Which model(s) have you tested this on?
2. When you modify the boot partition from U-boot, then boot into partition containing the app (which displays swapped partitions) and then you do regular reboot -- which partition would you boot into? The "current" (as marked in the app) or the "alternative"?

timodug, 2019-02-02:
Model is EA3500.
In fact, the issue could be somewhere else; here is the story:
After having rebooted to the alternative partition (call it '1') with  luci-app-advanced-reboot, I was stuck there because it does not have luci-app-advanced-reboot (it cannot be installed, it's a 18.06.1 snapshot build where 'opkg install' fail for this package).
To go back to partition 2 (18.06.2), I had to change 'bootcmd' (and saveenv) from U-Boot. Fine...until I look at advanced-reboot GUI where the two partitions seem swapped. And then if I reboot, it falls-down to partition 1...
So advanced-reboot seems correct regarding to what OpenWrt states.
Finally, advanced-reboot gave me the solution to switch permanently to partition 2:
- (current permanent is '1')
- Reboot, interrupt U-Boot
- Change  bootcmd to boot partition 2; boot
- Use advanced-reboot to reboot (a second time) to the partition you want become permanent.
Now 'reboot' effectively brings back to the same partition every time.

Note that in the advanced-reboot GUI, the distinction between the two partitions is done using the kernel version; more info would help when they are running the same one.

stangri, 2019-02-02:
AFAIK, two commands are required to switch the active partition on that specific device. Maybe just running one of them is not good enough for the lasting effect.

This luci app can only query the ```fw_printenv``` command, obviously it is returning unexpected (for you) data.

PS. You can set active partition from CLI, no need to reboot to u-boot to do that.

timodug, 2019-02-02:
Thank you a lot stangri for all that useful info.

stangri, 2019-02-03:
Regarding "more info would help" -- if you know how to fetch it, please share.

closed 2019-02-03 by stangri