mossdef-org/source.mossdef.org #53: [VPR bug?] loss of connectivity after disabling VPR

n8v8R, 2019-02-10:
After disabling VPR via LuCI, and returning traffic to wan, there is no connectivity however. Restarting/reloading `network` a/o `firewall` does not remedy the issue but only rebooting the router.
___
* {"kernel":"4.4.172-7bc33afbb1b35f5830b2b1b42c9cd8a0-0","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","release":{"distribution":"OpenWrt","version":"Chaos Calmer","revision":"r47055","codename":"omnia","target":"mvebu\/generic","description":"OpenWrt omnia 15.05"}}
* TOS 3.11.3 RC3
* vpn-policy-routing 0.0.3-3 
* luci-app-vpn-policy-routing git-19.035.52233-21b2..-36
* VPN 0.0.20190123
___
```
config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	option proto_control '1'
	option chain_control '1'
	option enabled '0'

config policy
	option interface 'wan'
	option name 'oc-ssh'
	option remote_addresses 'x.x.174.98'
	option remote_ports '56009'
	option chain 'PREROUTING'
	option proto 'tcp udp'

config policy
	option interface 'wan'
	option name 'oc-vps-cp'
	option remote_addresses 'foo.bar'
	option chain 'PREROUTING'
	option proto 'tcp udp'

config policy
	option interface 'wan'
	option name 'oc-vnc'
	option remote_addresses 'x.x.152.30'
	option remote_ports '5903'
	option chain 'PREROUTING'
	option proto 'tcp udp'

config policy
	option name 'lan'
	option local_addresses '192.168.84.0/24'
	option interface 'wg0'
	option chain 'PREROUTING'
	option proto 'tcp udp'
```

stangri, 2019-02-11:
I'll test on the mainline 18.06.1/2 when I get home but if I cannot reproduce that and it's TOS-specific I have no idea how to troubleshoot this.

n8v8R, 2019-02-11:
Sure, thank you for the effort.

I currently do not have an idea of how to debug it either but it had not been an issue previously, whereby previously could be a previous version of VPR or some previous settings on the router or a previous version of TOS. Which I concede are too many variables really.

And if it turns out to be another TOS3.x only issue then I can understand this becoming a bit old since you already made tremendous efforts to backport, which are developers are shy of.

The TOS developers are working on the next TOS4.x branch, which will be based on the then current OpenWRT branch and things should be simpler then. However, they are currently understaffed and TOS4.x development pace is clearly suffering.

n8v8R, 2019-02-12:
Just updated to

* vpn-policy-routing/stangri_repo/0.0.4-0
* luci-app-vpn-policy-routing/stangri_repo/git-19.042.42292-b7fabbd-37

and that seems to have cured the issue.

closed 2019-02-12 by n8v8R