mossdef-org/source.mossdef.org #55: [VPR feature suggestion] mac source based routing

n8v8R, 2019-02-11:
For single devices to be routed it currently requires to assign them a static ip. With mac source based routing those devices could be assigned a dynamic ip instead.

Not sure but thus far my understanding of `iptables` is such would be feasible with something like

`iptables -t nat -A PREROUTING -m mac --mac-source`

Also not certain whether additional kernel modules would be need, e.g. `iptables-mod-dhcpmac` & `kmod-ipt-dhcpmac`

stangri, 2019-07-24:
Implemented in vpn-policy-routing 0.0.7-1. Use colons in mac address. I didn't add any additional kmods and iptables didn't produce any errors when I included the mac address as the "local address".

Please test it, so I would cover it in README.

n8v8R, 2019-07-24:
Thank you for the effort. Unfortunately it does not seems to work though - it looks like that there is no ipt rule

```
config policy
	option chain 'PREROUTING'
	option interface 'wan'
	option name 'stv'
	option proto 'tcp udp'
	option local_address 'C0:97:27:40:84:36'
```

log [vpr log.txt](https://github.com/stangri/openwrt_packages/files/3425542/vpr.log.txt)
____
This been tested on 19.07 (TOS 5.x)

> {"kernel":"4.14.133","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"cznic,turris-omnia","release":{"distribution":"TurrisOS","version":"5.0-dev","revision":"6223df8","target":"mvebu/cortexa9","description":"TurrisOS 5.0-dev 6223df8"}}

with kernel conf (mac related) [kernel conf.txt](https://github.com/stangri/openwrt_packages/files/3425553/kernel.conf.txt)

`ls /lib/modules/$(uname -r)/ | grep mac`

> cmac.ko
> hmac.ko
> ip_set_bitmap_ipmac.ko
> ip_set_hash_mac.ko
> mac80211.ko
> xt_mac.ko

stangri, 2019-07-24:
Argh, my bad, fixed in 0.0.7-3.

I've rewritten a bunch of code to do with iptables entires, so there may be rough edges, but with my limited set of policies everything seems to work.

n8v8R, 2019-07-24:
Works now :1st_place_medal: 

> -A VPR_PREROUTING -p udp -m mac --mac-source C0:97:27:40:84:36 -m comment --comment stv -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p tcp -m mac --mac-source C0:97:27:40:84:36 -m comment --comment stv -c 0 0 -j MARK --set-xmark 0x10000/0xff0000

stangri, 2019-07-24:
Can you please confirm it's actually working? I've had a similar issue with the logical device/interface support where the rules would create properly, but it wouldn't affect the routing.
Thanks!

n8v8R, 2019-07-25:
Works fine on this node/setup. Seems you have implement counters on the mac ipt rules (for debugging?) and they are showing 

> -A VPR_PREROUTING -p udp -m mac --mac-source 94:65:2D:7D:DE:24 -m comment --comment android -c 70 3952 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p tcp -m mac --mac-source 94:65:2D:7D:DE:24 -m comment --comment android -c 425 73912 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p udp -m mac --mac-source 3E:10:D5:86:9F:67 -m comment --comment dvb-c -c 4 608 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p tcp -m mac --mac-source 3E:10:D5:86:9F:67 -m comment --comment dvb-c -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p udp -m mac --mac-source 3E:10:D5:9F:1F:89 -m comment --comment 1750e -c 4 608 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p tcp -m mac --mac-source 3E:10:D5:9F:1F:89 -m comment --comment 1750e -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p udp -m mac --mac-source C0:97:27:40:84:36 -m comment --comment stv -c 238 44571 -j MARK --set-xmark 0x10000/0xff0000
> -A VPR_PREROUTING -p tcp -m mac --mac-source C0:97:27:40:84:36 -m comment --comment stv -c 5096 381697 -j MARK --set-xmark 0x10000/0xff0000

Test various LAN clients and all are being routed as expected.

I would not reckon that by chance your rules weighing/order got mixed up (pardon for asking :woman_facepalming: )

This node's conf

```
config policy
	option chain 'PREROUTING'
	option interface 'wan'
	option name 'stv'
	option proto 'tcp udp'
	option local_address 'C0:97:27:40:84:36'

config policy
	option chain 'PREROUTING'
	option interface 'wan'
	option name '1750e'
	option proto 'tcp udp'
	option local_address '3E:10:D5:9F:1F:89'

config policy
	option chain 'PREROUTING'
	option interface 'wan'
	option name 'dvb-c'
	option proto 'tcp udp'
	option local_address '3E:10:D5:86:9F:67'

config policy
	option chain 'PREROUTING'
	option interface 'wan'
	option name 'android'
	option local_address '94:65:2D:7D:DE:24'
	option proto 'tcp udp'

config policy
	option local_address '192.168.84.0/24'
	option interface 'wg0'
	option chain 'PREROUTING'
	option proto 'tcp udp'
	option name 'to.lan'

config vpn-policy-routing 'config'
	option verbosity '2'
	option ipset_enabled '1'
	option dnsmasq_enabled '0'
	option strict_enforcement '1'
	option proto_control '1'
	option chain_control '1'
	option ipv6_enabled '0'
	option enabled '1'
```

stangri, 2019-07-25:
The order should be the same as before. Why?

n8v8R, 2019-07-25:
Probably a misunderstanding her - with the aforementioned setting on this node mac source routing is working as expected.

I was referring to a (potential/accidental) order mixup  concerning the issue you mentioned, that rules been created but the actual routing did not work.

closed 2019-07-25 by stangri