mossdef-org/source.mossdef.org #56: [VPR Bug?] No WG Traffic

euphoria360, 2019-03-01:
### Info and needed configs and logs

**Device:** _TP-Link WR-842ND_
**OS:** _OpenWrt 18.06.2, r7676-cddd7b4c77_

```
cat /etc/config/vpn-policy-routing

config vpn-policy-routing 'config'
        option ipv6_enabled '0'
        option strict_enforcement '1'
        option dnsmasq_enabled '1'
        option verbosity '2'
        option enabled '1'

config policy
        option chain 'PREROUTING'
        option name 'news'
        option remote_address 'bbc.com bbc.co.uk bbci.co.uk arstechnica.net arstechnica.com'
        option proto 'tcp udp'
        option interface 'wg0'

config policy
        option chain 'PREROUTING'
        option name 'other'
        option remote_address 'myip.com'
        option interface 'wg0'
        option proto 'tcp udp'
```

```
/etc/init.d/vpn-policy-routing status
vpn-policy-routing 0.0.4-1 running on OpenWrt 18.06.2. WAN (IPv4): wan/dev/192.168.49.1.
============================================================
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         192.168.49.1    0.0.0.0         UG    0      0        0 eth0
IPv4 Table 201: default via 192.168.49.1 dev eth0
IPv4 Table 201 Rules:
32765:  from all fwmark 0x10000 lookup 201
IPv4 Table 202: default via 10.180.83.84 dev wg0
IPv4 Table 202 Rules:
32764:  from all fwmark 0x20000 lookup 202
============================================================
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -m set --match-set wg0 dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IP Tables FORWARD
-N VPR_FORWARD
============================================================
IP Tables INPUT
-N VPR_INPUT
============================================================
IP Tables OUTPUT
-N VPR_OUTPUT
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create wg0 hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]
```

```
/etc/init.d/vpn-policy-routing reload
Creating table 'wan/192.168.49.1' [✓]
Creating table 'wg0/10.180.83.84' [✓]
Routing 'news' via wg0 [✓]
Routing 'other' via wg0 [✓]
vpn-policy-routing 0.0.4-1 started on wan/192.168.49.1 wg0/10.180.83.84 [✓]
vpn-policy-routing 0.0.4-1 monitoring interfaces: wan wg0 [✓]
```

### Issue Description
Hi and thanks for this great package. I've been using this on Openwrt 18.6.1 for few months and it was mostly ok. yesterday I clean installed OpenWRT 18.6.2 and tried whole day to make it work, but no luck!
not all intended Traffic is routed to wireguard interface, especially level 3 domains that are CNAME of another domain.

Wireguard Interface is set up on wg0 and working.
```
wg show all
interface: wg0
  public key: a2svsHWlAUsomegibberrishZWqz//hSU=
  private key: (hidden)
  listening port: 57259

peer: EcxHFjc20blahblahblahblahblahblahLOUIgI=
  endpoint: 190.2.141.163:51840
  allowed ips: 0.0.0.0/0
  latest handshake: 1 minute, 43 seconds ago
  transfer: 3.32 KiB received, 5.45 KiB sent
  persistent keepalive: every 25 seconds
```
here is the ping through wg0 interface:
```
ping myip.com -I wg0
PING myip.com (104.31.67.68): 56 data bytes
64 bytes from 104.31.67.68: seq=0 ttl=58 time=122.680 ms
64 bytes from 104.31.67.68: seq=1 ttl=58 time=124.160 ms
64 bytes from 104.31.67.68: seq=2 ttl=58 time=122.523 ms
64 bytes from 104.31.67.68: seq=3 ttl=58 time=121.246 ms
```

In VPR  have redirected **arstechnica.com** and **arstechnica.net** to wg0.
When I traceroute **arstechnica.com** on a PC in LAN, everything is ok:

```
tracert arstechnica.com

Tracing route to arstechnica.com [50.31.169.131]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  abyss-r.local [192.168.48.1]
  2   119 ms   119 ms   122 ms  190.2.141.163
  3   117 ms   119 ms   119 ms  172.16.0.1
  4   120 ms   124 ms   119 ms  190.2.141.3
  5   121 ms   119 ms   118 ms  109.236.95.184
  6   123 ms   120 ms   119 ms  be4381.rcr21.rtm01.atlas.cogentco.com [149.6.110.89]
  7   121 ms   129 ms   120 ms  be3384.ccr41.ams03.atlas.cogentco.com [154.54.58.165]
  8   122 ms   122 ms   127 ms  be3499.rcr22.ams05.atlas.cogentco.com [154.54.60.22]
  9   124 ms   126 ms   134 ms  level3.fra06.atlas.cogentco.com [130.117.15.194]
 10   214 ms   218 ms   217 ms  xe-4-2-2.cr2-chi1.ip4.gtt.net [89.149.143.97]
 11   217 ms   216 ms   233 ms  as23352.chi12.ip4.gtt.net [199.229.229.214]
 12   215 ms   216 ms   216 ms  0.ae4.cr1.ord6.scnet.net [204.93.204.85]
 13   223 ms   216 ms   216 ms  0.ae1.ar3.ord6.scnet.net [204.93.204.109]
 14   223 ms   242 ms   234 ms  vlan-41.aggrdl114-1.ord6.scnet.net [167.88.157.25]
 15   216 ms   216 ms   217 ms  ge-11-2-1.ar9.ord6.us.scnet.net [50.31.169.130]
 16   217 ms   217 ms   218 ms  ge-11-2-1.ar9.ord6.us.scnet.net [50.31.169.130]
 17   217 ms   219 ms   220 ms  ge-11-2-1.ar10.ord6.us.scnet.net [50.31.169.131]
```
as you can see 2nd hop is my WG endpoint. 

But if I traceroute **cdn.arstechnica.net**:
```
tracert cdn.arstechnica.net

Tracing route to vip1.g5.cachefly.net [205.234.175.175]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  abyss-r.local [192.168.48.1]
  2     1 ms    <1 ms     1 ms  192.168.49.1
  3    36 ms    31 ms    31 ms  94-183-140-6.shatel.ir [94.183.140.6]
  4    31 ms    31 ms    31 ms  94-183-140-1.shatel.ir [94.183.140.1]
  5    35 ms    34 ms    34 ms  172.18.74.113
  6    44 ms    36 ms    36 ms  172.18.218.145
  7    42 ms    37 ms    36 ms  172.18.196.65
  8    41 ms    36 ms    36 ms  172.18.196.90
  9    43 ms    39 ms    43 ms  10.202.4.132
 10    37 ms    37 ms    37 ms  10.21.211.20
 11   110 ms   109 ms   108 ms  et-5-0-1-0.ffttr6.frankfurt.opentransit.net [193.251.154.203]
 12   111 ms   108 ms   109 ms  et-14-0-7-0.ffttr7.frankfurt.opentransit.net [193.251.132.163]
 13   114 ms   112 ms   112 ms  be5511.agr41.fra03.atlas.cogentco.com [130.117.14.225]
 14     *        *        *     Request timed out.
 15   110 ms   110 ms   110 ms  vip1.g-anycast1.cachefly.net [205.234.175.175]
```
2nd hop is my upstream router, not WG endpoint. So no cdn.arstechnica.net content is being redirected to wg0. Can someone tell me where the problem is?

stangri, 2019-03-06:
Is the paste above the full output of `service vpn-policy-routing status`? It should list a section of `DNSMASQ ipsets`.

And just to confirm -- have you tried rebooting the testing device or otherwise purging the DNS cache on it?

euphoria360, 2019-03-07:
@stangri 
yes, it was the full output of service `vpn-policy-routing` status.
This is the current output (I've added abit more rules since then):

```
root@abyss-r:~# service vpn-policy-routing restart
vpn-policy-routing 0.0.4-1 stopped [✓]
Creating table 'wan/192.168.49.1' [✓]
Creating table 'wg0/10.180.83.84' [✓]
Routing 'lan' via wg0 [✓]
Routing 'telegram' via wg0 [✓]
Routing 'media' via wg0 [✓]
Routing 'news' via wg0 [✓]
Routing 'social' via wg0 [✓]
Routing 'torrent1' via wg0 [✓]
Routing 'torrent2' via wg0 [✓]
Routing 'other' via wg0 [✓]
vpn-policy-routing 0.0.4-1 started on wan/192.168.49.1 wg0/10.180.83.84 [✓]
vpn-policy-routing 0.0.4-1 monitoring interfaces: wan wg0 [✓]
root@abyss-r:~# service vpn-policy-routing status
vpn-policy-routing 0.0.4-1 running on OpenWrt 18.06.2. WAN (IPv4): wan/dev/192.168.49.1.
============================================================
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         192.168.49.1    0.0.0.0         UG    0      0        0 eth0
IPv4 Table 201: default via 192.168.49.1 dev eth0
IPv4 Table 201 Rules:
32747:  from all fwmark 0x10000 lookup 201
IPv4 Table 202: default via 10.180.83.84 dev wg0
IPv4 Table 202 Rules:
32746:  from all fwmark 0x20000 lookup 202
============================================================
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -d 151.101.60.193/32 -m comment --comment other_imgur_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 151.101.60.193/32 -m comment --comment other_imgur_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 52.205.36.130/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 52.205.36.130/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 52.86.8.163/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 52.86.8.163/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 34.232.230.241/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 34.232.230.241/32 -m comment --comment other_docker_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 108.168.194.93/32 -m comment --comment other_bintray_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 108.168.194.93/32 -m comment --comment other_bintray_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 205.185.216.10/32 -m comment --comment other_netdna-ssl_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 205.185.216.10/32 -m comment --comment other_netdna-ssl_com -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 185.178.208.182/32 -m comment --comment torrent2_nyaa_si -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 185.178.208.182/32 -m comment --comment torrent2_nyaa_si -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.47.104/32 -m comment --comment torrent2_eztv_io -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.47.104/32 -m comment --comment torrent2_eztv_io -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.48.104/32 -m comment --comment torrent2_eztv_io -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.48.104/32 -m comment --comment torrent2_eztv_io -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.108.19/32 -m comment --comment torrent2_ettv_tv -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.108.19/32 -m comment --comment torrent2_ettv_tv -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.109.19/32 -m comment --comment torrent2_ettv_tv -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.25.109.19/32 -m comment --comment torrent2_ettv_tv -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.71.121/32 -m comment --comment torrent2_btbit_org -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.71.121/32 -m comment --comment torrent2_btbit_org -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.70.121/32 -m comment --comment torrent2_btbit_org -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.70.121/32 -m comment --comment torrent2_btbit_org -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.17.3/32 -m comment --comment torrent2_1337x_to -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.17.3/32 -m comment --comment torrent2_1337x_to -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.16.3/32 -m comment --comment torrent2_1337x_to -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.31.16.3/32 -m comment --comment torrent2_1337x_to -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.27.212.30/32 -m comment --comment torrent2_torrentdownloads_me -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -d 104.27.212.30/32 -m comment --comment torrent2_torrentdownloads_me -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
[... cut ...]
I guess I found where the issue is:
If I add an entry into `vpn-policy-routing` LuCi page and **set the port to TCP**, upon service reload, it will be added into **DNSMASQ ipsets** section in `service vpn-policy-routing status`. Also it will be apllied much faster and subdomains are working.

But when I set the port to **UDP** or **TCP/UDP**, no. It will not get added to **DNSMASQ ipsets** section in `service vpn-policy-routing status`. It will be slow, and sometimes with errors. and no subdomain works.

stangri, 2019-03-07:
AFAIK HTTP is a TCP service, why would you want to set the domain-only policy to TCP/UDP?

euphoria360, 2019-03-08:
@stangri, lots of my tunneled services are websites/HTTP. but not all.
Torrent trackers use udp mostly. And seems some google services use QUIC which is based on UDP (That's gonna become HTTP/3 probably). Telegram and Skype use UDP for VoIP.

And unfortunately, something in each of those examples above need to be tunnelled by me

stangri, 2019-03-08:
WON'T FIX -- DNSMASQ's ipset matching can only work for TCP packets, that's the iptables limitation.

"Torrent trackers use udp mostly." - not for the web-sites and you still need a UDP rule for the whole internet for device running a torrent client.
" google services use QUIC which is based on UDP " -- set up UDP port forwarding for port 80.
"Telegram and Skype use UDP for VoIP." -- set the DSCP tags for these apps.

The only alternative would be to route all traffic thru the tunnel (and exclude as needed) route or all traffic from specific client thru the tunnel.

This is a feature, not a bug, for additional discussion please visit the forum.

stangri, 2019-09-18:
I may have read an article which led me to believe DNSMASQ ipset only supports TCP incorrectly. Currently ipsets should be protocol agnostic, but I'll be modifying that behaviour soon.

stangri, 2019-09-27:
Please test the updated version.

closed 2019-09-27 by stangri