mossdef-org/source.mossdef.org #57: [VPR feature suggestion] Rule Enable/Disable Option

simonbytes, 2019-03-14:
It would be handy to add an Enable option to rules in the list. Like in openwrt firewall rules list.

This picture of your openwrt_package shows it does only give the possibility to delete a rule to temporarily disable it. Even a moved rule would be readed obviosly.
[image]

n8v8R, 2019-03-16:
dup https://github.com/stangri/openwrt_packages/issues/38

simonbytes, 2019-03-17:
Well, if the width of the luci screen is important, I will try to look for an alternate way to make disabling a rule possible and report back.

closed 2019-04-04 by stangri

stangri, 2019-07-26:
@simonbytes implemented in vpn-policy-routing 0.0.7-5 and luci-app-vpn-policy-routing 44. Check the README for more info.