mossdef-org/source.mossdef.org #61: Installing as per Readme.md causes Luci Certificate Untrusted

GAndreone, 2019-04-12:
OpenWrt 18.06.2, r7676-cddd7b4c77

This error occurs after a power up/down of the router.

stangri, 2019-04-12:
Please post exact commands and the error.

GAndreone, 2019-04-12:
#### Causes Web certificate not trusted error##

# VPN Policy-Based Routing
opkg update; opkg install ipset resolveip ip-full kmod-ipt-ipset iptables
opkg update; opkg remove dnsmasq; opkg install dnsmasq-full

opkg update
opkg list-installed | grep -q uclient-fetch || opkg install uclient-fetch
opkg list-installed | grep -q libustream || opkg install libustream-mbedtls
echo -e -n 'untrusted comment: LEDE usign key of Stan Grishin\nRWR//HUXxMwMVnx7fESOKO7x8XoW4/dRidJPjt91hAAU2L59mYvHy0Fa\n' > /tmp/stangri-repo.pub && opkg-key add /tmp/stangri-repo.pub
! grep -q 'stangri_repo' /etc/opkg/customfeeds.conf && echo 'src/gz stangri_repo https://raw.githubusercontent.com/stangri/openwrt-repo/master' >> /etc/opkg/customfeeds.conf
opkg update
opkg install vpn-policy-routing luci-app-vpn-policy-routing
--------------------------------------------------------------------------------

# VPN Client & Server Simultaneously
if [ -s /etc/config/vpn-policy-routing ]; then
  uci set vpn-policy-routing.config.output_chain_enabled='1'
  uci add_list vpn-policy-routing.config.ignored_interface='vpnserver0'
  uci add vpn-policy-routing policy
  uci set vpn-policy-routing.@policy[-1]=policy
  uci set vpn-policy-routing.@policy[-1].comment='VPN Server'
  uci set vpn-policy-routing.@policy[-1].interface='wan'
  uci set vpn-policy-routing.@policy[-1].local_ports='1194'
  uci commit vpn-policy-routing
fi

# Create another firewall forwarding (in the code below replace the vpnclient with the firewall zone for your VPN client, refer to the tail of /etc/config/firewall):
uci add firewall forwarding
uci set firewall.@forwarding[-1].src='vpnserver0'
#uci set firewall.@forwarding[-1].dest='vpnclient'
uci set firewall.@forwarding[-1].dest='wan'
uci commit firewall

# Restart/reload the service:
service vpn-policy-routing reload

stangri, 2019-04-12:
As you can see, the code above has nothing to do with certificates.

GAndreone, 2019-04-12:
`You are absolutely correct! But the fact is that the certificate error only occurs after I follow the above procedure!  At the moment with out the procedure the router boots powers up without a problem.  Your code is great. Maybe it has to do with an older version of Luci you added??

GAndreone, 2019-04-12:
I rebuilt the system 2 times to prove it. It takes a long time for the rebuild.

stangri, 2019-04-12:
Maybe it has something to do with the routing/MITM attack and/or the way certificate was issued/used by the web-site.

GAndreone, 2019-04-12:
This router TL-WR1043NDv1.7 id behind DVA-5592 router supplied by Wind in Italy. The TL-WR1043ND is just used as a OpenVPN server & Client. The Client talks to a Router in Canada and I VPN from my system in Canada to the Router in Italy. The OpenWrt wan is DVA-5592 firewall.  The certificate it is complaining about is the CA of the OpenWrt router. It was created using Easy-RSA on the router.

stangri, 2019-04-12:
This issue is definitely not related to my package. Post on the forum if you need help with your manually generated certificate.

closed 2019-04-12 by stangri