mossdef-org/source.mossdef.org #62: [VPR change request] Use append instead of insert in OUTPUT chain

yitzhaq, 2019-04-18:
When rules are added, it seems this is done using insert instead of append. In the case of the root OUTPUT chain this is a bit problematic, since it specifies -g VPR_OUTPUT, which causes any already set rules to be ignored. This makes it quite hard to add things cleanly to the OUTPUT chain in other contexts, particularly in places like firewall.user, which gets loaded before VPR does. You also don't wanna add stuff to the VPR_OUTPUT chain, both because it doesn't organically exist before VPR is loaded, and because VPR shouldn't reasonably have to expect anything else to mess about in this chain.

It seems the most pragmatic solution to this would be to use -A instead of -I. Would that be possible here, or are there concerns I haven't considered?

stangri, 2019-04-19:
Good point, it used to be append, but I've needed to change it to insert because it was ignored in combination with some other package (I suspect mwan3).

Rather than append, I'd prefer to keep insert with jump, rather than goto. Would that address your concern?

yitzhaq, 2019-04-19:
As long as other rules in the chain don't end up getting skipped (I'd have to test), I think I'll be happy.

stangri, 2019-04-19:
Please test 0.0.4-5 from https://dev.melmac.net/openwrt-repo/

yitzhaq, 2019-04-19:
OMG melmac.net. A kindred spirit! Here's the last birthday cake I got - some might categorize me as a bit of a fanatic.
[image]

On a somewhat more related note, unfortunately this does not seem to work for me, though that might be due to idiosyncrasies with my setup. ICMP routes as expected (using rules set in firewall.user), but TCP doesn't work. That said, I haven't tested that the same wouldn't happen with append instead of insert too - perhaps it's a timing issue of some sort.

My ugly workaround, which does work, is cronjobs like this, running every minute (wrapped by idemptables so the rule list doesn't grow indefinitely):

iptables -t mangle -I OUTPUT -d 1.1.1.1 -m conntrack --ctstate NEW -j MARK --set-xmark 0x20000/0xff0000

(1.1.1.1 just as example)

My use case is to enable connection tracking just for router connections, so that outgoing traffic from the router is routed through a tunnel, while still enabling incoming connections from world on WAN. I don't want conntracking on all rules (as it breaks port forwarding through tunnels), only the OUTPUT rules, which is why I don't use your append settings. If it was possible to set custom appends on a per-rule (or possibly per-chain) basis, rather than globally, that would also probably solve my challenge, though I understand you're hesitant about web config width (FWIW I'm on a 12" display and already need to scroll, though I haven't given it a second thought).

If you want to do any testing yourself, or have clever suggestions for better ways to accomplish this, by all means. If not, my current workaround does work, so there's no need to bend over backwards to accommodate my particular use case. Due to the concerns I originally raised, your last change might still be worthwhile though - provided it doesn't cause general breakage, of course. From my limited testing it does seem to work well both on its own, and together with the cron-added rules.

yitzhaq, 2019-04-21:
You're right, my testing was flawed, sorry. I was testing with HTTPS, and the problem seems to be with the TLS handshaking session. HTTP and TCP itself appears fine, but the handshake issue happens regardless of where or when the output rules are added (and regardless of whether VPR rules are active), so I guess I have some more debugging to do.

This suggests your release should be good to go as far as I'm concerned, at least. Should you happen to have any helpful pointers for my (now unrelated) issue, I'd love to hear them.

Thanks again for the fix, and apologies for the confusion.

closed 2019-05-14 by stangri

stangri, 2019-07-26:
Just the heads up, I've changed the iptables rules options to APPEND by default in 0.0.7-7. Can be switched to INSERT with the ```iptables_rule_option``` option in config.

yitzhaq, 2019-09-01:
Thanks for the heads up, and sorry for the belated response. Just out of curiosity, what was the background for your decision to change the default?