mossdef-org/source.mossdef.org #63: luci-app-vpn-policy-routing page load error

sages, 2019-04-30:
Running latest snapshot build on ath79 wr842n-v3
Get the following error on the VPR luci page.

`Failed to execute cbi dispatcher target for entry '/admin/services/vpn-policy-routing'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/vpn-policy-routing.lua:98: attempt to concatenate local 'tmpfsVersion' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/vpn-policy-routing.lua:98: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:866: in function </usr/lib/lua/luci/dispatcher.lua:860>`

luci-app-vpn-policy-routing - git-19.115.41336-35ace20-40
vpn-policy-routing - 0.0.5-0
OpenWrt SNAPSHOT, r9911-fdf1aae77d

stangri, 2019-04-30:
Thank you for the report, fixed in git-19.120.24458-1b79446-41.

sages, 2019-04-30:
Installed update and appears to be working ok.

closed 2019-04-30 by stangri