mossdef-org/source.mossdef.org #64: [VPR bug] conflict with firewall at boot time - wrong routing

n8v8R, 2019-05-14:
{"kernel":"4.14.113","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"armada-385-turris-omnia","release":{"distribution":"TurrisOS","version":"4.0-beta1","revision":"0663455801","target":"mvebu\/cortexa9","description":"TurrisOS 4.0-beta1 0663455801"}}

- vpn-policy-routing 0.0.5-0
- luci-app-vpn-policy-routing git-19.120.25736-3b37..-41
___
https://forum.turris.cz/t/turris-os-4-0-beta1-is-released/10107/73
https://forum.turris.cz/t/turris-os-4-0-beta1-is-released/10107/75

upon booting the router - despite VBP up/running all/any traffic is routed via WAN.

What I discovered is that there is a conflict with firewall, that becomes apparent in the syslog timeline

> 14:52:33 to netifd: Network device 'wg0' link is up
> 14:52:33 to vpn-policy-routing [5674]: Creating table 'wan/xx.46.104.107/::/0' [✓]
> 14:52:34 to firewall: Reloading firewall due to ifup of wan (pppoe-wan)
> 14:52:34 to vpn-policy-routing [5674]: Creating table 'wg0/xx.46.104.107/fd30:d64c:1eed:4c3a::12 dev ' [✓]
> 14:52:34 to vpn-policy-routing [5674]: Routing 'oc-ssh' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing 'oc-vps-cp' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing 'oc-vnc' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing 'stv' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing '1750e' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing 'dvb-c' via wan [✓]
> 14:52:35 to vpn-policy-routing [5674]: Routing 'to.lan' via wg0 [✓]
> 14:52:35 to vpn-policy-routing [5674]: service started on wan/xx.46.104.107/::/0 wg0/xx.46.104.107/fd30:d64c:1eed:4c3a::12 dev  [✓]
> 14:52:36 to vpn-policy-routing [5674]: service monitoring interfaces: wan wg0 [✓]
> 14 14:52:37 to firewall: Reloading firewall due to ifup of wg0 (wg0)

There are 2 issues, VPR is generating the same route for wg0 as for wan and then also firewall reloads after VPR finished and deletes/wipes any routing tables generated by VPR -> https://github.com/stangri/openwrt_packages/issues/44
___
Restarting VPR manually restores the expected behaviour and respective fw rules

n8v8R, 2019-05-14:
For some reason this happens only at some boot times but not always.

stangri, 2019-05-14:
Thanks for creating the report. 

Can you please e-mail me the ifconfig output and post the PROCD level from the file /etc/init.d/firewall.

stangri, 2019-05-14:
For the restart on firewall reload, try this: save file from https://raw.githubusercontent.com/stangri/openwrt_packages/master/vpn-policy-routing/files/vpn-policy-routing.iface.hotplug as /etc/hotplug.d/iface/70-vpn-policy-routing

n8v8R, 2019-05-14:
> If you personally know anyone at Turris

Never met anyone in person, I am afraid, and the communication over the net is a bit curt. But I might ask them never the less.
___
> make sure that their flavour of OpenWrt is compatible with my packages

It should be getting easier since this trunk 4.0 beta 1 is based on OpenWRT 18.06 and TOS 5.x will be following on the heels of OpenWRT 19.x, whenever that might be

n8v8R, 2019-05-14:
let us close this for the time being as I cannot reproduce it consistently. I will keep an eye on it and reopen if necessary.

Moreover, firewall reload is not the issue, as mentioned in #44, but it was firewall restart which is not the case here.

N.B. Are you still getting messages from the Turris Forum? If yes, then I would contact you there about the response from TO about their inclination to sponsor a device for you.

closed 2019-05-14 by n8v8R

n8v8R, 2019-05-15:
Like said it does not happen consistently but  enclosed the unfettered boot log when it just happened again

[bootlog.zip](https://github.com/stangri/openwrt_packages/files/3182022/bootlog.zip)