mossdef-org/source.mossdef.org #66: [VPR question] How to make a large set of remote IPs/subnets as the remote_address of policy?

jeffpeng78, 2019-06-05:
I have a quite large set of IPs/subnets like following
```
43.227.8.0/22
43.227.32.0/22
43.227.36.0/22
43.227.40.0/22
43.227.44.0/22
43.227.48.0/22
43.227.52.0/22
43.227.56.0/22
43.227.60.0/22
..... Total 8365 records.
```

And I would like to make a policy based on this list as the remote_address.
What's the best way to do that?

As far as I know, ipset can handle this kind of big IP list.
Can I make an ipset from my IP list and make it the remote_address of my policy?

Thanks a lot for your help.

stangri, 2019-06-05:
https://github.com/stangri/openwrt_packages/blob/master/vpn-policy-routing/files/README.md#custom-user-files

closed 2019-07-24 by stangri

stangri, 2019-07-26:
Just a heads up, I've re-worked how the custom user files work, refer to the README link above.

dananichev, 2019-11-12:
@stangri i hate to bring this closed issue back to life but i have a question: does custom user files work only for WAN-specific routing? I tried to place `TARGET_IPSET='vpn'` into one of them but my traffic still goes trough WAN not VPN interface:

```bash
#!/bin/sh
# This file is heavily based on code from https://github.com/Xentrk/netflix-vpn-bypass/blob/master/IPSET_Netflix.sh

TARGET_IPSET='vpn'

TARGET_URL="https://antifilter.download/list/ipsum.lst"
TARGET_FNAME="/var/tmp_rkn_subnet"

curl "$TARGET_URL" --output "$TARGET_FNAME"
awk -v ipset="$TARGET_IPSET" '{print "add " ipset " " $1}' "$TARGET_FNAME" | ipset restore -!
rm -f "$TARGET_FNAME"

```

Thanks,
Dmitry

stangri, 2019-11-13:
If you're looking for help with your user script, post in the forum: https://forum.openwrt.org/t/vpn-policy-based-routing-web-ui-discussion/ and make sure to include required information: https://github.com/stangri/openwrt_packages/blob/master/vpn-policy-routing/files/README.md#getting-help