mossdef-org/source.mossdef.org #67: [VPR] boot time-out not working

n8v8R, 2019-06-15:
> {"kernel":"4.14.123","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"armada-385-turris-omnia","release":{"distribution":"TurrisOS","version":"4.0-beta3","revision":"3755d71","target":"mvebu\/cortexa9","description":"TurrisOS 4.0-beta3 3755d71"}}
____
```
config vpn-policy-routing 'config'
	option boot_timeout '90'
```
 From the log - the gap is only 6 seconds

> 2019-06-15T13:58:**12**+01:00 to kernel: [    0.000000] Booting Linux on physical CPU 0x0
> 2019-06-15T12:58:**18**+01:00 to vpn-policy-routing [6779]: service waiting for wan gateway...

If you take a closer look at the hour though there is a gap of one 1 hour - could that be the cause?

There seems to be an issue with busy box logger and syslog as mentioned in this downstream issue

https://gitlab.labs.nic.cz/turris/turris-os-packages/issues/100

stangri, 2019-06-15:
Boot time out is how long VPR waits to discover wan gateway from the time
it starts on boot.

n8v8R, 2019-06-15:
I see, a misconception on my end then.

> Time (in seconds) for service to wait for WAN gateway discovery on boot. 

For me it implied that VPR waits that amount of time till is starts WAN gateway discovery and not the time when it gives up.

To start WAN gateway discovery 6 seconds into the boot process seems a bit early, that with PROC S level 94. This node's PPPoE WAN gateway is usually not ready within 30 seconds from boot time 0.

closed 2019-06-15 by n8v8R