mossdef-org/source.mossdef.org #68: [info on repo] tls connectivity issues with DNS64 embedded ipv4 addresses

n8v8R, 2019-06-16:
This is for your info, and perhaps any other user that might experience the issue.

With IPv6 upstream connectivity getting updates from the repo fails with a resolver that deploys DNS64, likely caused by versions of libgnutls | libopenssl that having issues with DNS64 records.

DNS lookup for `raw.githubusercontent.com` reveals the absence of an AAAA record

>     ;; QUESTION SECTION:
>     ;raw.githubusercontent.com. IN AAAA
> 
>     ;; ANSWER SECTION:
>     raw.githubusercontent.com. 25 IN CNAME github.map.fastly.net.

And the DNS lookup of its `CNAME github.map.fastly.net` showing the same absence of an AAAA record.

Subsequent the resolver's DNS64 module correctly synthesis from the A record of `raw.githubusercontent.com` and produces an IPv4-embedded IPv6 address, which in a packet dump reads:

> Destination: 64:ff9b::9765:2485
> [Destination Embedded IPv4: 151.101.36.133]

running `gnutls-cli -d 4 -V raw.githubusercontent.com` reveals that ipv6 connectivity fails in the first instance and subsequent ipv4 connectivity producing a fatal error

>     Connecting to ‘64:ff9b::9765:7085:443’…
>     Connecting to ‘151.101.112.133:443’…
>     |<4>| HSK[0x14c2750]: CLIENT HELLO was queued [235 bytes]
>     |<2>| WRITE: -1 returned from 0x4, errno: 104
>     |<3>| ASSERT: buffers.c[errno_to_gerr]:228
>     |<3>| ASSERT: buffers.c[_gnutls_io_write_flush]:720
>     |<3>| ASSERT: handshake.c[handshake_client]:2817
>     *** Fatal error: Error in the push function. |<2>| WRITE: -1 returned from 0x4, errno: 32
>     |<3>| ASSERT: buffers.c[errno_to_gerr]:228
>     |<3>| ASSERT: buffers.c[_gnutls_io_write_flush]:720
>     |<3>| ASSERT: record.c[_gnutls_send_tlen_int]:554
>     Could not connect to 151.101.112.133:443: Socket not connected

Similar `openssl` fails to connect.

As a causality train packages from the repo cannot be updated in lieu of connectivity.

A bit odd in the first place that neither `raw.githubusercontent.com` or `github.map.fastly.net` feature AAAA records.

n8v8R, 2019-06-19:
@stangri 

it seems a bug of VPR confusing routing of upstream ipv4/6 connectivity for the router
___
> {"kernel":"4.14.127","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"armada-385-turris-omnia","release":{"distribution":"TurrisOS","version":"4.0-beta3","revision":"6ee650d","target":"mvebu\/cortexa9","description":"TurrisOS 4.0-beta3 6ee650d"}}

- `vpn-policy-routing` 0.0.6-0 with [vpn-policy-routing.txt](https://github.com/stangri/openwrt_packages/files/3304779/vpn-policy-routing.txt)   

- upstream ISP authentication PPPoE
- upstream ISP connectivity native IPv6 PD /56 range and DS-Lite for ipv4
- upstream VPN connectivity `wireguard` which provides _ipv4 only_
___
enabling VPR (takes a few seconds for propagating the routing tables/ip rules) and then from cli exec

```
curl -o /dev/null https://raw.githubusercontent.com/stangri/openwrt-repo/master/Packages.gz -S
v
``` 
produces

>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to raw.githubusercontent.com:443
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to raw.githubusercontent.com:443

for extended debugging then ran `openssl s_client -connect raw.githubusercontent.com:443 -status -debug -msg -nbio_test` with this output [openssl debug.log](https://github.com/stangri/openwrt_packages/files/3304831/openssl.debug.log)
___

Now, if stopping/disabling `VPR` these connectivity issues are not present and `curl` and `openssl` connect as expected.

Having repeatedly started/stopped `VPR` the result is reproduced each time.

The strange thing is that the tcp packets emitted by the router with VPR enabled are embedding the ipv4 address in the ipv6 address which emulates NAT64 behaviour

> Destination: 64:ff9b::9765:2485
> [Destination Embedded IPv4: 151.101.36.133]

stangri, 2019-09-11:
Just to confirm, does this only happen in the mixed (legacy/IPv6) environment?

n8v8R, 2019-09-20:
The ISP provides only two sets of connectivity (VDSL PPPoE)

1. sole ipv4 
2. ipv6 dhcp delegated prefix + ipv4 ds-lite

The issue is not present with 1 but present with 2, though I could not undertake tests with recent versions of VPBR since the remote endpoint is currently offline.

stangri, 2019-09-21:
Uhm, the mixed environment support is WIP, actually selectively routing IPv6 is no easy feat on OpenWrt at the moment. The consensus is that you need to configure NAT6/NPTv6 for it to work.

n8v8R, 2019-09-21:
Uhm, not sure if we got our wires crossed. With the mixed ipv4/6 scenario there is no issue unless VPR is enabled and no need for NAT6/NPTv6, as it should not be necessary. If ipv6 is not available it falls back to ipv4

stangri, 2019-09-22:
Is that with IPv6 enabled or disabled in vpr?

stangri, 2019-09-27:
Also, when you regain access to that system, could you please post the VPR config, as well as the outputs of ```reload``` and ```support```?

stangri, 2020-03-29:
Closing for lack of response.

closed 2020-03-29 by stangri

beatstick, 2024-02-11:
I can confirm this issue still exists with pbr on openwrt 23.02.

I recently changed my provider and now I only receive an ipv6, no ipv4. Wireguard is is running on ipv4. I enable the package ds-lite in openwrt and everything works. But pbr does not start. It gives the error:

"service failed to discover WAN gateway!"

I used this hacky workaround which made it start up:

edit: /etc/init.d/vpn-policy-routing ( insert around line 222 )
 
``
is_wan_up() {
	local sleepCount=1

       #ADD-THE-LINE-BELOW-HERE
       wanGW="192.168.1.1"

	while [ -z "$wanGW" ] ; do
``

Now PBR starts without errors, but it does not work. There is no more Internet connectivity. 

This is from my pvr config:
``
config pbr 'config'
	option verbosity '2'
	option strict_enforcement '1'
	option src_ipset '0'
	list ignored_interface 'vpnserver wgserver'
	option boot_timeout '30'
	option iprule_enabled '0'
	option webui_protocol_column '0'
	option webui_chain_column '0'
	option webui_sorting '1'
	option webui_enable_column '1'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option dest_ipset '0'
	option webui_show_ignore_target '0'
	option ipv6_enabled '1'
	option enabled '1'
	option rule_create_option 'add'
	option resolver_set 'dnsmasq.nftset'
``

stangri, 2024-02-12:
> I can confirm this issue still exists with pbr on openwrt 23.02.

I don't think the issue created 4 years ago, for a drastically different version of OpenWrt and drastically different version of pbr (well, predecessor of pbr) affects you.

You actually have two issues:
1. pbr not starting properly on IPv6-only system
2. you changing part of the code and code not working properly after

While I can't help you with the second issue, I can see what can be done about the first one.

You'll need to check out Getting Help section of the `pbr` README and provide information requested there. You can either: * open a new issue with that information here (and please see how it needs to be formatted so it can be read) or 
* post the information requested in Getting Help section in the `pbr` thread on the OpenWrt forum.