mossdef-org/source.mossdef.org #69: [luci-app-vpn-policy-routing] columns too wide

n8v8R, 2019-07-25:
* 19.07 (TOS 5.x)

> {"kernel":"4.14.133","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"cznic,turris-omnia","release":{"distribution":"TurrisOS","version":"5.0-dev","revision":"6223df8","target":"mvebu/cortexa9","description":"TurrisOS 5.0-dev 6223df8"}}

*  luci-app-vpn-policy-routing git-19.204.23774-bd44189-44
____

Columns for Protocol | Chain | Iface are spread too wide

[image]

stangri, 2019-07-25:
AFAIK, this is auto-set by Luci, I have no control over this.

closed 2019-07-25 by stangri

n8v8R, 2019-07-27:
from the cross referenced upstream issue tracker

> There is no such app in this repo, please clarify your issues with the developer directly. If you're are not satisfied with the measurements of the default templates, you can change nearly every aspect of your app design with a bit of customization of these.

n8v8R, 2019-07-27:
appears to be a bug in the material theme https://github.com/openwrt/luci/issues/2927