mossdef-org/source.mossdef.org #71: [bug]  started on iface with errors

n8v8R, 2019-08-09:
https://forum.turris.cz/t/vpn-policy-based-routing-possible/7204/67

stangri, 2019-08-10:
Also here: https://forum.openwrt.org/t/vpn-policy-based-routing-web-ui-discussion/10389/515

I've tried to lessen dependency on ```ifconfig``` as it's due to be deprecated, but I'm guessing the new way of determining the gateway is not working in some cases. 

Please roll back to the previous version from the repo (check previous commits) and I'm asking anyone affected to post the output of:
```sh
ip -4
ip -4 r
route -4
```

I'll try to fix it asap.

p4p4-GitHub, 2019-08-10:
Here is my output, I'm using Turris OS 4.0 beta 8 and the newest vpn-policy-routing 0.0.7-15 with OpenVPN:

 root@turris:~# ip -4 r
0.0.0.0/1 via 10.XX.X.61 dev tun0 
default via 185.XXX.XX.1 dev eth2 proto static src 185.XXX.XX.188 
10.XX.X.1 via 10.XX.X.61 dev tun0 
10.XX.X.61 dev tun0 proto kernel scope link src 10.XX.X.62 
10.XXX.XXX.0/24 dev br-guest_turris proto kernel scope link src 10.XXX.XXX.1 linkdown 
128.0.0.0/1 via 10.XX.X.61 dev tun0 
185.XXX.XX.0/22 dev eth2 proto kernel scope link src 185.XXX.XX.188 
185.XXX.XXX.38 via 185.XXX.XX.1 dev eth2 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
root@turris:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.XX.0.61      128.0.0.0       UG    0      0        0 tun0
default         ve-92-mlx-XXX-1 0.0.0.0         UG    0      0        0 eth2
10.XX.0.1       10.XX.0.61      255.255.255.255 UGH   0      0        0 tun0
10.XX.0.61      *               255.255.255.255 UH    0      0        0 tun0
10.XXX.XXX.0    *               255.255.255.0   U     0      0        0 br-guest_turris
128.0.0.0       10.XX.0.61      128.0.0.0       UG    0      0        0 tun0
185.XXX.XX.0    *               255.255.252.0   U     0      0        0 eth2
185.XXX.XXX.38  ve-92-mlx-XXX-1 255.255.255.255 UGH   0      0        0 eth2
192.168.1.0     *               255.255.255.0   U     0      0        0 br-lan
root@turris:~#

n8v8R, 2019-08-10:
@stangri though for a different package the issue of gateway discovery been discussed here https://gitlab.labs.nic.cz/turris/foris-controller/issues/74, just in case of interest

meantime, the requested output [routing.txt](https://github.com/stangri/openwrt_packages/files/3488508/routing.txt)

n8v8R, 2019-08-10:
@stangri 
> Please roll back to the previous version from the repo (check previous commits)

That is not working really on TOS since it utilises an `opkg`  wrapper for system maintenance that does not allow to specify package releases, e.g. `vpn-policy-routing_0.0.7-11` does not work.

stangri, 2019-08-10:
> That is not working really on TOS since it utilises an `opkg` wrapper for system maintenance that does not allow to specify package releases, e.g. `vpn-policy-routing_0.0.7-11` does not work.

Switch to the commit of the last known version (https://github.com/stangri/openwrt-repo/tree/fbf9e01937071f4deceb78877010d6226755e066) and download the ipk to your router for local install.

Jorman, 2019-08-10:
Same problem here. With 0.7-15
```
Sat Aug 10 16:01:57 2019 user.notice vpn-policy-routing [6935]: Creating table 'wan/192.168.1.1' [✓]
Sat Aug 10 16:01:58 2019 user.notice vpn-policy-routing [6935]: Creating table 'PIA_VPN/10.14.10.6 peer 10.14.10.5' [✗]
Sat Aug 10 16:01:58 2019 user.notice vpn-policy-routing [6935]: Creating table 'VPN_SERVER/10.8.0.1' [✓]
Sat Aug 10 16:01:58 2019 user.notice vpn-policy-routing [6935]: Routing 'Nas' via PIA_VPN [✓]
Sat Aug 10 16:01:58 2019 user.notice vpn-policy-routing [6935]: service started on wan/192.168.1.1 VPN_SERVER/10.8.0.1 with errors [✗]
Sat Aug 10 16:01:58 2019 user.notice vpn-policy-routing [6935]: ERROR: Failed to set up 'PIA_VPN/10.14.10.6 peer 10.14.10.5'
Sat Aug 10 16:01:59 2019 user.notice vpn-policy-routing [6935]: service monitoring interfaces: wan PIA_VPN VPN_SERVER [✓]
```
After force a downgrade to 0.7-7 all is working like expected

stangri, 2019-08-10:
Please test 0.0.7-17. In the service status it now marks the default interface and it no longer prints the list of monitored interfaces, since it's a duplicate of the previous line pretty much.

Jorman, 2019-08-10:
In my side not working:
```
Sat Aug 10 16:39:46 2019 user.notice vpn-policy-routing [15522]: Creating table 'wan/192.168.1.1[✓]' [✓]
Sat Aug 10 16:39:47 2019 user.notice vpn-policy-routing [15522]: Creating table 'PIA_VPN/0.0.0.0' [✓]
Sat Aug 10 16:39:47 2019 user.notice vpn-policy-routing [15522]: Creating table 'VPN_SERVER/0.0.0.0' [✓]
Sat Aug 10 16:39:48 2019 user.notice vpn-policy-routing [15522]: Routing 'Nas' via PIA_VPN [✓]
Sat Aug 10 16:39:48 2019 user.notice vpn-policy-routing [15522]: service started on (strict mode): wan/192.168.1.1[✓] PIA_VPN/0.0.0.0 VPN_SERVER/0.0.0.0
```
The nas (this's the only device under pbr) can't use internet, but with 0.7-7
```
opkg install --force-downgrade vpn-policy-routing_0.0.7-7_all.ipk
Downgrading vpn-policy-routing on root from 0.0.7-17 to 0.0.7-7...
Stopping service and removing rc.d symlink for vpn-policy-routing
vpn-policy-routing 0.0.7-17 stopped [✓]
vpn-policy-routing 0.0.7-17 stopped [✓]
Command failed: Not found
Configuring vpn-policy-routing.
Creating table 'wan/192.168.1.1' [✓]
Creating table 'PIA_VPN/10.14.10.5' [✓]
Creating table 'VPN_SERVER/10.8.0.1' [✓]
Routing 'Nas' via PIA_VPN [✓]
vpn-policy-routing 0.0.7-7 started on wan/192.168.1.1 PIA_VPN/10.14.10.5 VPN_SERVER/10.8.0.1 [✓]
vpn-policy-routing 0.0.7-7 monitoring interfaces: wan PIA_VPN VPN_SERVER [✓]
```
And the nas is now under vpn, and all the lan over normal isp

n8v8R, 2019-08-10:
@stangri  0.0.7-17 works for me, i.e. routing as expected. Just prints a weird syntax in LuCI

> Success: wan/xxx.46.104.107\033[0;32m[\xe2\x9c\x93]\033[0m wg0/192.168.45.2

stangri, 2019-08-10:
> In my side not working:
Can you e-mail me or post the unsanitized output of: ```ubus call network.interface.PIA_VPN status```

Jorman, 2019-08-10:
I don't have your email, maybe a pm over openwrt forum?
Btw I don't see any personal data to be masked, so I paste here the code:
```
root@OpenWrt:~# ubus call network.interface.PIA_VPN status
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 91493,
        "l3_device": "tun1",
        "proto": "none",
        "device": "tun1",
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [

        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ]
        },
        "data": {

        }
}
```
Let me know if you need more data

h4waii, 2019-08-10:
@stangri 0.0.7-17 not working for me.

`ip -4 r`
```
default via 192.168.0.1 dev eth1.2 proto static 
10.8.0.0/24 dev tun0 proto kernel scope link src 10.8.0.4 
10.10.10.0/24 dev br-lan proto kernel scope link src 10.10.10.1 
192.168.0.0/24 dev eth1.2 proto kernel scope link src 192.168.0.50 
192.168.2.0/24 dev WIREGUARD proto kernel scope link src 192.168.2.1 
192.168.3.0/24 dev wlan1-1 proto kernel scope link src 192.168.3.1 
192.168.4.0/24 dev br-GUEST_OPEN proto kernel scope link src 192.168.4.1 
```
`route`
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1     0.0.0.0         UG    0      0        0 eth1.2
10.8.0.0        *               255.255.255.0   U     0      0        0 tun0
10.10.10.0      *               255.255.255.0   U     0      0        0 br-lan
192.168.0.0     *               255.255.255.0   U     0      0        0 eth1.2
192.168.2.0     *               255.255.255.0   U     0      0        0 WIREGUARD
192.168.3.0     *               255.255.255.0   U     0      0        0 wlan1-1
192.168.4.0     *               255.255.255.0   U     0      0        0 br-GUEST_OPEN
```

stangri, 2019-08-11:
Thank you for all the feedback! I'll need to run some more experiments with various tunnels locally before publishing the updates, for now I've reverted to the old gateway detection if the new system fails.

Please try vpn-policy-routing 0.0.7-19 and luci app version 47.

n8v8R, 2019-08-11:
@stangri 0.0.7-19 and luci app version 47 works for me

p4p4-GitHub, 2019-08-11:
@stangri 0.0.7-19 and luci app version 47 works for me too! Thx!

Jorman, 2019-08-11:
Just updated, the revert function works.
I only see one error when upgrading, I don't have much log, hope you can find what is from this:
```
Simple OPKG Updater v0.3.5

Done | Updating package lists
Done | Getting upgradable packages list

Packages available for upgrade: 4

+-----+-----------------------------+-----------------------------+-----------------------------+
|   # | Package                     | Current                     | Update                      |
+-----+-----------------------------+-----------------------------+-----------------------------+
|   1 | luci-app-simple-adblock     | git-19.221.45989-15e9526-32 | git-19.222.06794-9def6fa-32 |
|   2 | luci-app-vpn-policy-routing | git-19.222.06794-9def6fa-46 | git-19.222.06794-9def6fa-47 |
|   3 | simple-adblock              | 1.7.1-13                    | 1.7.1-15                    |
|   4 | vpn-policy-routing          | 0.0.7-7                     | 0.0.7-19                    |
+-----+-----------------------------+-----------------------------+-----------------------------+

Proceed with upgrade? (Y/y to proceed) y

.... | Upgrading packages

Upgrading luci-app-simple-adblock on root from git-19.221.45989-15e9526-32 to git-19.222.06794-9def6fa-32...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/luci-app-simple-adblock_git-19.222.06794-9def6fa-32_all.ipk
Upgrading luci-app-vpn-policy-routing on root from git-19.222.06794-9def6fa-46 to git-19.222.06794-9def6fa-47...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/luci-app-vpn-policy-routing_git-19.222.06794-9def6fa-47_all.ipk
Upgrading simple-adblock on root from 1.7.1-13 to 1.7.1-15...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/simple-adblock_1.7.1-15_all.ipk
Stopping service and removing rc.d symlink for simple-adblock
Stopping simple-adblock 1.7.1-13...
Reloading dnsmasq [✓]
simple-adblock 1.7.1-13 stopped.
Upgrading vpn-policy-routing on root from 0.0.7-7 to 0.0.7-19...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/vpn-policy-routing_0.0.7-19_all.ipk
Stopping service and removing rc.d symlink for vpn-policy-routing
vpn-policy-routing 0.0.7-7 stopped [✓]
vpn-policy-routing 0.0.7-7 stopped [✓]
Command failed: Not found
Configuring vpn-policy-routing.
Creating table 'wan/192.168.1.1[✓]' [✓]
Creating table 'PIA_VPN/10.14.10.5' [✓]
Creating table 'VPN_SERVER/10.8.0.1' [✓]
Routing 'Nas' via PIA_VPN [✓]
vpn-policy-routing 0.0.7-19 started on wan/192.168.1.1[✓] PIA_VPN/10.14.10.5 VPN_SERVER/10.8.0.1
Configuring simple-adblock.
Starting simple-adblock 1.7.1-15...
[DL] Blocked Hosts:   raw.githubusercontent.com [✓]
[DL] Blocked Domains: ssl.bblck.me [✓]
[DL] Blocked Domains: s3.amazonaws.com [✓]
[DL] Blocked Hosts:   adaway.org [✓]
[DL] Blocked Domains: dshield.org [✓]
[DL] Blocked Domains: ransomwaretracker.abuse.ch [✓]
[DL] Blocked Hosts:   pgl.yoyo.org [✓]
[DL] Blocked Domains: s3.amazonaws.com [✓]
[DL] Blocked Domains: s3.amazonaws.com [✓]
[DL] Blocked Hosts:   www.malwaredomainlist.com [✓]
[DL] Blocked Hosts:   winhelp2002.mvps.org [✓]
[DL] Blocked Hosts:   someonewhocares.org [✓]
[DL] Blocked Domains: mirror1.malwaredomains.com [✓]
Sorting combined list [✓]
Optimizing combined list [✓]
Whitelisting domains [✓]
Formatting merged file [✓]
Creating DNSMASQ servers file [✓]
Removing temporary files [✓]
Restarting DNSMASQ [✓]
simple-adblock 1.7.1-15 is blocking 47980 domains [✓]
Configuring luci-app-vpn-policy-routing.
uci: Entry not found
uci: Entry not found
uci: Entry not found
uci: Entry not found
Configuring luci-app-simple-adblock.
uci: Entry not found
uci: Entry not found
uci: Entry not found
uci: Entry not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/simple-adblock is different from the conffile in the new package. The new conffile will be placed at /etc/config/simple-adblock-opkg.
 * resolve_conffiles: Existing conffile /etc/config/vpn-policy-routing is different from the conffile in the new package. The new conffile will be placed at /etc/config/vpn-policy-routing-opkg.
Done |
Upgrade finished

Please check for config file conflicts!
```

stangri, 2019-08-12:
I don't know why the ```uci: Entry not found``` comes up, but the ```resolve_conffiles``` can be ignored.

Jorman, 2019-08-12:
Ok, and what about this?
vpn-policy-routing 0.0.7-7 stopped [✓]
Command failed: Not found
Configuring vpn-policy-routing.

stangri, 2019-08-13:
> Command failed: Not found

Fixed in 0.0.7-21.

stangri, 2019-08-14:
Please reopen if 0.0.7-21 cannot detect gateways.

closed 2019-08-14 by stangri