mossdef-org/source.mossdef.org #74: [VPBR bug] routing not working with multiple WG instances

n8v8R, 2019-08-30:
* {"kernel":"4.14.140","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"cznic,turris-omnia","release":{"distribution":"TurrisOS","version":"5.0-dev","revision":"456191c","target":"mvebu/cortexa9","description":"TurrisOS 5.0-dev 456191c"}}
* above is based on OpenWrt 19.x
* vpn-policy-routing 0.1.0-18
____
To keep it simple just two WG instances are up and running and only one (WG1) to be routed trough.

From the log [vpbr info.txt](https://github.com/stangri/openwrt_packages/files/3560794/vpbr.info.txt)  it would appear that the routing table is created but not the `ipt` rules pertaining to WG1

stangri, 2019-08-30:
I see both TCP and UDP iptables rules for to_lan created with the counters running.

Are both wg0 and wg1 connected to Mullvad?

n8v8R, 2019-08-30:
Huh, missed it then.

One instance (WG0) is connected to my VPS node, and works as expected when routed. The other instance (WG1) connects to some VPN endpoint (not Mullvad though) and there is no traffic.

The problem is that I cannot run a trace since setting a designated ICMP iface does not work (an issue I meant to open) - router client ICMP traffic escapes via ISP WAN.

stangri, 2019-08-30:
I suspect there's something going on with the VPN endpoint.

n8v8R, 2019-08-30:
Uhm, bummer :woman_facepalming: had the iface in the wrong fw zone :face_with_head_bandage: 

Works now, pardon for having wasted your valuable time. Will look into the ICMP issue more in detail prior opening an issue, though it really seems not to be working.
____

Off topic - any idea whether nested VPN routing works - one VPN over another VPN?

closed 2019-08-30 by n8v8R