mossdef-org/source.mossdef.org #75: [VPBR bug] ICMP per iface not working

n8v8R, 2019-08-31:
* {"kernel":"4.14.141","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"cznic,turris-omnia","release":{"distribution":"TurrisOS","version":"5.0-dev","revision":"fdcbf64","target":"mvebu/cortexa9","description":"TurrisOS 5.0-dev fdcbf64"}}
* vpn-policy-routing 0.1.0-18
____
as per [vpbr info.txt](https://github.com/stangri/openwrt_packages/files/3562337/vpbr.info.txt) ICMP is appended to `ipt` output chain.

Which seems curious, least my understanding that chain does not perform routing but permits egress traffic but maybe I am wrong, though would expect it in the same chain as tcp/udp.

Tracert/traceroute from W10/Android client however is escaping via WAN only.

stangri, 2019-08-31:
What happens if you do traceroute from the router?

Also, based on the status output, can you please post/PM me the output of:
```sh
which dnsmasq
uci get network.wan.ifname
```

n8v8R, 2019-08-31:
The router's self generated egress is not set to be routed via VPN, only its clients.

`dnsmasq` is not installed, using `odhcpd`

` uci get network.wan.ifname`
> eth2

stangri, 2019-08-31:
Please try 0.1.0-20.

n8v8R, 2019-08-31:
Unfortuntely the same outcome.

One curious thing is happening when the ICMP rule is enabled - 2FA at Gitgunb signin fails in FF 
> PR_CONNECT_RESET_ERROR

Once the ICMP rule is disabled the error does not show.
____
Why ICMP is appended to ipt output chain and not to prerouting as tcp/udp does?

stangri, 2019-09-11:
> Why ICMP is appended to ipt output chain and not to prerouting as tcp/udp does?

It's been a while since I've looked into it and implemented it, but as far as I remember, that's for the ICMP traffic originating from the router itself. You can create a separate policy to affect the ICMP traffic of the client devices themselves.

n8v8R, 2019-09-11:
How would you go about the client policy with the protocol selection is limited to `tcp` | `udp`?

stangri, 2019-09-11:
Ah, right. Can you insert an iptables rule manually to see if it achieves the desired effect?

You can have it in another custom user file then.

n8v8R, 2019-09-20:
Noticed that the supported protocols are now configurable via LuCI (`ALL` and `ICMP` already added) which supposedly sort this matter, hence closing.

Cannot test at the moment since the remote endpoint is currently down, will revisit only if necessary when I can test.

Thanks for the effort of having added the code!

closed 2019-09-20 by n8v8R