mossdef-org/source.mossdef.org #76: [VPR bug] Refactoring broke my VPR

yitzhaq, 2019-09-01:
Hey,

Ever since commit 96df7ae725dc6c77ef19568ebd2da73a39661072 on July 24th, where you did "major refactoring" (and where MAC support was seemingly added), VPR has stopped working for me; none of my rules are applying. iptables-save shows they are being set, but they do not take effect. Rolling back to the previous commit, with no changes to config, makes it work again. Every version I have tested since that commit have failed similarly.

Happy to provide any output that would help you pin down the problem.

Cheers

stangri, 2019-09-11:
Great report, looking forward to part II. ;)

yitzhaq, 2019-09-11:
Heh, well.. IANAD, and with a substantial commit containing a "major refactoring" I didn't know quite where to start. I also didn't really want to spam the report with a lot of potentially unrelated details[1], and my hope was that by identifying the commit which introduced the breakage, it might give you an idea of potential candidates, which in turn could help me drill down further. As mentioned, I'm happy to provide any detail which might be helpful, and/or make this report, uh, greater.

[1] ```# iptables-save | wc -l```
```2409```

stangri, 2019-09-11:
Full VPR config + output of status with both old and new version.

Are you using mwan3 by any chance?

yitzhaq, 2019-09-11:
Nope, not using mwan3. I did notice a possible source of the problem when doing a manual start/stop, some errors that weren't logged.

/etc/rc.common: eval: line 2: ::33af:cd8d/96: not found
Creating table 'wan/eth1/51.170.128.1/2a01:79c:abcd:3674::1/30' [✓]
Creating table 'wireguard0/0.0.0.0/fc00:bbbb:bbbb:bb01::846/128' [✓]
/etc/rc.common: eval: line 2: fe80::7fb5:f6b3:ed48:6c04/64: not found
Creating table 'tun0/10.15.0.12/fdda:d0d0:cafe:1302::102a/64' [✓]

So, seemingly related to IPv6. Given what you were adding in the commit in question (MAC address support), I guess there needs to be some evaling to determine if something is a MAC address or an IPv6 address - could that be the source somehow?

These are affected interfaces:

6rd-wan_6 Link encap:IPv6-in-IPv4
          inet6 addr: 2a01:79c:abcd:3674::1/30 Scope:Global
          inet6 addr: ::33af:cd8d/96 Scope:Compat
          UP RUNNING NOARP  MTU:1280  Metric:1
          RX packets:180315 errors:0 dropped:0 overruns:0 frame:0
          TX packets:193545 errors:45 dropped:556 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:81458286 (77.6 MiB)  TX bytes:39764410 (37.9 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.15.0.12  P-t-P:10.15.0.12  Mask:255.255.0.0
          inet6 addr: fe80::7fb5:f6b3:ed48:6c04/64 Scope:Link
          inet6 addr: fdda:d0d0:cafe:8302::200a/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:1307477 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1391288 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:659662253 (629.1 MiB)  TX bytes:365009752 (348.0 MiB)

(Random characters changed in IPv6 addresses for posting, but should presumably not affect parsing)

If this leads nowhere, I can email you the details you requested directly.

stangri, 2019-09-12:
Have you tried disabling IPv6 in VPR settings?

Also, could you please PM/email me the output of `ip -6 a list`? You can obfuscate things, I just need to know the format.

Thanks!

closed 2019-09-27 by stangri

yitzhaq, 2021-02-11:
I see now I dropped the ball on this report - sorry about that. The issue turned out to be embarrassingly stupid; I was testing routing with traceroute, but my rules were all TCP/UDP. D'oh - major facepalm moment. Dunno why it worked before with ICMP even when set to TCP/UDP, but when set to ALL it's working as expected. Sorry for disturbing!