mossdef-org/source.mossdef.org #77: VPR: error accessing Luci

craibuc, 2019-09-11:
I restarted my router and attempted to access the Luci interface (http://192.168.1.1/cgi-bin/luci).  

When I did, I got this error:

```
/usr/lib/lua/luci/controller/vpn-policy-routing.lua:4: attempt to call field 'lookup' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/vpn-policy-routing.lua:4: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:563: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:220: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```
Foris works as expected.

Is there a way to temporarily disable VPR via `ssh` so I can access Luci?

stangri, 2019-09-11:
What's a foris?

Which version of OpenWrt?

Which version of luci-app-vpn-policy-routing?

You can try removing the above-mentioned app from console to regain access
to Luci.

craibuc, 2019-09-11:
Foris is the Omnia's "simple" interface, accessible via http://192.168.1.1/foris/config/.

It's Turris OS v3.11.6.  I don't know if this corresponds w/ OpenWRT, however.

These packages were installed on 9/7/19:
 • Installed version 0.1.1-1 of package vpn-policy-routing
 • Installed version git-19.249.68069-204293a-56 of package luci-app-vpn-policy-routing

stangri, 2019-09-11:
Thank you, could you please test it with luci-app-vpn-policy-routing version git-19.254.43391-5e7e9b5-57?

craibuc, 2019-09-11:
OK.

How do I reference a package from Github?

`# opkg install ?`

stangri, 2019-09-11:
How did you install these packages in the first place?

craibuc, 2019-09-11:
Via Luci.  As I can't access Luci, I'll need to install the package via opkg's CLI.

stangri, 2019-09-11:
> Via Luci.

Did you add my repo to your router?

craibuc, 2019-09-11:
Never mind.

# opkg update
...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/Packages.sig
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   163  100   163    0     0   1012      0 --:--:-- --:--:-- --:--:--  1012
Signature check passed.
...
# opkg install vpn-policy-routing luci-app-vpn-policy-routing
Installing vpn-policy-routing (0.1.1-1) to root...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/vpn-policy-routing_0.1.1-1_all.ipk
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 12144  100 12144    0     0  58951      0 --:--:-- --:--:-- --:--:-- 59239
Configuring vpn-policy-routing.
Creating table 'vpn_interface/tun0/0.0.0.0' [✓]
Creating table 'wan/eth1/72.50.209.209' [✓]
Routing 'Guest VPN' via vpn_interface [✓]
vpn-policy-routing 0.1.1-1 started on (strict mode): vpn_interface/tun0/0.0.0.0 wan/eth1/72.50.209.209[✓] 
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/vpn-policy-routing is different from the conffile in the new package. The new conffile will be placed at /etc/config/vpn-policy-routing-opkg.
Upgrading luci-app-vpn-policy-routing on root from git-19.249.68069-204293a-56 to **git-19.254.43391-5e7e9b5-57**...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/luci-app-vpn-policy-routing_git-19.254.43391-5e7e9b5-57_all.ipk
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  6144  100  6144    0     0  48000      0 --:--:-- --:--:-- --:--:-- 48377
Configuring luci-app-vpn-policy-routing.

craibuc, 2019-09-11:
Rebooted.

Your patched fixed the issue.

Thanks.

closed 2019-09-11 by craibuc