mossdef-org/source.mossdef.org #80: VPR not routing

FangsMcWolf, 2019-09-21:
There's a chance that it's something I did to mess things up, but for the life of me I can't figure it out.

I was on OpenWRT 18.06.02 then upgraded to 18.06.04.  While on .02, VPR was working fine (I did have another issue but not the fault of VPR and not on the router).  Now on .04, when I have a VPN enabled, it affects all devices, not just those that are supposed to be affected.

Strict enforcement enabled.
Using DNSMASQ (and DNSMASQ full is installed).
IPSET disabled for local policies.
IPv6 support disabled.

When reading another issue, I saw a reference to seeing the routing tables that are set up via DNSMASQ.  I'm wondering how to do this so that I can do some troubleshooting.  Also, wondering what else I should do in order to get more information.

[code]Sat Sep 21 01:32:54 2019 daemon.err modprobe: xt_set is already loaded
Sat Sep 21 01:32:54 2019 daemon.err modprobe: ip_set is already loaded
Sat Sep 21 01:32:54 2019 daemon.err modprobe: ip_set_hash_ip is already loaded
Sat Sep 21 01:32:54 2019 daemon.err uhttpd[3153]: /etc/rc.common: eval: line 2: fe80::c256:27ff:fe73:2019/64: not found
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Creating table 'wan/eth1/73.31.93.1' [✓]
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Creating table 'providervpn/tun0/0.0.0.0' [✓]
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Routing 'Plex_Ports' via wan [✓]
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Routing 'VPN_200-207' via providervpn [✓]
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Routing 'Plex_APP' via wan [✓]
Sat Sep 21 01:32:55 2019 user.notice vpn-policy-routing [24270]: Routing 'All' via wan [✓][/code]

Even though the only real "error" to wonder about doesn't seem to be related to VPR, I felt it best to leave it in the copy/paste of the little bit of system log in case it actually is relevant.

Anything you can guide me on in regards to making sure the interface (using tun0) is set up properly, as well as other things, would be greatly appreciated.  Since the VPN itself works, it seemed appropriate to ask you (VPR) for assistance in trying to narrow down what the issue is.

stangri, 2019-09-21:
I'm not sure what exactly are you reporting. Could you please post the steps to reproduce the bug, what's expected and what's really happening?

FangsMcWolf, 2019-09-21:
VPR isn't routing the VPN to only specified clients while routing WAN to other clients.

When adding the VPR package (the main one), an error occurs, similar to the one in the system log shown above.  The luci package installs without issue.  (The only other 'error' is the config file already existing, but of course that one can be ignored.)

When I enable a VPN server (ie, OpenVPN, start a selected server), all clients start using the VPN, not just the ones specified.  It's as though the routes aren't actually being applied.

FangsMcWolf, 2019-09-21:
So after toying a teeny bit, the error that is being given is being generated at about line 97 `eval "$1=${gw}"` except that I don't understand WHY it's generating an error.  I added a line to echo what it is trying to do and if I do it manually (like add a line to do it without using variables), then it doesn't show an error.  Still isn't working, but doesn't throw an error at me.  Any ideas what could be causing that?  I have a feeling that addressing that may fix the issue I'm having.

FangsMcWolf, 2019-09-21:
More digging and found that by uncommenting a couple of lines, my main issue clears up.

```#	elif is_netmask "$laddr" || is_netmask "$raddr"; then
#		insert_policy "$comment" "$iface" "$laddr" "$lport" "$raddr" "$rport" "$proto" "$chain"
```

stangri, 2019-09-22:
Thanks I'll check what could be happening at line 97, other than that there is no information here about vor version, configs, reproducibility, etc, so I'm closing this.

closed 2019-09-22 by stangri

FangsMcWolf, 2019-09-22:
Not even the part of removing a couple of #'s (from vpr.init) which fixes my issue?