mossdef-org/source.mossdef.org #81: Started "with errors"

FangsMcWolf, 2019-09-24:
This is an easy one.

Add a rule with multiple ports and set the protocol to "All," and it will tell you that errors were generated.  Turns out that iptables says that a protocol must be selected (ie, can't be "all") when using the multiport option.

I have the following installed:

iptables, iptables-mod-conntrack-extra, and iptables-mod-ipopt, all version 1.6.2-1

stangri, 2019-09-24:
Please test with vpn-policy-routing 0.1.3-3.

FangsMcWolf, 2019-09-24:
Okay cleared up in -3.

Something else I've noticed, not really a bug but just a cosmetics thing, is that there are extra blank lines after the tun0 entry (when listing the service status).  Not sure if you're aware of it or not.  Either way, I was on -1, didn't know there was a -3 out already.

stangri, 2019-09-24:
> Something else I've noticed, not really a bug but just a cosmetics thing, is that there are extra blank lines after the tun0 entry (when listing the service status). Not sure if you're aware of it or not. Either way, I was on -1, didn't know there was a -3 out already.

Can't reproduce.

closed 2019-09-27 by stangri