mossdef-org/source.mossdef.org #84: HTTPS DNS Proxy

systemcrash, 2019-10-03:
I see the quad entries in the LUA source. In my 18.06.04 GUI I see only CF+Goog. Are the more recent changes in your code for 19.x only?

It seems only bootstrapped with those two:
```
# uci show https_dns_proxy
https_dns_proxy.@https_dns_proxy[0]=https_dns_proxy
https_dns_proxy.@https_dns_proxy[0].listen_addr='127.0.0.1'
https_dns_proxy.@https_dns_proxy[0].listen_port='5053'
https_dns_proxy.@https_dns_proxy[0].user='nobody'
https_dns_proxy.@https_dns_proxy[0].group='nogroup'
https_dns_proxy.@https_dns_proxy[0].url_prefix='https://dns.google.com/resolve?'
https_dns_proxy.@https_dns_proxy[1]=https_dns_proxy
https_dns_proxy.@https_dns_proxy[1].bootstrap_dns='1.1.1.1,1.0.0.1'
https_dns_proxy.@https_dns_proxy[1].url_prefix='https://cloudflare-dns.com/dns-query?ct=application/dns-json&'
https_dns_proxy.@https_dns_proxy[1].user='nobody'
https_dns_proxy.@https_dns_proxy[1].group='nogroup'
https_dns_proxy.@https_dns_proxy[1].listen_addr='127.0.0.1'
https_dns_proxy.@https_dns_proxy[1].listen_port='5054'
```

Also no way to free-edit any new entries. Fixable?

stangri, 2019-10-03:
What is the bug you're reporting?

systemcrash, 2019-10-04:
Not a bug per se. I just added your repo to get luci-app-https_dns_proxy installed, and my 18.06.04 GUI only presented CF and google as options. Your lua code seemed to have more. 

I added them into the lua file manually in my ```/usr/lib/lua/luci/model/cbi/``` and they now show up, but using any of the quad 9 options seemed to be problematic for me. 

Also another point: The URIs included e.g. dns.google.com are looked up every 30 seconds or so via normal DNS UDP port 53 (bootstrapping?), but everything else goes DoH. 

I tried changing (after checking that the X509 certs also cover their corresponding IPs) the URIs to e.g. 8.8.8.8, and ```https://cloudflare-dns.com/dns-query?ct=application/dns-json``` to ```https://1.1.1.1/dns-query?ct=application/dns-json``` and my openwrt basically ground to a halt and the DNS daemon ```/usr/sbin/https_dns_proxy -a 127.0.0.1 -p 5054 -u nobody -g nogroup...``` ate up all available CPU after that. Running curl against those IP based URLs works fine from my desktop tho. *shrug*

I think it's beneficial if users can freely add/edit new entries, somehow.

stangri, 2019-10-04:
> I think it's beneficial if users can freely add/edit new entries, somehow.

You can edit the ```/etc/config/https_dns_proxy``` directly. 

Quad9 resolvers don't seem to work with the ```https_dns_proxy``` package, that's why I haven't pushed the Luci app update to the official repo. 

If there's no actual issue, please start a thread at OpenWrt forum next time.

closed 2019-10-04 by stangri