mossdef-org/source.mossdef.org #91: [vpn-policy-routing] Wish: rules pointing existing ipset set

y-alexey, 2019-12-15:
Please make it possible to directly point in the rules compliance with a particular ipset set. 
I have ready-made files for "ipset restore", parser and scripts for renew it.  These are many thousands of subnets and in this form it load much faster than one-by-one line.  If you app already have the opportunity to write a rules for matching a specific set, indicate how to do it.  I did not understand from the description.

stangri, 2019-12-15:
> Please make it possible to directly point in the rules compliance with a particular ipset set.

There already is a documentation with two sample scripts for user-defined custom scripts. If your question is about something else, re-open (and you may want to comment in Russian, because I can't make sense of what you're asking in English).

closed 2019-12-15 by stangri

y-alexey, 2019-12-15:
Ок
Попробую подробно.

Как известно, существует проблема с блокировками ркн. Существует список имён и адресов. Имеется масса ресурсов, где это можно так или иначе получить. Результат обработки лежит у меня в репозитории в виде оптимальном для загрузки в ipset/dnsmasq бытового роутера.

Сейчас это работает либо с заворотом всего совпадающего в прозрачный прокси tor с помощью port forward правил fw3, либо с помощью прямого указания имени ipset в правилах mwan3 для VPN интерфейса.

Хотелось бы в идеале получить приоритетную маршрутизацию в VPN, а если он лежит, то в tor. Я не уверен, что ваше решение может с этим помочь, потому и спрашиваю.

В примерах у вас просто списки. Загрузка списка подсетей и хостов в созданный вашим приложением сет может оказаться в разы дольше, чем выполнение команды ipset restore/swap. Вот в этом вопрос. Не указывать в правилах файл с хостами, а указывать имя сета с сотней тысяч позиций. Периодически обновляющегося внешним скриптом без перезагрузки конфигурации и изменения правил маршрутизации.

stangri, 2019-12-15:
Так сделайте rename/swap в скрипте.

y-alexey, 2019-12-15:
Я там выше поправил. Одним сетом можно обойтись только если это будет list. Но если ваша программа создаёт, например, hash:ip, то свопнуть его с list:set не выйдет.

stangri, 2019-12-15:
> Я там выше поправил. Одним сетом можно обойтись только если это будет list. Но если ваша программа создаёт, например, hash:ip, то свопнуть его с list:set не выйдет.

Ну поправьте тогда iptables rules в user-script. Я не понимаю чего вы хотите от VPR.

y-alexey, 2019-12-15:
Чего я хочу, я прямо сразу сказал - чтобы одним из возможных критериев в правилах было совпадение с сетом, указанным по имени. Это наглядно и интуитивно прозрачно, что и требуется от пользовательских интерфейсов, обычно.

Относительно остального я подумаю. Ночь уже тут.

stangri, 2019-12-15:
Эта возможность есть и реализуется через user-script в котором через ipt/iptables вы добавляете правила нацеленные на ваш сет. Можете посмотреть что создает программа через команду support и добавить аналогичные (я бы рекомендовал через ipt).

y-alexey, 2019-12-18:
Давайте терминологию как-то согласуем. Я считаю, что то, что реализовано через написание правил iptables руками, не реализовано с помощью fw3, vpr или mwan3. И наоборот, эти программы реализуют лишь те функции, для которых сами создают в результате команды iptables.

Я вот чего понять не могу. Как, где и в результате чего ваш пакет создаёт сет. (FW3, например, имеет для этого "config ipset") И по какому принципу его называет? Я не могу же указать в ваших правилах своё имя сета?

Допустим, у меня есть три сета, и мне нужно совпадения пихать в конкретный интерфейс. Насколько я понял, мне тут ваш пакет вообще не поможет. А поможет прямое написание правил iptables руками для каждого сета. Или, всё-таки, есть возможность как-то на сеты ссылаться? Или можно их собрать в мета сет и на него сослаться в правилах vpr?
Я вижу в примере имя сета, как бы, совпадает с именем целевого интерфейса. Это так работает?