mossdef-org/source.mossdef.org #92: [vpn-policy-routing] Issue: invalid conditional in insert_policy()

dl12345, 2020-03-22:
In the function `insert_policy()`, the following code behaves incorrectly

```
   if [ -z "$proto" ] || [ "$proto" = 'all' ]; then
        echo "${lport}${raddr}${rport} AND $raddr....."
        if [ -z "${lport}${raddr}${rport}" ] && [ -n "$laddr" ]; then
            proto='all'
        elif [ -z "${laddr}${lport}${rport}" ] && [ -n "$raddr" ]; then
            proto='all'
        elif [ -n "$lport" ] || [ -n "$rport" ]; then
            proto='tcp udp'
        else
            proto='tcp'
        fi
    fi
```

If, in the policy, a local subnet is specified and a remote subnet is specified, with protocol set to "ALL", this code will ensure that a rule will be added setting the protocol to only tcp, which is clearly not what is intended. 

I have such a rule since I don't want my guest network routing through a VPN tunnel and only certain destination subnets are accessed via this tunnel. The net result is a rule for only tcp added when in fact ALL was specified as the protocol.

I don't see why the `|| [ "$proto" = 'all' ]` check is needed. If the user wants all protocols, just respect it and set `proto = all`. 

So that conditional should read just `if [ -z "$proto" ]; then`

stangri, 2020-03-22:
> If, in the policy, a local subnet is specified and a remote subnet is specified, with protocol set to "ALL", this code will ensure that a rule will be added setting the protocol to only tcp, which is clearly not what is intended.

You're right, that's broken.

> I don't see why the || [ "$proto" = 'all' ] check is needed. If the user wants all protocols, just respect it and set proto = all.

It's the artifact of implementation of `AUTO` protocol option. Your suggestion seems to solve the problem with differentiating between `AUTO` and `all`. Let me sleep on it.

stangri, 2020-03-29:
Fixed in 0.2.1-5 available from my repo. If you could test it before I send the PR to openwrt repo, I'd appreciate it!

stangri, 2020-03-31:
PR: https://github.com/openwrt/packages/pull/11728, thank you!

closed 2020-03-31 by stangri